XCSoar/XCSoar #1569: Make speed to fly altitude dependent

flugplatz-berching, 2024-09-19:
XCSoar version
--------------
7.43
----------------------------------------------------------------------
After reading the XCS handbook I believe that the speed to fly is calculated altitude independent. At least the handbook does not mention, that altitude is taken into account.
However, in real life the polar ist slightly shifted due to altitude. See chapter 3 of following document.
https://journals.sfu.ca/ts/index.php/op/article/download/1796/1717

At moderate altitude, this makes little to no difference, however, if flying in wave it starts to become noticable.

Possibly altitude dependency is already in the code. I did not find where XCS does the polar calculations.
If this is already the case than this issue shall be closed.

-------------------------------------------------------------------

As XCS has already code to calculate the air density ratio in [AirDensity.cpp](https://github.com/XCSoar/XCSoar/blob/6696635126fc457cf03fe4f6c5eb57f09d7c598f/src/Atmosphere/AirDensity.cpp#L20) it should be possible to take the altitude dependency of the polar into account.

Unfortunately, I did not find where in the code the polar is adjusted for weight of glider, etc. Hints appreciated.

flugplatz-berching, 2024-09-20:
Here is my first try at modifying GlidePolar.cpp
https://github.com/XCSoar/XCSoar/commit/4d5dc87a0c43c08661ed8f80507ea40002a13dfe

Where is GlidePolar called from, so that I can update the added argument of the function?

flugplatz-berching, 2024-09-21:
Would calling SetAltitude in the BasicComputer.cpp make sense?

dspreitz, 2024-09-21:
I need help understanding following warning message:
src/Engine/GlideSolvers/GlidePolar.cpp:51: error: undefined reference to 'AirDensityRatio(double)'

The line [51](https://github.com/dspreitz/XCSoar/blob/6d31dc129b2021bc22f7a56322692a25542fb7d1/src/Engine/GlideSolvers/GlidePolar.cpp#L51) is:
  const auto density_factor = sqrt(AirDensityRatio(altitude));

I have an include statement in the file:
#include "Atmosphere/AirDensity.hpp"

groundhogxc, 2024-09-23:
I applaud any efforts to make XCS an ever more precise tool, but I would like you to look into two things:

First: Looking at the equations and figures in that paper, it appears to me that this is all "regular" polar calculations, which means the horizontal axis is TAS. Now, STF is usually used as IAS! So this means that there is already a transformation with altitude going on which does depend on density. It is not clear to me that Westerboer is taking this into account, but maybe I do not understand the paper well enough.

I have however discussed the question of altitude corrections with respect to *IAS* with others before, and our conclusion at the time was that it it snot worth it - but we might have been wrong!

And to my other point:
STF calculation assumes McReady theory; however, when flying in wave, fundamental assumptions about that theory do not apply. Most significantly the fact that in that theory it is assumed that any lift is moving with the airmass, and is not stationary, which is not the case with wave lift. This assumption then also means that wind needs not be taken into account in McReady-Theory, as you are basically optimizing for cross-country velocity *relative to the moving airmass*. This is also quite far fromt he situation in wave. 

I am not saying that it does not make sense to look at Dolphin speed or STF while flying wave, but it is not the "mathematically correct" speed you should fly. Going against the wind you need to go even faster, with a tailwind you go slower than STF etc.
That being said, there are regions in the world where thermals go up to 5000m, so maybe it starts to become interesting there. But as I mentioned in a), I think the error is smaller than you think from looking at the figures in the paper.

But again I might be totally wrong, and I would love if someone could explain it to me.

groundhogxc, 2024-09-23:
Cool! Thank you for the reference. Very interesting!

dspreitz, 2024-09-23:
Dear [groundhogxc](https://github.com/groundhogxc),

thanks for your review and comments.

> First: Looking at the equations and figures in that paper, it appears to me that this is all "regular" polar calculations, which means the horizontal axis is TAS.

In a polar diagram both axis are usually expressed as normed values, i.e. excluding the effect of altitude on the value - this corresponds to IAS.
See following link on the [IDAFlieg](https://idaflieg.de/?page_id=1802) site: [Link](https://www.j2mcl-planeurs.net/dbj2mcl/planeurs-biblio/fac-similes/Ermittlung_der_Flugleistung_von_Segelflugzeugen_(Wende_2003).pdf) Page 55

Your comments on the applicability of McReady theory in wave are obviously true.

My understanding of the XCSoar code is poor, but from what I read in [GlidePolar.cpp](https://github.com/XCSoar/XCSoar/blob/39f49e8ed2727793ee66817edbdac932b2f181dc/src/Engine/GlideSolvers/GlidePolar.cpp#L51) I believe all the calculations are done on polar coefficient level, which are given in IAS, not TAS.

In my understanding (which may be wrong, then pls educate me) the effect of altitude/density should be taken into account in the polar calculations/coefficients.
[SinkRate](https://github.com/XCSoar/XCSoar/blob/39f49e8ed2727793ee66817edbdac932b2f181dc/src/Engine/GlideSolvers/GlidePolar.cpp#L117) is slightly different, when you take altitude into account.

groundhogxc, 2024-09-24:
> In a polar diagram both axis are usually expressed as normed values, i.e. excluding the effect of altitude on the value - this corresponds to IAS. See following link on the [...]

Well, I think we mean the same thing! But it is tricky to define. Not sure how to read it from that page 55, though.

Let me unroll what I meant:
When a polar is given, it is usually for MSL ~0 at standard conditions. So that means there is IAS=TAS anyway.
Now, when people do things like the stretching with sqrt(density) to transform to higher altitude, *the x axis is still TAS*.
That is all I meant. So basically all polar calculations happen in TAS usually.

So to get to the IAS table of the Dutch guys, they computed the polar and then STF in TAS and then converted the values to IAS.

I accidentally found some old notes about this, but only the first page - from Summer 2012, ahem. Since it is not complete, not sure how we got to our result. Either an error or we just never considered such high McReady values and then maybe the effect is smaller :-).
If I find some time I will do the calculation for myself just to make sure I understand.