XCSoar/XCSoar #1570: Multiple failures at starup if Page0 has InfoBoxes = none

dssherrill, 2024-09-20:
XCSoar versions having the problem
-------------------------------------------------
XCSoar v7.43-Android

XCSoar versions not having the problem
-------------------------------------------------
XCSoar v7.42-Android

System information
------------------
Samsung Galaxy S10+ model SM-G975U with Android 12
also
Sony Xperia model D6603 with Android 6.0.1

Steps to reproduce the behavior
-------------------------------------------------
Open Config -> System -> Look -> Pages and set the first page to InfoBoxes = None
Exit XCSoar and restart
If possible, swipe from Page0 to Page1

Expected behavior
-----------------
Normal map displayed

Actual behavior
---------------
The to 10% of the map is garbage.  If the layout is Reverse Landscape, the entire map is garbage.

Now try an input gesture.  Notice that the red trace is offset from your finger.  

Open the main menu by double tapping or drawing the check mark gesture and try to click the Quit button.  You must touch on the upper half of the button to activate it.   Next, in the Yes/No pop-up dialog, you must touch above the top of the "YES" button to activate it.  

Do you have any idea what may have caused this?
-----------------------------------------------
No, but I can provide two user profiles that differ by only one line, but one file works and the other fails.

Do you have an idea how to solve the issue?
-------------------------------------------
No, but the same problem exists in the XCSoar JET version found on the Google Play Store (version 7.43_JET_1.23 Update v1.23 [16 Jun 2024]).  That suggests the bug was introduced before 16 June 2024

Screenshot using "Good" profile
-------------------------------------------
[image]

Screenshots using "Bad" profile
-------------------------------------------
Top of map is missing
[image]

Another view with top of map missing
[image]

Page 1 with InfoBoxes shifted down and partially off the screen; sometimes they seem to wrap around to the top of the screen
[image]

Page 0 if Reverse Landscape is also set
[image]

dssherrill, 2024-09-20:
An Android hardware dependency seems a little unlikely.  Perhaps this bug is an interaction between more than one profile setting.  Attaching 2 profiles with .TXT appended to the file names to allow uploading.  The "good" file works.  To create the "bad" file I used XCSoar to copy the "good" profile to a new profile named "bad", clicked "Select", and clicked "Continue".  After XCSoar was running I changed the Page0 InfoBox selection from "Auto" to "None".  

[good.prf.txt](https://github.com/user-attachments/files/17073553/good.prf.txt)
[bad.prf.txt](https://github.com/user-attachments/files/17073554/bad.prf.txt)

lordfolken, 2024-09-20:
I was able now to reproduce it on a Nokia, and also on the Pixel. its less evident on the pixel, as no corruption is displayed. 

The main culprint seems to be the fullscreen option. if that is off, all works normal. Can you confirm?

dssherrill, 2024-09-20:
Confirmed.  When full screen is off, XCSoar runs normally even if Page0 has InfoBoxes=None.
Tested on Samsung Galaxy S10+

dssherrill, 2024-09-20:
However, full screen off produces the problem if Display orientation = Reverse Landscape.

lordfolken, 2024-09-26:
Reverse landscape and reverse portrait modes are borked even in the Linux target.

lordfolken, 2025-01-16:
So commenting this line moves the map on Android when there is no Infoboxen with only a Page=0.

[src/ui/window/Window.hpp](https://github.com/XCSoar/XCSoar/blob/273cc7f332972732a379c0aa4430d0e4f58eff27/src/MainWindow.cpp#L859)

Further investigation pending.