XCSoar/XCSoar #1582: MACOSX Target Failing

lordfolken, 2024-10-02:
XCSoar versions having the problem
-------------------------------------------------

Current head

XCSoar versions not having the problem
-------------------------------------------------

System information
------------------

Steps to reproduce the behavior
-------------------------------

Expected behavior
-----------------

The Pipeline to pass.

Actual behavior
---------------

Fails to compile with following errors:

```
 /usr/local/include/boost/function_types/detail/pp_cc_loop/preprocessed.hpp:51:24: note: expanded from macro 'BOOST_FT_cond'
# define BOOST_FT_cond BOOST_FT_CC_PASCAL
                       ^
/usr/local/include/boost/function_types/detail/pp_cc_loop/preprocessed.hpp:64:6: error: 'BOOST_FT_CC_FASTCALL' is not defined, evaluates to 0 [-Werror,-Wundef]
# if BOOST_FT_cond
     ^
/usr/local/include/boost/function_types/detail/pp_cc_loop/preprocessed.hpp:63:24: note: expanded from macro 'BOOST_FT_cond'
# define BOOST_FT_cond BOOST_FT_CC_FASTCALL
                       ^
/usr/local/include/boost/function_types/detail/pp_cc_loop/preprocessed.hpp:76:6: error: 'BOOST_FT_CC_CLRCALL' is not defined, evaluates to 0 [-Werror,-Wundef]
# if BOOST_FT_cond
     ^
/usr/local/include/boost/function_types/detail/pp_cc_loop/preprocessed.hpp:75:24: note: expanded from macro 'BOOST_FT_cond'
# define BOOST_FT_cond BOOST_FT_CC_CLRCALL
                       ^
/usr/local/include/boost/function_types/detail/pp_cc_loop/preprocessed.hpp:88:6: error: 'BOOST_FT_CC_THISCALL' is not defined, evaluates to 0 [-Werror,-Wundef]
# if BOOST_FT_cond
     ^
/usr/local/include/boost/function_types/detail/pp_cc_loop/preprocessed.hpp:87:24: note: expanded from macro 'BOOST_FT_cond'
# define BOOST_FT_cond BOOST_FT_CC_THISCALL
                       ^
/usr/local/include/boost/function_types/detail/pp_cc_loop/preprocessed.hpp:100:6: error: 'BOOST_FT_CC_IMPLICIT_THISCALL' is not defined, evaluates to 0 [-Werror,-Wundef]
# if BOOST_FT_cond
     ^
/usr/local/include/boost/function_types/detail/pp_cc_loop/preprocessed.hpp:99:24: note: expanded from macro 'BOOST_FT_cond'
# define BOOST_FT_cond BOOST_FT_CC_IMPLICIT_THISCALL
                       ^
In file included from src/Computer/GlideComputer.cpp:4:
In file included from src/Computer/GlideComputer.hpp:11:
In file included from src/Computer/TaskComputer.hpp:6:
In file included from src/Computer/RouteComputer.hpp:6:
In file included from ./src/Task/ProtectedRoutePlanner.hpp:6:
In file included from ./src/Task/RoutePlannerGlue.hpp:6:
In file included from ./src/Engine/Route/AirspaceRoute.hpp:7:
In file included from ./src/Engine/Airspace/Airspaces.hpp:7:
In file included from ./src/Engine/Airspace/AirspacesInterface.hpp:9:
In file included from /usr/local/include/boost/geometry/index/rtree.hpp:33:
In file included from /usr/local/include/boost/geometry/algorithms/detail/covered_by/interface.hpp:22:
In file included from /usr/local/include/boost/geometry/algorithms/detail/within/interface.hpp:32:
In file included from /usr/local/include/boost/geometry/strategies/concepts/within_concept.hpp:26:
In file included from /usr/local/include/boost/function_types/result_type.hpp:19:
In file included from /usr/local/include/boost/function_types/is_callable_builtin.hpp:14:
In file included from /usr/local/include/boost/function_types/components.hpp:414:
In file included from /usr/local/include/boost/function_types/detail/pp_loop.hpp:52:
/usr/local/include/boost/function_types/detail/pp_cc_loop/preprocessed.hpp:16:6: error: 'BOOST_FT_CC_IMPLICIT' is not defined, evaluates to 0 [-Werror,-Wundef]
[... 210 lines cut ...]
In file included from ./src/Task/ProtectedRoutePlanner.hpp:6:
In file included from ./src/Task/RoutePlannerGlue.hpp:6:
In file included from ./src/Engine/Route/AirspaceRoute.hpp:7:
In file included from ./src/Engine/Airspace/Airspaces.hpp:7:
In file included from ./src/Engine/Airspace/AirspacesInterface.hpp:9:
In file included from /usr/local/include/boost/geometry/index/rtree.hpp:49:
/usr/local/include/boost/geometry/index/detail/exception.hpp:27:1: error: function 'throw_runtime_error' could be declared with attribute 'noreturn' [-Werror,-Wmissing-noreturn]
{
^
/usr/local/include/boost/geometry/index/detail/exception.hpp:32:1: error: function 'throw_logic_error' could be declared with attribute 'noreturn' [-Werror,-Wmissing-noreturn]
{
^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
gmake: *** [build/compile.mk:104: output/OSX64/dbg/src/Computer/TaskComputer.o] Error 1
gmake: 
unlink: output/host/tools/GenerateSineTables/../dirstamp: Not a directory

```

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2024-12-09:
Yes it is. It started to fail, after github did some updates on its osx vms.

yorickreum, 2024-12-09:
Can you point me to an action log showing this issue? The most recent one on master that I could find, https://github.com/XCSoar/XCSoar/actions/runs/12211912717/job/34069744689, looks fine to me.

yorickreum, 2024-12-09:
Maybe PR #1620 was the issue, would need to see a full log to be sure.

closed 2024-12-16 by lordfolken