XCSoar/XCSoar #1586: The application shuts down when launched

Edouardpara, 2024-10-06:
The application does not work on my tablet: Samsung Galaxy tabA9.. Android 14.

https://github.com/user-attachments/assets/853694ff-8890-4e31-9b35-beb4d0e94366

XCSoar versions having the problem
-------------------------------------------------

XCSoar versions not having the problem
-------------------------------------------------

System information
------------------

Steps to reproduce the behavior
-------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2024-10-06:
We put a list in the bugreport template to locate the error. 
This is probably the icon problem on ldpi screens. Try a testing release https://download.xcsoar.org/testing

Edouardpara, 2024-10-06:
Good morning. I just tried the test version, it works perfectly.Thank you very much for your feedback

Envoyé par AOL sur Android