XCSoar/XCSoar #1587: v7.44 Update translations

Joerg-CH, 2024-10-07:
With introduction of PR #1551 (Better thermal and flarm gauge), some translations are missing:
    Center top
    Center bottom
    Center top (avoid infoboxes)
    Center bottom (avoid infoboxes)
    Top left
    Top right
    Top left (avoid infoboxes)
    Top right (avoid infoboxes)

Could someone update the source file in weblate? I would update then the german translation.
In de.po is currently the "POT-Creation-Date: 2023-12-21".

lordfolken, 2024-10-08:
@felixhaedicke would you mind to add me on weblate as an admin? I would implement the webhook.

lordfolken, 2024-10-08:
I added the weblate user as a collaborator. According to: https://docs.weblate.org/en/latest/vcs.html

Sundown3867, 2024-10-08:
We dealt with this already in #459 - maybe try to contact the same person

lordfolken, 2024-10-08:
Yeah because we didnt integrate it then, the strings are 9 months out of date and no automation is configured. Id rather invest the time for integration and recieve automatic pull requests.

lordfolken, 2024-10-08:
@orangesunny could you add me please on weblate?

lordfolken, 2024-10-12:
So I wrote to weblate to ask for admin access. 

In the meantime there seems to be a merge conflict in the weblate git, and translations are currently disabled.

lordfolken, 2024-10-12:
@MaxKellermann could you add me as admin on weblate please?

Sundown3867, 2024-10-12:
Hallo Phillipp
I have contacted Weblate admin and got answer that you are a project owner
now

Best regards
MB

lordfolken, 2024-10-15:
Sady despite a confirmational email from weblate, now access as of yet. Asked them now for the 3rd time.

lordfolken, 2024-10-15:
@felixhaedicke could you please make me an admin on weblate? (or fix the integration...)

StanimirS, 2024-10-17:
Hello, I am using XCSoar with a Kobo, my issue is that once I change the language to something that isn't using the english alphabet(japanese, cyrilic, hebrew) I lose the labels of most buttons. Also if a map generated by the map generator doesn't have populated areas names in english characters it displays question marks. I don't know how related it is to this task, but it seems like it has a connection.

lordfolken, 2024-10-18:
Please open a new bug, with 
platform details, and version
Exact steps to reproduce the problem.

lordfolken, 2024-10-18:
Update: just got an invite. so its good now.

lordfolken, 2024-10-18:
Translations should now be auto updated.

closed 2024-10-18 by lordfolken