XCSoar/XCSoar #1599: Non english characters issue on Kobo

StanimirS, 2024-10-18:
XCSoar versions having the problem 
-------------------------------------------------
7.43

XCSoar versions not having the problem
-------------------------------------------------
unknown, this is the first time I have tried to use it

System information
------------------

Kobo Glo model number: N514

Steps to reproduce the behavior
-------------------------------
Install XCSoar. 
Go to menu -> config -> system -> look -> language, input -> language -> pick any language that has characters that are not english(Bulgarian, Russian, Japanese, Hebrew).

Close the menu and go back to the main screen with the map. 

Open the menu again. 

Most of the buttons are completely empty and don't have strings, making navigation impossible. 

Also maps that include names of populated areas that don't have english language characters do not display them, but instead are replaced with "????"(this is the case for both when the language is set to english and not english).

Expected behavior
-----------------
All the buttons have the correct language translations in place of the english strings (for both the buttons and the names of the populated areas on the map).

Actual behavior
---------------
Most of the buttons are completely empty and don't have strings, making navigation impossible. 
Also maps that include names of populated areas that don't have english language characters do not display them, but instead are replaced with "????".

Do you have any idea what may have caused this?
-----------------------------------------------
I believe this is because there are language packs missing or because the OS isn't using utf8. 

Do you have an idea how to solve the issue?
-------------------------------------------
We need to ensure other languages are supported on the OS by using utf8.

lordfolken, 2024-10-18:
@XCSoar/kobo-team anyone can have a look please?

fraktal-sb, 2024-10-18:
tested on Kobo GLO-HD with current head (7.44) and language "German": OK in map and menu
[image]

[image]

StanimirS, 2024-10-18:
[image]
[image]
[image]
[image]

This is what I see

fraktal-sb, 2024-10-19:
Can confirm language "Bulgarian" not working on my KOBO GLO-HD, but it is working on UNIX. (see screencopys)

I would recommend to factory reset the KOBO and initialize it in the bulgarian language and then again install XCSoar. Maybe this helps. If not we have to look about the process how we distribute languages on KOBO

[image]

[image]

lordfolken, 2024-10-19:
https://github.com/XCSoar/XCSoar/blob/b07f8818563741d7b4123657ee1ff1ae9d040ea2/build/kobo.mk#L70-L76

Can this font do cryllic?

fraktal-sb, 2024-10-19:
Answer from ChatGPT:

The Bitstream Vera font does not support Cyrillic. It is designed primarily for Latin characters. If you need Cyrillic characters, you should choose a font specifically designed for them, such as Arial, Times New Roman, or other fonts that support Cyrillic.

StanimirS, 2024-10-19:
> Can confirm language "Bulgarian" not working on my KOBO GLO-HD, but it is working on UNIX. (see screencopys)
> 
> I would recommend to factory reset the KOBO and initialize it in the bulgarian language and then again install XCSoar. Maybe this helps. If not we have to look about the process how we distribute languages on KOBO
> 
> ![2012-03-23-22-47-56-screen](https://private-user-images.githubusercontent.com/69430537/378059627-b27f2d90-477d-48ad-b5aa-6c78dbcf3015.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkzNzQ3MTIsIm5iZiI6MTcyOTM3NDQxMiwicGF0aCI6Ii82OTQzMDUzNy8zNzgwNTk2MjctYjI3ZjJkOTAtNDc3ZC00OGFkLWI1YWEtNmM3OGRiY2YzMDE1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDE5VDIxNDY1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThhOTBhMzQ5ZWE3Mzc5OGFiZDg5YWE3YjgxOTVjNTQ4NzU5YWE0NzJlZjM1OGUyNWVlODliYzMzZWNiMGRmMTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.UMde8xG4AXixb7Ihh4ktbNHnQ9LVpMhrQofjgFA914o)
> 
> ![Bildschirmfoto zu 2024-10-19 09-05-30](https://private-user-images.githubusercontent.com/69430537/378059636-0f4a3c0e-4e94-4399-9ff4-b6b539daf44d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkzNzQ3MTIsIm5iZiI6MTcyOTM3NDQxMiwicGF0aCI6Ii82OTQzMDUzNy8zNzgwNTk2MzYtMGY0YTNjMGUtNGU5NC00Mzk5LTlmZjQtYjZiNTM5ZGFmNDRkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDE5VDIxNDY1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI3ZGMwOGUzZTQzMmI0OTAzNTVmZGNkYzg4ODM3N2Q0NzY5NzkyMTEyMWUwODc4OTY2MWQ1ODNkYTAyMzQzNmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.qxr8naD2Y1fxV9ydZgHRNtaPKMVwbEebWnJ9axwhA9k)

Even if this works, I may decide to fly in some other location - turkey, japan, etc. Would I have to reset my Kobo every time?

fraktal-sb, 2024-10-20:
Reset of your KOBO is not necessary. As I can see now the problem lies in the used XCSoar KOBO- version font. This font is hardcoded in the make- procedure software. Reset does not help. Simply changing the font to an other with cyrillic capabilitys causes surely other problems. If there is someone who wants to make and maintain the change that is certainly OK. For the moment I only can recommend to use western european languages on KOBO.

StanimirS, 2024-10-20:
Is there a way to generate a map that has only western european languages? Since the map generator uses native language for the map locations.

fraktal-sb, 2024-10-20:
Answer from ChatGPT (no idea this is correct):

Yes, in XCSoar's map generator, you can customize the map to include only specific languages. To generate a map that features only Western European languages, you'll need to adjust the language settings in the map generation options.

Here's a general guide on how to do that:

    Open the XCSoar Map Generator: Start the map generation tool.
    Select Language Options: Look for an option to choose or filter languages. You may need to deselect languages that are not part of Western Europe.
    Choose Map Area: Specify the geographical area for the map if you want to focus on a particular region.
    Generate the Map: Once you’ve made your selections, proceed to generate the map.

If you have any trouble finding these options, the documentation for XCSoar may provide more specific guidance, or you could check forums and user communities for additional tips.

lordfolken, 2024-10-20:
Mapgen is another issue and another repo. Can we stick to the current problem?

lordfolken, 2024-10-20:
So i copied robot-regular/bold/bolditalic.ttf etc to /opt/xcsoar/share/fonts/ and this seems to fix the problem. But the font is smaller, but it looks a lot more .. . decent.

lordfolken, 2024-10-20:
@StanimirS can you review the attached PR and give me feedback if that solves the not displayed problem?

StanimirS, 2024-10-20:
> Mapgen is another issue and another repo. Can we stick to the current problem?

Sorry I was just trying to find a solution for the problem I have. I will try the new font tomorrow since I don't have access to the device right now (I don't actually own it). I will inform you of the results. I downloaded the build.

fraktal-sb, 2024-10-20:
looks good, even the German version looks better with this new font (more space for longer labels); tested with English and Bulgarian.

[image]

[image]

[image]

lordfolken, 2024-10-20:
> Answer from ChatGPT (no idea this is correct):
Nice example of why you need to have domain knowledge to be able to use an LLM positvily. The mentioned option is something the LLM dreamed up, it doesn´t  exist. :-)

StanimirS, 2024-10-22:
> > > Mapgen is another issue and another repo. Can we stick to the current problem?
> > 
> > 
> > Sorry I was just trying to find a solution for the problem I have. I will try the new font tomorrow since I don't have access to the device right now (I don't actually own it). I will inform you of the results. I downloaded the build.
> 
> Please, don't misunderstand. I think the idea has merit, but this being a bugtraker the topic has to stay on the topic. Feel free to make an entry under discussions.

I didn't misunderstand, and wasn't sarcastic or anything, sorry if it came out that way, I genuinely just want to help improve this project since I will be using it to fly myself, when I get a Kobo. 

I tested the new font, it works well. A bit on the small side compared to the huge airports, but it is usable now, which was the main issue.

[image]

lordfolken, 2024-10-22:
Yeah the font size is smaller then the old, but I only changed the font face ill check as to why. Text size 130% in the settings seems ideal.

closed 2024-11-24 by lordfolken