XCSoar/XCSoar #1603: autoupdate pot files for translations

lordfolken, 2024-10-24:
I found this: 
https://github.com/tarantool/doc/commit/1040a4f5c6061ea3dfa426a82746fcc81f67786e

This would autocommit the pot file once a merge from a PR is done. 

@Joerg-CH

lordfolken, 2024-10-24:
i did some tests in my branch https://github.com/lordfolken/XCSoar/actions/workflows/build-translation.yml. The problem i'm having is that it updates the timestamp even if there is no change in the pot or po file.

Joerg-CH, 2024-10-25:
I made some modifications in the workflow, but need more testing
https://github.com/Joerg-CH/XCSoar/blob/auto-translate/.github/workflows/build-translation.yml

lordfolken, 2024-10-25:
The sed you cold do sed 's/sed 's/\"POT-Creation-Date: .*//'  That way the line would be deleted to the end. '11d' is positional and could break when there is a change in the line order or format of the file.  Check the pattern with https://regexp101.com

Joerg-CH, 2024-10-25:
That's a good hint, thanks!
But what I am more concerned of is whether the 'if .. else .. fi' is working correctly when there is a difference between .pox1 and .pox2.

Joerg-CH, 2024-10-26:
I did some modifications, and now it seems to work:
- In case of no difference found, the old file (with old timestamp) stays in the po-folder. The workflow ended with success.
- In case of a differece, the xcsoar.pot is updated to the latest version, and a new timestamp set. 

But the process is then completed with an exit code 127, causing the workflow run fail.
(see https://github.com/Joerg-CH/XCSoar/actions/runs/11532607588/job/32104588888):
">Run if diff -q po/xcsoar.pot po/xcsoar.pox; then
Files po/xcsoar.pot and po/xcsoar.pox differ
Differences found, run normal update-po
/__w/_temp/0e6498d0-39ae-4fa9-acb2-953083b16ad2.sh: 8: -: not found
Error: Process completed with exit code 127."
Is there something like a reset necessary to start the second time 'make update-po' correctly?
Does anyone have an idea how to avoid this error?

lordfolken, 2024-10-27:
You need to remove the  yaml code in that run statement.
Its complaining about - not being found

but i think you just fixed it.

lordfolken, 2024-10-27:
sed: can't read po/xcsoar.pot2: No such file or directory is the current problem.

Joerg-CH, 2024-10-31:
I created a new workflow (build-translations.yml), which keeps the timestamps as old, when no differences between old and updated xcsoar.pot are found. When differences found (i.e. if there are new strings in the .cpp files), the xcsoar.pot and all <language>.po files in the folder po/ will be updated.
See https://github.com/Joerg-CH/XCSoar/pull/2/commits/c286f63edfa777ed1e4365e5fe6de7ed0d4b2ffd

closed 2025-02-07 by lordfolken