XCSoar/XCSoar #1605: IGC file GNSS altitude is orthometric/MSL altitude, but should be WGS84 ellipsoid altitude

nbonniere, 2024-10-29:
XcSoar version 7.42 Android.

According to the IGC flight recorder specification, the IGC GNSS altitude needs to be the WGS84 ellipsoid altitude.  XcSoar uses the orthometric altitude instead.

_CHAPTER 3
IGC FLIGHT DATA FILES - PRINCIPLES
Detail on the format of the IGC file is in Appendix A, this chapter contains the main principles
3.1 Earth Model. Lat/long and GNSS altitude figures must be with respect to the WGS 84 ellipsoid (see WGS84 in the
Glossary)._

Given an NMEA GGA sentence:

$GPGGA,152145.00,4537.06717,N,07438.94746,W,1,07,1.25,47.4,M,-33.4,M,,*55
The orthometric/MSL altitude is 47.4 meters, and the Geoid separation is -33.4 meters
Therefore the ellipsoid altitude is 47.4 + -33.4 = 14.0 meters

The IGC record from XcSoar (2024-10-05-XCS-0NB-01.igc) is:
B1608184537038N07438599WA-0030**00049**01512
the GNSS altitude is 49 meters  - this is incorrect, this value is the orthometric/MSL altitude, not the ellipsoid altitude.

https://en.wikipedia.org/wiki/Orthometric_height

h = H + N

ellipsoid height  h
Geoid height N
orthometric height H

Given the above GGA sentence, the ellipsoid altitude should be calculated as:

IGC_GNSS_altitude = 47.4 + -33.4 = 14.0

(NOT as IGC_GNSS_altitude = 47.4)

A review of a many files from PowerFlarm, and LXnav showing the use of the proper ellipsoid altitude confirm this problem.

The IGC record just before take-off for the PowerFlarm (4A5Gxxx1.IGC) is:
B1608254537036N07438610WA-0042**00019**01107
the GNSS altitude is 19 meters, this is the correct ellipsoid altitude.

The IGC record from the LXnav Nano (4A5Vxxx1.igc) is:
B1608394537027N07438780WA-0025**00014**00700109314263
the GNSS altitude is 14 meters, this is the correct ellipsoid altitude.

lordfolken, 2024-10-30:
So the b-record is formatted here:
https://github.com/XCSoar/XCSoar/blob/b07f8818563741d7b4123657ee1ff1ae9d040ea2/src/IGC/IGCWriter.cpp#L178-L196

The IGCFix is defined here:
https://github.com/XCSoar/XCSoar/blob/b07f8818563741d7b4123657ee1ff1ae9d040ea2/src/IGC/IGCFix.cpp#L9-L39

basic.gps_altitude for android sensors receives the geoid correction here:
https://github.com/XCSoar/XCSoar/blob/b07f8818563741d7b4123657ee1ff1ae9d040ea2/src/Device/AndroidSensors.cpp#L71-L76.

What brand of phone version os and what is your gps source?  Also please try a current testing version of XCSoar https://download.xcsoar.org/testing or at least 7.43.

nbonniere, 2024-10-30:
I use an LG G7 ThinQ cellphone, and have 3 sources, the phone's internal GNSS, and by bluetooth I connect to an LXnav Nano, and an LXnavigation ERA-57. In addtion, a PowerFlarm-Portable is connected by RS-232 serial to the ERA-57 for Flarm data. It all works very well.  Perhaps one of these 3 sources provides bad data, but I don't know which source was ultimately used for the generation of the IGC B records. I had checked the release notes for 7.43 and saw no mention of changes to GNSS altitude. I went through the commits since 7.42 to look for mention of GNSS altitude changes, but found none, so I entered a problem report. I can certainly try 7.43 and try each source one at a time to see what happens.

lordfolken, 2024-10-30:
Thank you for that information. 

Its in the end its relativly simple. gps_altitude is always stored geoid internally. In the IGCWriter/Parser XCSoar writes / reads those directly, without conversion to WGS84 as you described.  

These 2 bugs are 13 years old.

lordfolken, 2024-10-30:
The PR will fail, as i haven't updated the checks.

lordfolken, 2024-10-30:
Could you check that PR if you get the expected results?

nbonniere, 2024-10-31:
The PR seems to make 2 corrections. One to change the altitude at parsing, and one to change the altitude when writing the IGC B record. The second one looks good, but the first is not required. Using the 'MSL' GNSS altitude is best. You certainly don't want to make the correction twice.
I've encountered a GNSS module that reported a zero Geoid all the time in its GGA NMEA sentence. In the case of a zero Geoid, a Geoid look-up (EGM96) is needed to correct the NMEA sentence, and calculate the proper 'MSL' altitude.

moshe-braner, 2024-11-05:
When the GNSS module does report the geoid height in the GGA sentence, as most of them do, then it may be more accurate and efficient to use that rather than to look up the EGM96 approximation, as is done in the proposed commit.  Who knows how the GNSS module gets that number, maybe its own lookup table.  But what is certain is that adding the MSL altitude and the geoid height from the GGA sentence recreates the exact WGS84 ellipsoid altitude which was computed inside the module from the signals from the GNSS satellites.  And that is what should be output to the IGC file.

moshe-braner, 2024-11-08:
Proposed a better fix: https://github.com/XCSoar/XCSoar/pull/1610

moshe-braner, 2024-11-09:
reorganized commits and comments into https://github.com/XCSoar/XCSoar/pull/1611

moshe-braner, 2024-11-13:
Minor other bug: on an Android phone, after using an external FLARM device for GNSS input, then switching to the 'internal sensor' device and disabling the external source the GPS source indication in the IGC file still says "FLARM".  And conversely in an Android phone where the internal device was used, then disabled and an external device was sending data via UDP, the following line appeared in the IGC file header:
    HFGPS:Internal GPS (Android)

lyusupov, 2024-12-12:
> [!TIP]
> XCSoar is NOT listed as IGC certified flight recorder.

Therefore, Rule A3.2.7.5 ( H-records ) is applicable for this soaring software.
https://xp-soaring.github.io/igc_file_format/igc_fr_specification_with_al8_2023-2-1_0.pdf

[image]

<br>
<br>

When (if) XCSoar writes the geoid altitude into the B-record of IGC file- you are safe to add an appropriate Header ( H-record ) and that's it.
If is responsibility of scoring software such as **SeeYou**, to parse the H-record first then identify the altitude value in every B-record and process correspondingly.

closed 2026-09-14 by lordfolken

lordfolken, 2026-09-19:
Follow-ups from leftovers on this thread (now under #2391):

- #3179 — missing/zero GGA geoid uses EGM96 instead of the receiver value
- #3180 — `HFGPS` names Device A, not the GPS that filled the log