XCSoar/XCSoar #1606: TaskAbort toggle

XCNav, 2024-10-30:
XCSoar versions having the problem
-------------------------------------------------

7.42
XCSoar versions not having the problem
-------------------------------------------------

7.39

System information
------------------

Any Android device i tested Android 11 and later

Steps to reproduce the behavior
-------------------------------

Starting 7.42 TaskAbort toggle on Android stopped working correctly. Using XCRemote or a Keyboard, the TASKABORT toggle function is set to "T" when the GLB_XCI_XCREMOTE file is downloaded and set in the INPUT settings.
.

Expected behavior
-----------------

Normal behaviour would be to abort the task and set the nearest Alternate, indicating up to ten alternates when "T" is pressed once. On the second press, xcs should continue and return to the set task

Actual behavior
---------------

Present behaviour is that XCS only detects the first press and disregards the second press and sticks in the alternate menu. In the windows version everything is working as supposed

Do you have any idea what may have caused this?
-----------------------------------------------

NO

Do you have an idea how to solve the issue?
-------------------------------------------

OPENING AN ISSUE 😉

lordfolken, 2024-10-31:
Can you list the xci entry that doesent work anymore?

XCNav, 2024-10-31:
> Can you list the xci entry that doesent work anymore?

mode=default
type=key
data=T
event=AbortTask toggle

XCNav, 2024-11-07:
In open soar the toggle function is still working as supposed

lordfolken, 2024-11-07:
It seems to me that letters are no longer recognized. 

This works:

mode=default
type=key
data=9
event=AbortTask toggle

XCNav, 2024-11-09:
That is not true, as initially T is recognized and aborts the task.
"9" is set by default to  show the target info of a set task. Recheck against the windows version where everything seems to work as it is supposed to. Also opensoar does not show this bug

lordfolken, 2025-02-25:
So... this took hours... looking at every aspect from file parsing, to keyscan code conversions, to event mapping. 
I finally found the issue at least for the linux case. 
Basically the letters don't get uppercased before they are compared to the already uppercased event table. 
https://github.com/lordfolken/XCSoar/pull/new/issue-1606-lettereventkeys

Now this will not be merged like this as this is a work around.

XCNav, 2025-02-25:
Wow, thanks for looking for the needle in the hey! Great work.

lordfolken, 2025-03-31:
7766304af23b04a0af0b1463db9dface1b7a7e92 is my current bet.

XCNav, 2026-01-09:
@lordfolken Problem still exists in IOS and Android. I tried to change to:
mode=default
type=key
data=9
event=AbortTask toggle
 with same behaviour.
You can abort the task, but it does not resume, no matter if you try with a letter or a number.
even when you abort via NAV2 "Abort Task" you can not resume the task via  NAV2 "Task Resume". In IOS it states "TASK RESUMED" but it does not set actual waypoint active again.