XCSoar/XCSoar #1612: Condor 3 device requested to correct wind being out by 180 degrees.

JBlyth, 2024-11-19:
I have been a beta over past months for Condor 3. The development team has discovered an error in the NMEA stream for wind in earlier Condor versions, where the wind data is out by 180 degrees. With XCSoar, you have corrected this error in the Condor device by using the reciprocal of the wind calculation to correct that error in the Condor NMEA output string.

Line 58 from https://github.com/XCSoar/XCSoar/blob/master/src/Device/Driver/Condor.cpp reads:-
info.ProvideExternalWind(wind.Reciprocal());

In Condor 3, this error has been corrected in the interests of accuracy, so it would be very helpful if XCSoar could be updated to resolve this change for Condor 3 users.

It may be easiest to just add a Condor 3 device to the list with that line changed/or removed to stop the reciprocal being used.

That would maintain compatibility with Condor 2 while giving Condor 3 users the correct wind data.

lordfolken, 2024-11-19:
Does anything else change with the protocol? Now that 3.01 is out is there a change? Also is there documentation mentioning the update?

The lxnav driver should allready interpret the wind correctly.

JBlyth, 2024-11-20:
Thanks for the reply. There are no other changes in the protocol. Just the "correction" to the wind direction which was an identified bug in Condor 2.

I understand that the LXNav driver may well work correctly, but it would be very useful to have a Condor 3 device to avoid confusion. (I don't use XCSoar personally, but I was asked to make the request.)

closed 2024-11-24 by lordfolken