XCSoar/XCSoar #1621: Receive airspace warnings for class E (OpenAIP), even though the warning is disabled

mlankenau, 2024-12-10:
XCSoar versions having the problem
-------------------------------------------------
7.43

XCSoar versions not having the problem
-------------------------------------------------
dont know

System information
------------------
Stefly Nav 7"

Steps to reproduce the behavior
-------------------------------
1. Install OpenAIP Airspace for Germany
2. Start low on some airport, run the first minutes of a IGC File, takeof and get into the Class E aboce
3. Receive Warning about class E, even though the Warning for Class E is disabled

Expected behavior
-----------------
I dont want to reveive the class E warning, it is disabled.

Actual behavior
---------------
I receive the Class E warning

Do you have any idea what may have caused this?
-----------------------------------------------
No

Do you have an idea how to solve the issue?
-------------------------------------------
No

mlankenau, 2024-12-10:
I think I found a solution.
For example: Class E, Type Unclassified.
I removed the warning for Unclassified. Not sure what Type Unclassified means.

lordfolken, 2024-12-10:
[image]

So this file was downloaded manually, not via File Manager in XCSoar? Because the openaip files I provide in XCSoar don't have this problem.

fraktal-sb, 2025-05-22:
is this now also solved with https://github.com/XCSoar/XCSoar/pull/1751 ?

lordfolken, 2026-01-08:
Agreed. Thanks