XCSoar/XCSoar #1625: Cant build for iOS

LeNinjaHD, 2024-12-11:
XCSoar versions having the problem
-------------------------------------------------
latest commit

XCSoar versions not having the problem
-------------------------------------------------
none

System information
------------------
M2 Pro Max MacBook (macOS Sequioa)

Steps to reproduce the behavior
-------------------------------
Download Repo, install dependencies, build for iOS

Expected behavior
-----------------
It should build normally

Actual behavior
---------------
Reports:
`/Users/<user>/Documents/XCSoar/output/src/fmt-10.2.1/include/fmt/os.h:11:10: fatal error: 'cerrno' file not found `
after some time.

Do you have any idea what may have caused this?
-----------------------------------------------
Maybe my system setup

Do you have an idea how to solve the issue?
-------------------------------------------
Reinstall C maybe?

yorickreum, 2024-12-12:
I do feel like I need more information to debug this. 
Are there warnings thrown prior to this error?
With which parameters are building?
Which version of CLANG do you use?

LeNinjaHD, 2024-12-14:
> I do feel like I need more information to debug this. Are there warnings thrown prior to this error? With which parameters are building? Which version of CLANG do you use?

Apple clang version 16.0.0 (clang-1600.0.26.4)
Target: arm64-apple-darwin24.0.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin

I am building with "make TARGET=IOS64 ipa"

Full log is here: https://pastebin.com/sDRAp5ZG

Another thing which i wanted to ask is on how to get the executable on your phone or in the iPhone Simulator. I dont really need to build it myself if i can figure out how to use this: https://download.xcsoar.org/testing/IOS64/

yorickreum, 2024-12-14:
Can you try ```gmake -j$(nproc) TARGET=IOS64  V=2 OPTIMIZE="-O0" ipa```?
```make``` often referes to an outdated version of GNU Make on macOS.

yorickreum, 2024-12-14:
To install on your phone, you need to pack and sign the executable into an ipa. Then you can just drag & drop it on your iPhone in Finder, when it's connected. Signing you can do e.g. with https://www.iosappsigner.com/.

To install in the simulator, you'd probably need to build it with the simulator SDK.

Keep in mind that XCSoar is currently not functional on iOS, it has some rendereing bugs, etc. that need to be addressed.

yorickreum, 2025-01-27:
> Keep in mind that XCSoar is currently not functional on iOS, it has some rendereing bugs, etc. that need to be addressed.

With the new pull requests, there now is a mostly functional version of XCSoar for iOS. If you want, I can provide you with access to the app via Apple TestFlight, @LeNinjaHD.

LeNinjaHD, 2025-01-27:
> With the new pull requests, there now is a mostly functional version of XCSoar for iOS. 🥳
>
> If you want, I can provide you with access to the app via Apple TestFlight, @LeNinjaHD.

Yes please!!

yorickreum, 2025-01-27:
> Yes please!!

Please let me know your Apple ID (email address you are registered with at Apple). Feel free to contact me directly via xcsoar@yorickreum.de.

Anyways, are you still interested in building yourself? Then please provide more information so we can debug this. Otherwise, imo we can close this issue.

closed 2025-01-30 by lordfolken