XCSoar/XCSoar #1626: Integrate PureTrack.io data

XCNav, 2024-12-13:
XCSoar version
--------------

7.43

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

Integrate data and user portal for puretrack and show PT info on Map and Flarmscreen

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

Most people using XCSoar on a mobile device. Puretrack is using an online connection to show realtime traffic and as well climb information merging data from different platforms. That would increase safety during flight, as you are now able to seeADSB-B and non flarm equipped traffic such as paragliders etc. On some PG contests it became mandatory to run PT.
I suggest a log-in portal for registered users as well as a set-up menu. I will forward this issue to Tim Bromhead the main developer and head behind PureTrack. Here you can find the API
https://puretrack.io/help/api

lordfolken, 2024-12-13:
This requires a paid subscription. This is comodifying data that is provided free of charge by a lot of volunteers running these data sources in their spare time.

XCNav, 2024-12-15:
@lordfolken You do not pay for the data. As you said you can get it partialy elsewhere for free. You pay for the work, engineering background, enthusiasm and hard work of someone who made it possible to gather all this different sources in one app, and to maintain this service in the future! For my opinion that makes flying safer. Everybody can decide of his own, if he would like to pay for that service or not. XCSoar could have the ability to support flight safety. It is not without reason that this app is mandatory in some PG contests already! By the way, there is a free subscription available as well.

lordfolken, 2024-12-16:
So for me things speaking against this:
 * its an intermediary with a business model so by definition unfree. 
 * its a http rest api, that requires stable tcp connections. This very sub optimal in a cockpit. 

The basis of this service is a fragmented tracking eco system. A fragmented tracking eco system is in no one's interest. I don't think it makes sense to support more tracking options than Flarm/ADSB/OGN. 

The PG Community has created FANET for their need to signal different states.  Support for that I would also welcome, but has to be initiated by the PG community as this is not something I have devices for, nor does it make sense to implement something I don't regularly use.

I think i would favor #1584 over this. That is direct integration of OGN data, and feed it via cloud.xcsoar.org via our at this point 15 years old udp tracking/traffic/thermal protocol.

lordfolken, 2024-12-16:
At least tracking should already be possible now, as the integrated skylines tracking option is stated as a source in https://puretrack.io/help/api

I don't really know how this is done... other than scraping the https://skylines.aero/tracking page. 

And I wasn't contacted even though I run https://skylines.aero at my own cost and somehow try to keep it breathing.

XCNav, 2024-12-17:
> I am sure if the people form PureTrack made a proper PR to get their data in XCSoar no one would reject it (if it is up to standards).
> 
Tim will join soon. We already discussed the possibility to show the free subscription data via API as well

> XCSoar is not an anti collision service nor should it be. ..... So please provide some reference/study that agrees with your statement that this improves safety when making these claims.
I totally agree. I am not talking about short term traffic avoidance, more about helping to build the big picture, what's going on around me, eg can I expect PG traffic around the next corner etc., and plan ahead.
As an airline pilot it is my daily business framing my situation by any information (atc, Tcas etc.) i can get.

ipearx, 2024-12-18:
Hi Tim here from puretrack.io, thanks @XCNav for your enthusiasm to integrate PureTrack into XCSoar.
I'd be keen to help make it happen, I've had a number of people ask for it.

For sending location from XCSoar to PureTrack, @lordfolken  yes we're just using the skylines homepage data at the moment. It's not great, as we only fetch every minute or so and doesn't include type or course info. We've got 179 users configured to use that so far. I assumed this was fine under the Open Data Commons Attribution License (ODBY) mentioned on the homepage? Let me know if not. It would be great to get proper support for sending location to PureTrack built into XCSoar. Unfortunately my C skills aren't up there, so haven't had a chance to try and contribute myself yet.

For showing location data, yes the model at the moment is for a Pro account to use the PureTrack API data. This is the same model as SafeSky, they charge for sending data to external devices/api use. Of course I'd love to provide full open data for all apps, but just can't afford it yet, I'd need to beef up the infrastructure to handle that. All data from all sources is free to view on the PureTrack website however.
I'd certainly consider building a free API that gave limited data for XCSoar to use, e.g. aircraft within a certain radius of the user. If there was interest for doing that?

As for why it is useful to show PureTrack data:
- Yes it's not a replacement for FLARM or collision alerting. But it helps immensely with situational awareness. It's worth considering the design to reflect this. e.g. hiding old data promptly, or indicating it's old somehow. PureTrack starts fading out the icons when >5mins old for example.
- It shows a LOT of data sources on top of OGN (30+ so far) including ADSB, SPOT, InReach, cellular trackers, apps etc. OGN is great in Europe, but many countries don't have good OGN coverage set up (and never will due to the geography). e.g. Australia, New Zealand, Brazil, USA. In NZ we use SPOT/InReach all over the South Island, and cellular trackers all over the North Island.

As for sending to PureTrack:
- PureTrack can be used as an OGN gateway, for free. So getting data from XCSoar into PureTrack would mean any user can also send their location to OGN, which then shares to any other tracking system e.g. SafeSky, OGN websites etc.
- In France for Paragliders PureTrack can be used to send data to the FFVL tracking system, which their search/rescue uses.

Hope that helps, happy to answer questions or discuss ideas

lordfolken, 2024-12-23:
(i'll move my misgony towards fragmentation of the tracking eco systems of flying things to somewhere else. ) 
(I have some reservations as to steer users of Free (libre) Software towards non-free services. We however do already support weglide, lifetrack24, which are per definition non-free) 

#skylines you are correct that the site allows resharing by the licence type. Even commercialization. What I would discuss there is a better integration than scraping a website. Unfortunately I think the projects is dead, unless @Turbo87 disagrees?

#puretrack we have lifetrack24 https://github.com/XCSoar/XCSoar/tree/master/src/Tracking/LiveTrack24 which is at least http based, so an implementation could look similar, at least for the sending part. 

For receiving traffic we would have to look at how to integrate the data. A tcp rest api doesn't seem very efficient for the purpose, which is why XCSoar's devs invented a udp based traffic and thermal exchange protocol around 15 years ago.. 

Is the OGN Gateway you mention 2 way?

Puretrack seems to require an API key per application. This api key would be publicly visible in the source code.

Turbo87, 2024-12-24:
> skylines: What I would discuss there is a better integration than scraping a website. 

I would assume that PureTrack uses the SkyLines API, same as the SkyLines frontend. Since SkyLines does not use server-side rendering scraping the website would be pretty complicated...

> skylines: Unfortunately I think the projects is dead, unless @Turbo87 disagrees?

no, I think that's a fair statement. with WeGlide so strong there is little motivation on my side to still put much effort into SkyLines.

ipearx, 2024-12-28:
> > skylines: What I would discuss there is a better integration than scraping a website.
> 
> I would assume that PureTrack uses the SkyLines API, same as the SkyLines frontend. Since SkyLines does not use server-side rendering scraping the website would be pretty complicated...

After checking I don't scrape the website, I just use the one API endpoint https://skylines.aero/api/live every minute, which is almost perfect. It gets all current locations of current pilots in a single call. It's just missing:
- Bearing/heading
- Type of point i.e. Glider/Paraglider/Plane etc

> > skylines: Unfortunately I think the projects is dead, unless @Turbo87 disagrees?
> 
> no, I think that's a fair statement. with WeGlide so strong there is little motivation on my side to still put much effort into SkyLines.

A number of Paraglider pilots use it however, while WeGlide only supports gliding... (I asked, it was  a deliberate decision, so unlikely to be added in the near future)

ipearx, 2024-12-28:
> (I have some reservations as to steer users of Free (libre) Software towards non-free services. We however do already support weglide, lifetrack24, which are per definition non-free)

Keep in mind WeGlide and PureTrack do provide a LOT of value for free. They just also have paid upgrade plans as well. My 'philosophy' as it were is: Features needed for search and rescue/safety are free (e.g. all basic tracking and finding people), nice to haves (e.g. Climb markers, fancier maps) are paid.

> #puretrack we have lifetrack24 https://github.com/XCSoar/XCSoar/tree/master/src/Tracking/LiveTrack24 which is at least http based, so an implementation could look similar, at least for the sending part.

Yes I was certainly considering making an endpoint in PureTrack compatible with LiveTrack24 if that would make it easy?

> > For receiving traffic we would have to look at how to integrate the data. A tcp rest api doesn't seem very efficient for the purpose, which is why XCSoar's devs invented a udp based traffic and thermal exchange protocol around 15 years ago..

TCP is not ideal but as I've only got web apps, I haven't had a need to build a UDP endpoint. Happy to look into it.

> Is the OGN Gateway you mention 2 way?

Yes we take OGN data IN, and OUT if that's what you mean? Can read more at:
https://puretrack.io/news/new-send-puretrack-data-to-ogn-open-glider-network

> Puretrack seems to require an API key per application. This api key would be publicly visible in the source code.

That's fine, we can just make a key like 'xcsoar'.

lordfolken, 2026-03-09:
PureTrack integration PR is now open: https://github.com/XCSoar/XCSoar/pull/2294\n\nStatus: this is currently **untested against a live PureTrack backend**, but it should work based on the Insert API documentation and XCSoar's existing tracking patterns.\n\nOpen questions still to settle:\n- tracking interval default\n- canonical server address/default endpoint\n- app key/device ID provisioning and expected validation constraints

lordfolken, 2026-09-10:
Related architecture: #3086.

PureTrack belongs with the other live feeds, not with File Manager / repository `type=`.

- Traffic: add a source tag on the existing `TrafficList` (same pattern as FLARM / FLARMNet / Cloud), not a parallel list the map and FLARM screen both walk.
- Thermals (if PureTrack ever sends them): the single thermal list in #3093, tagged the same way — not another loop in `MapWindowThermal.cpp`.

This issue stays the PureTrack integration; those two are the lists to write into.