XCSoar/XCSoar #1627: Airspace not working

glider-user, 2024-12-15:
XCSoar versions having the problem
-------------------------------------------------
The actual Namiban airspace V1.20 generates an error, despite working on other devices like LX:
Error in airspace at line 306: Airspace has no name. 

error shows up at version 7.42 and 7.43

airspace from:
https://ssn.org.na/wordpress/airspace-2024-25/

XCSoar versions not having the problem
-------------------------------------------------
unknown

System information
------------------

Samsung S10 and Samsung S22 Android 14

Steps to reproduce the behavior
-------------------------------

Load Namibian Airspace V1.20

Expected behavior
-----------------

Actual behavior
---------------
Error in airspace at line 306: Airspace has no name. 

Do you have any idea what may have caused this?
-----------------------------------------------
no

Do you have an idea how to solve the issue?
-------------------------------------------
no

lordfolken, 2024-12-15:
The Airspace from the file:

```
AC TMZ
AN Windhoek TMZ
AL GND
AH FL145
V X=22:28:39 S 017:28:14 E
DC 50.0
V X=22:28:39 S 017:28:14 E
DC 60.0
```

DC Is the radius of a circle at the center point X=22:28:39 S 017:28:14 E. 
The radius is specified twice. XCSoar doesn't expect that and says that the name is missing. 

So clearly this file contains errors, that might lead to an airspace violation if not fixed. Therefore XCSoar doesn't parse it and barks at the file.

lordfolken, 2024-12-15:
Same error at line 331. 
Please talk to the supplier of your file and address the issue.

lordfolken, 2024-12-15:
Config->Config->File Manager offers the latest Nambia Airspace file from OpenAIP, this one has the named airspaces and parses without fault. NA-ASP-National-OpenAip.txt Updated on a daily basis.

lordfolken, 2024-12-16:
Nex

closed 2024-12-16 by lordfolken