XCSoar/XCSoar #1641: Small Flarm radar still displayed when targets outside 5km radius

fraktal-sb, 2025-01-08:
XCSoar versions having the problem
-------------------------------------------------

7.44
XCSoar versions not having the problem
-------------------------------------------------

not tested

System information
------------------

UNIX

Steps to reproduce the behavior
-------------------------------

replay/ use the attached (following) NMEA Log file

[Test2.log](https://github.com/user-attachments/files/18346999/Test2.log)

see also the following screenshots

IMG_1 shows the situation when a new Flarm- object appears in the 5 km range (local time 14:48:07)
IMG_2 shows the situation at time 14:51:51, the Flarm Object is "far away"
IMG_3 shows the situation at 14:52:11 at the big radar display

[image]
[image]
[image]

Expected behavior
-----------------

when no more Flarm- objects within the 5km range the small radar should disappear. The desired behavior is given when switching back from the big radar page to the map page

Actual behavior
---------------

the small Flarm- radar within the map is still displayed alltough all flarm objects are outside the 5 km range

Do you have any idea what may have caused this?
-----------------------------------------------

no

Do you have an idea how to solve the issue?
-------------------------------------------

possible solution: within the XCSoar/src/Device/Driver/FLARM/StaticParser.cpp should also be a condition to filter traffic outside the near 5km range.

use the attached

ubx, 2025-01-14:
I don't think this is a bug; rather, it is the expected behavior. 

First, the small FLARM radar has a maximum distance of 2000 m, as noted in the [XCSoar User Manual](https://download.xcsoar.org/releases/7.43/XCSoar-manual.pdf) on page 111.  
The manual states that "Traffic further away than 2000 m is drawn at the 2000 m ring," and this is the case at 14:51:51.  

At 14:51:51, you can see three objects:  
- **100** (9990 m above),  
- **34** (3357 m above), and  
- **4** (441 m below).  

The first two are of type *'aircraft with jet/turboprop engines'*, i.e., passive traffic. The last one is a *'paraglider (soft)'*, which is considered active traffic. You can refer to the relevant code [here](https://github.com/XCSoar/XCSoar/blob/273cc7f332972732a379c0aa4430d0e4f58eff27/src/FLARM/Traffic.hpp#L146C1-L154C4).  

By the way, the `Test2.log` file you provided contains many erroneous `$PFLAA` sentences. For instance:  
- At line 269: missing values.  
- The object with ID `39103C!FJLKN` has an inconsistent `unique_id`. At one point, it is labeled as an 'aircraft with reciprocating engines' and at another as a 'paraglider (soft),'.
Just to name a few.

fraktal-sb, 2025-01-14:
but what is the point if I change back from Flarm radar back to the map page? 

see some seconds later at Time 14:52:21.

The small flarm radar is no more displayed.

[image]

ubx, 2025-01-14:
> but what is the point if I change back from Flarm radar back to the map page?

I thougt the point was "Small Flarm radar still displayed when targets outside 5km radius"

fraktal-sb, 2025-01-15:
No Problem; I will further analyse the program code and make some tests. My opinion is that the small radar should disappear when there is nothing to display.

lordfolken, 2025-01-15:
The way this is supposed to work:
  * All traffic is displayed on the Map
  * All traffic is displayed on the BigTrafficGauge
  * If there is traffic below 4km then the Flarm Radar is shown

What seems to be happening when replaying the the log:
 * the Gauge appears when there is traffic below 4km
 * the Gague does **_not_** disappear when the traffic is now out of the 4km range
 * the Gauge disappears when you enter the BigTrafficGauge and then go back to the map when the traffic is out of 4km range. 

The function for determining traffic in close range:
https://github.com/XCSoar/XCSoar/blob/273cc7f332972732a379c0aa4430d0e4f58eff27/src/FLARM/List.cpp#L25-L29

The logic for showing the traffic gauge:
https://github.com/XCSoar/XCSoar/blob/273cc7f332972732a379c0aa4430d0e4f58eff27/src/MainWindow.cpp#L1163-L1182

With the return statement the traffic gauge is never hidden. 

If someone cares to make an improvment. I'd be happy to add it. Please also update the documentation to 4000m.

fraktal-sb, 2025-01-18:
To explain the errors in the test file:
1) Aircraft type 8 and 7: Different setup of transponder and Flarm. Transponder setup B4 (Ultralight) is interpreted as type 7 Flarm (Paraglider). Setup in Flarm=8 aircraft with reciprocating engine(s). Categories are not fully compatible.
2) No data in line 269: RelativeEast is missing. The reason: Bearingless target for which no exact position is known. In this case, relativeNorth gives the estimated distance.

PR #1648 (2025-01-20): Make Flarm gauge disappear when traffic is farther than 4 km away
Manual updated for en, de and fr. Others pending.

Closes #1641
closed 2025-01-20 by lordfolken