XCSoar/XCSoar #1643: Different Flarm target colours in map and radar display

fraktal-sb, 2025-01-09:
XCSoar versions having the problem
-------------------------------------------------

7.44

XCSoar versions not having the problem
-------------------------------------------------

not tested

System information
------------------

UNIX

Steps to reproduce the behavior
-------------------------------

use the following NMEA log file and switch to the radar and map page

[Test2.log](https://github.com/user-attachments/files/18347165/Test2.log)

Expected behavior
-----------------

Within map and radar page we should display targets with the same colour

Actual behavior
---------------

Flarm target colour is allways "blue" in the radar page.

see the following screenshots

[image]
[image]

Do you have any idea what may have caused this?
-----------------------------------------------

no

Do you have an idea how to solve the issue?
-------------------------------------------

use the same logic for colouring the Flarm targets within map and radar page. Because the Flarm ID text is allways related to the selected target this text should use the same logic for colouring.

fraktal-sb, 2025-01-09:
Maybe this is more a proposal then a bug report. When switching to the radar view the same plane as in the map view is displayed. Therefore (my opinion or proposal) the colour should be the same and all other planes should appear in neutral colour. Selecting an other plane should show it in the colour according distance/warning logic within the map view (same colours as in the details view). But I can also live with the current solution. We can leave this issue open for a while and when nobody is interested close it.

fraktal-sb, 2025-01-12:
To make more clear what the requirement should be I analysed the program code.

In ./XCSoar/src/Look/TrafficLook.hpp we find the colours used to display targets in the map window.

We find there

static constexpr Color safe_above_color{0x1d,0x9b,0xc5};
static constexpr Color safe_below_color{0x1d,0xc5,0x10};
static constexpr Color warning_color{0xfe,0x84,0x38};
static constexpr Color warning_in_altitude_range_color{0xff,0x00,0xff};
static constexpr Color alarm_color{0xfb,0x35,0x2f};

Within ./XCSoar/src/Renderer/TrafficRenderer.cpp

they are used in canvas.DrawPolygon to draw the traffic in the map window.

within ./XCSoar/src/Gauge/FlarmTrafficWindow.cpp traffic is rendered in

FlarmTrafficWindow::PaintRadarTarget

Here the only used colour codes are warning and alarm (warning_brush, alarm_brush)

See section // Set the arrow color depending on alarm level

The requirement would be to render the traffic in FlarmTrafficWindow the same way as in ./XCSoar/src/Renderer/TrafficRenderer.cpp

The selected Flarm target could be rendered with an additional thicker (black) border around the triangle polygon.

This enhancement would make the map and radar display more consistent. Targets in the radar page look the same way as in the map page. The colour information "above, below, in_altitude_range" enhances the overview within the radar page

lordfolken, 2025-01-13:
Are you willing to implement?

Akso have a look at the #1295 PR.

fraktal-sb, 2025-01-13:
At the moment I am not familiar with the environment and procedure for software changes with github. But I can prepare the implementation.

lordfolken, 2025-01-13:
You can fork this repo. Commit your changes and create a pull request to the main repo. 

https://xcsoar.readthedocs.io/en/latest/policy.html#github more info here.

lordfolken, 2025-01-30:
I in general like it. Consistency is key here.