XCSoar/XCSoar #1649: undeclared identifier 'THERMALRECENCY'

yorickreum, 2025-01-19:
XCSoar versions having the problem
-------------------------------------------------

latest from git https://github.com/XCSoar/XCSoar/commit/273cc7f332972732a379c0aa4430d0e4f58eff27

XCSoar versions not having the problem
-------------------------------------------------

not aware of one

System information
------------------

Compiling on macOS ARM for iOS64.

Steps to reproduce the behavior
-------------------------------

Compile with `gmake -j$(nproc) TARGET=IOS64 DEBUG=1 USE_CCACHE=y V=2 OPTIMIZE="-O0" ipa` and compile again, and again ... the compile error sometimes shows up.

Expected behavior
-----------------

Compiling should succeed in every run, given I do not change the code in between.

Actual behavior
---------------

Compiling sometimes fails with 
```console
ccache clang++ src/Computer/ThermalRecency.cpp -c -o output/IOS64/dbg/src/Computer/ThermalRecency.o -MD -MF output/IOS64/dbg/src/Computer/ThermalRecency.d -MT output/IOS64/dbg/src/Computer/ThermalRecency.o -O0 -g   -std=c++20 -fno-threadsafe-statics -fmerge-all-constants -fcoroutines-ts -fvisibility=hidden -ObjC++ -fobjc-arc -stdlib=libc++ -Wall -Wextra -Wwrite-strings -Wcast-qual -Wpointer-arith -Wsign-compare -Wundef -Wmissing-declarations -Wredundant-decls -Wmissing-noreturn -Wvla -Wno-missing-field-initializers -Wcast-align -Werror -Wno-error=deprecated-declarations   -I./src/unix -I./output/include -isystem ./output/src/boost_1_85_0 -DBOOST_NO_IOSTREAM -DBOOST_MATH_NO_LEXICAL_CAST -DBOOST_UBLAS_NO_STD_CERR -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_SYSTEM_NO_DEPRECATED -DBOOST_NO_STD_LOCALE -DBOOST_LEXICAL_CAST_ASSUME_C_LOCALE -DBOOST_NO_CXX98_FUNCTION_BASE -I./src -I./src/Engine -I./output/IOS64/include -DHAVE_POSIX -DHAVE_VASPRINTF -isystem ./output/IOS64/lib/aarch64-apple-darwin/include -DEYE_CANDY -DXCSOAR_TESTING -isystem /include        -isystem /Users/yorickreum/Coding/XCSoar/output/IOS64/lib/aarch64-apple-darwin/include   -miphoneos-version-min=10.0 -arch arm64 -target aarch64-apple-darwin -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS.sdk     
ccache clang++ src/VALI-XCS.cpp -c -o output/IOS64/dbg/src/VALI-XCS.o -MD -MF output/IOS64/dbg/src/VALI-XCS.d -MT output/IOS64/dbg/src/VALI-XCS.o -O0 -g   -std=c++20 -fno-threadsafe-statics -fmerge-all-constants -fcoroutines-ts -fvisibility=hidden -ObjC++ -fobjc-arc -stdlib=libc++ -Wall -Wextra -Wwrite-strings -Wcast-qual -Wpointer-arith -Wsign-compare -Wundef -Wmissing-declarations -Wredundant-decls -Wmissing-noreturn -Wvla -Wno-missing-field-initializers -Wcast-align -Werror -Wno-error=deprecated-declarations  -I./src/unix -I./output/include -isystem ./output/src/boost_1_85_0 -DBOOST_NO_IOSTREAM -DBOOST_MATH_NO_LEXICAL_CAST -DBOOST_UBLAS_NO_STD_CERR -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_SYSTEM_NO_DEPRECATED -DBOOST_NO_STD_LOCALE -DBOOST_LEXICAL_CAST_ASSUME_C_LOCALE -DBOOST_NO_CXX98_FUNCTION_BASE -I./src -I./src/Engine -I./output/IOS64/include -DHAVE_POSIX -DHAVE_VASPRINTF -isystem ./output/IOS64/lib/aarch64-apple-darwin/include -DEYE_CANDY -DXCSOAR_TESTING -isystem /include   -isystem /Users/yorickreum/Coding/XCSoar/output/IOS64/lib/aarch64-apple-darwin/include -DZLIB_CONST -isystem /Users/yorickreum/Coding/XCSoar/output/IOS64/lib/aarch64-apple-darwin/include   -miphoneos-version-min=10.0 -arch arm64 -target aarch64-apple-darwin -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS.sdk     
ccache clang++ src/Math/FastTrig.cpp -c -o output/IOS64/dbg/src/Math/FastTrig.o -MD -MF output/IOS64/dbg/src/Math/FastTrig.d -MT output/IOS64/dbg/src/Math/FastTrig.o -O0 -g   -std=c++20 -fno-threadsafe-statics -fmerge-all-constants -fcoroutines-ts -fvisibility=hidden -ObjC++ -fobjc-arc -stdlib=libc++ -Wall -Wextra -Wwrite-strings -Wcast-qual -Wpointer-arith -Wsign-compare -Wundef -Wmissing-declarations -Wredundant-decls -Wmissing-noreturn -Wvla -Wno-missing-field-initializers -Wcast-align -Werror -Wno-error=deprecated-declarations   -I./src/unix -I./output/include -isystem ./output/src/boost_1_85_0 -DBOOST_NO_IOSTREAM -DBOOST_MATH_NO_LEXICAL_CAST -DBOOST_UBLAS_NO_STD_CERR -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_SYSTEM_NO_DEPRECATED -DBOOST_NO_STD_LOCALE -DBOOST_LEXICAL_CAST_ASSUME_C_LOCALE -DBOOST_NO_CXX98_FUNCTION_BASE -I./src -I./src/Engine -I./output/IOS64/include -DHAVE_POSIX -DHAVE_VASPRINTF -isystem ./output/IOS64/lib/aarch64-apple-darwin/include -DEYE_CANDY -DXCSOAR_TESTING -isystem /include    -miphoneos-version-min=10.0 -arch arm64 -target aarch64-apple-darwin -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS.sdk     
ccache clang++ src/Math/FastMath.cpp -c -o output/IOS64/dbg/src/Math/FastMath.o -MD -MF output/IOS64/dbg/src/Math/FastMath.d -MT output/IOS64/dbg/src/Math/FastMath.o -O0 -g   -std=c++20 -fno-threadsafe-statics -fmerge-all-constants -fcoroutines-ts -fvisibility=hidden -ObjC++ -fobjc-arc -stdlib=libc++ -Wall -Wextra -Wwrite-strings -Wcast-qual -Wpointer-arith -Wsign-compare -Wundef -Wmissing-declarations -Wredundant-decls -Wmissing-noreturn -Wvla -Wno-missing-field-initializers -Wcast-align -Werror -Wno-error=deprecated-declarations   -I./src/unix -I./output/include -isystem ./output/src/boost_1_85_0 -DBOOST_NO_IOSTREAM -DBOOST_MATH_NO_LEXICAL_CAST -DBOOST_UBLAS_NO_STD_CERR -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_SYSTEM_NO_DEPRECATED -DBOOST_NO_STD_LOCALE -DBOOST_LEXICAL_CAST_ASSUME_C_LOCALE -DBOOST_NO_CXX98_FUNCTION_BASE -I./src -I./src/Engine -I./output/IOS64/include -DHAVE_POSIX -DHAVE_VASPRINTF -isystem ./output/IOS64/lib/aarch64-apple-darwin/include -DEYE_CANDY -DXCSOAR_TESTING -isystem /include    -miphoneos-version-min=10.0 -arch arm64 -target aarch64-apple-darwin -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS.sdk     
clang++ -O0 -g -L./output/IOS64/lib/aarch64-apple-darwin/lib -stdlib=libc++    -miphoneos-version-min=10.0 -arch arm64 -target aarch64-apple-darwin -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS.sdk   -o output/IOS64/bin/vali-xcs output/IOS64/dbg/src/Logger/GRecord.o output/IOS64/dbg/src/util/MD5.o output/IOS64/dbg/src/Version.o output/IOS64/dbg/src/VALI-XCS.o output/IOS64/dbg/io.a output/IOS64/dbg/fmt.a output/IOS64/dbg/libos.a output/IOS64/dbg/util.a  -lz -L/Users/yorickreum/Coding/XCSoar/output/IOS64/lib/aarch64-apple-darwin/lib -lfmt   -lpthread -framework Foundation -framework UIKit   
src/Computer/ThermalRecency.cpp:11:7: error: use of undeclared identifier 'THERMALRECENCY'
    ? THERMALRECENCY[x]
      ^
1 error generated.
gmake: *** [build/compile.mk:103: output/IOS64/dbg/src/Computer/ThermalRecency.o] Error 1
gmake: *** Waiting for unfinished jobs....
gmake: *** [build/vali.mk:12: output/IOS64/bin/vali-xcs] Interrupt: 2
```

Do you have any idea what may have caused this?
-----------------------------------------------

It's complaining that ```THERMALRECENCY``` is not defined in https://github.com/XCSoar/XCSoar/blob/master/src/Computer/ThermalRecency.cpp
I do agree, I can't find where the variable is declared.

Do you have an idea how to solve the issue?
-------------------------------------------

I don't even understand why it's compiling without the error usually.
One can usually circumvent the problem by running ```gmake clean``` and building again from scratch. This seems to solve it. But I'd really like to understand what's happening here.

dssherrill, 2025-01-20:
> latest from git 273cc7f

Try  gmake -j1.  If a clean build does not produce the same result every
time, then a problem with parallel builds is a prime suspect.  Some years
ago -j1 was required for my local XCSoar build to succeed, but the symptom
was different:  a file was sometimes missing.  Or perhaps a lock file was
sometimes present.  I do not remember the details.

   ...david

yorickreum, 2025-01-25:
Hi David, thanks for getting back to me. Passing ```-j1``` doesn't seem to change anything. I also just wonder how this supposed to work normally, meaning how ```THERMALRECENCY``` is supposed to be defined.

yorickreum, 2025-02-01:
> This is generated code with tools/GenerateSineTables.cpp This will generate the definition in ./output/include/MathTables.h

Thanks, that helped. Indeed, in the cases where the build fails, ```MathTables.h``` is just an empty file. Deleting it immediately fixes the build capability. Remains the question why it's empty sometimes ... will dig into that as soon as I find the time.