XCSoar/XCSoar #1655: Unable to calculate wind

Popets, 2025-01-29:
XCSoar versions having the problem 7.43
-------------------------------------------------

System information
------------------
Kobo Glo HD with BlueFly Vario

I have been flying XCSoar with Kobo Glo HD and I have noticed that it doesn't calculate wind direction. I have a few thermals circled for about an hour but it never calculates the wind. I have enabled circle calculation and zig zag calculation, disabled external sources. I am not sure if I am missing some setting but I am unable to get the wind to update.

Steps to reproduce the behavior
-------------------------------
Fly around in thermals.

Expected behavior
-----------------
Wind is calculated and windspeed infobox is updated and wind arrow is displayed on map.

Actual behavior
---------------

No wind available both in info boxes and map

Do you have any idea what may have caused this?
-----------------------------------------------

I do not know what is causing it

Do you have an idea how to solve the issue?
-------------------------------------------
I do not know how to solve the issue

ubx, 2025-01-30:
I tried replying with [2025-01-25-XCS-AAA-03.txt](https://github.com/user-attachments/files/18594915/2025-01-25-XCS-AAA-03.txt) on 7.44-Kobo and 7.43-PC but could not reproduce the problem.

[image] \\\

[image] \\\

[image]  \\\

[image]

Popets, 2025-01-30:
Could you point me to which build you are using so I can put it on my Kobo? the 7.44 in particular, since there are changes in the newer prs that are not released that I use.

Popets, 2025-01-30:
Also could you tell me the terrain settings you are using, since mine isn't nearly as well defined? @ubx

ubx, 2025-01-30:
> Could you point me to which build you are using so I can put it on my Kobo? the 7.44 in particular, since there are changes in the newer prs that are not released that I use.

Version 7.44 is not yet released, but if you want to try it, you can get it, for example, from [here](https://github.com/XCSoar/XCSoar/actions/runs/13047785525/artifacts/2509127536)

> Also could you tell me the terrain settings you are using, since mine isn't nearly as well defined? @ubx

[image]

Popets, 2025-01-31:
Ok with this version I am getting it to display wind. I am not sure why it wasn't doing so before.