XCSoar/XCSoar #1657: battery low warning for no reason.

hjr, 2025-01-31:
XCSoar versions having the problem
-------------------------------------------------
most recent 7.4x

XCSoar versions not having the problem
-------------------------------------------------
unknown

System information
------------------
XCTN7v2 a XCS specialized nav android 11 with touch scrren.

Steps to reproduce the behavior
-------------------------------
switch on sit and wait.

Expected behavior
-----------------
Battery gauge shows "Ok".

Actual behavior
---------------
From time to time there is a low voltage warning. But the XCTN7v2 is connected to the on board 12V supply and should "never" show something like this.

Do you have any idea what may have caused this?
-----------------------------------------------
no. Maybe an Android API issue.

Do you have an idea how to solve the issue?
-------------------------------------------
ask Android experts

lordfolken, 2025-01-31:
So we are recieving android intents here. Basically the os is telling us that the device is unplugged. The warning comes at less than 10 percent battery. If its less than 5 xcsoar shutsdown to preserve the flight log. 

Intents
https://github.com/XCSoar/XCSoar/blob/master/android/src/BatteryReceiver.java

Logic for battery checks:
https://github.com/XCSoar/XCSoar/blob/master/src/BatteryTimer.cpp

Percentage definitions:
https://github.com/XCSoar/XCSoar/blob/master/src/BatteryTimer.hpp

While I think the shutdown should be overridable... the rest of the logic is sound. One thing I do notice is that the default values in xcsoar are 0 so if the intents are broadcasted then this may be an issue.

As far as I gather from the product description the device doesn't have a backup battery?

@xcnav do you have an idea?

hjr, 2025-01-31:
> 
> [@XCNav](https://github.com/XCNav) do you have an idea?

That is the wrong link. -> https://xcvario.com/product/xctouchnav/

lordfolken, 2025-01-31:
> You are right, the android devices itself has a little coin cell to keep time and date. Does Android through warnings becaus of this coin cell?

I meant a backup batttery in the sense that the device keeps running without 12v from the aircraft battery. According to the product page this is not the case.

So if this device reports battery levels its purley based on voltages. Potentially you have voltage drops on your electrics.

XCNav, 2025-02-03:
> So we are recieving android intents here. Basically the os is telling us that the device is unplugged. The warning comes at less than 10 percent battery. If its less than 5 xcsoar shutsdown to [...]

Hi, I can only speak for our XCNav products. Maybe The XCTN works similar.
XCNav has an internal voltage regulation up to 35V input voltage and us not using the onboard power Jack of the mainboard. That may result in a lower voltage than 12V as power input on the mainboard due to the loss of the power distribution. I suggest to involve @iltis42 from XCV for more information.

lordfolken, 2025-02-08:
So i today learn about the android battery scale factor. https://developer.android.com/training/monitoring-device-state/battery-monitoring

The attached PR takes the scale into account and also checks for invalid values from the battery readout. 

Please test the PR (there is an apk attached to it.)

iltis42, 2025-02-08:
sure we want to test, bue where can we get the .apk, i don't this here, nor in the attached PR

lordfolken, 2025-02-08:
You can download it from the PR github action page: https://github.com/XCSoar/XCSoar/actions/runs/13216917352?pr=1668

lordfolken, 2025-02-08:
New build: https://github.com/XCSoar/XCSoar/actions/runs/13218004070 with a battery present check.

closed 2025-02-09 by lordfolken