XCSoar/XCSoar #1663: XCSoar crashes at Android 13 and 14

Blaubart, 2025-02-07:
I tried to install XCSoar V7.43 at Android 13 and 14. If I start XCSoar I can choose Fly or Sim. With both option XCSoar crashes. Last version that works is V7.41. 
This is written in xcsoar.log:

`[2025-01-29T20:32:34Z] Display dpi=95,101
[2025-01-29T20:32:34Z] OpenGL: mda=false npot=true stencil=0x88f0
[2025-01-29T20:32:36Z] Startup dialog
[2025-01-29T20:32:36Z] Loading profiles
[2025-01-29T20:32:36Z] Failed to load profile: Failed to open '/storage/emulated/0/Android/media/org.xcsoar/default.prf': No such file or directory
[2025-01-29T20:32:36Z] Loading language file
[2025-01-29T20:32:36Z] Loading input events file
[2025-01-29T20:32:36Z] Loading Topography File...
[2025-01-29T20:32:36Z] ReadWaypoints
[2025-01-29T20:32:36Z] Failed to read waypoint file: /storage/emulated/0/Android/media/org.xcsoar/user.cup
[2025-01-29T20:32:36Z] Failed to open '/storage/emulated/0/Android/media/org.xcsoar/user.cup': No such file or directory
[2025-01-29T20:32:36Z] RASP load
[2025-01-29T20:32:36Z] No rasp data file: Failed to open ZIP archive /storage/emulated/0/Android/media/org.xcsoar/xcsoar-rasp.dat
[2025-01-29T20:32:36Z] ReadAirspace
[2025-01-29T20:32:36Z] Register serial devices
[2025-01-29T20:32:36Z] ProgramStarted
[2025-01-29T20:32:36Z] cannot open /storage/emulated/0/Android/media/org.xcsoar/lua/init.lua: No such file or directory`

The folder "/Android/media/org.xcsoar" exits, but there is only the file xcsoar.log.
It seems to be a problem during installation of XCSoar. Default files and folders are not created. If I use my files from an other tablet and put them to Android/media XCSoar dosen’t crash anymore, but stops starting.

ubx, 2025-02-07:
Why do you say that XCSoar crashes? The log only indicates, for instance, that `Failed to load profile: Failed to open '/storage/emulated/0/Android/media/org.xcsoar/default.prf: No such file or directory`. If `default.prf `does not exist, XCSoar will create it in Fly mode upon exit.

Everything works fine on the Pixel 9 with Android 13."

Blaubart, 2025-02-07:
Sorry if I used the wrong expression. XCSoar starts, and is closed immediately. Thats looks for my like software crashes.
I'm not the only one with this problem:
https://forum.xcsoar.org/viewtopic.php?p=25280#p25280

lordfolken, 2025-02-07:
The "cannot  open" messages, are just indicators that it tried to open files, and would open them if they existed. These messages are wrongly worded, and are not per se errors. 

Note: XCSoar auto exits if your battery is below 5%.  (in fly)

Can you tell me:
  * does it work in sim mode?
  * what device are you running?
  * can you try latest development builds: https://download.xcsoar.org/tesitng/ANDROID ?
  * If at all possible can you please attach a logcat of xcsoar start?

In general i can say that XCSoar has been tested up to android 15, and seems to work. 

Potentially you have ldpi screen and suffer from this problem:
[  - fix crash on startup when loading icons on ldpi screens](https://github.com/XCSoar/XCSoar/blob/473a810159de28b73401ff0e91d030197024060b/NEWS.txt#L24)

Blaubart, 2025-02-07:
- does it work in sim mode?
No is the same

- what device are you running?
a no name Android board from China

- can you try latest development builds: https://download.xcsoar.org/tesitng/ANDROID ?
This version works

- If at all possible can you please attach a logcat of xcsoar start?
still necessary?

Blaubart, 2025-02-08:
I forget to witch displays i'm using:
https://www.texim-europe.com/product/display-and-touch-solutions/display-panels/tft-lcd-display/detail/am-1280800p3tzqw-00h-ampi
https://www.texim-europe.com/product/display-and-touch-solutions/display-panels/tft-lcd-display/detail/ch070oldlth-002-chef

lordfolken, 2025-02-08:
So there is another bug report here #1657 and i had a look at the code. 

First Problem I see:
https://github.com/XCSoar/XCSoar/blob/df93d8962b874542878c030a2ded4b7c36103ea5/android/src/BatteryReceiver.java#L15-L16

So the value is initialized to 0. If your droid fails to report proper battery levels, this will be at 0 therefore triggering the shutdown here:
https://github.com/XCSoar/XCSoar/blob/df93d8962b874542878c030a2ded4b7c36103ea5/src/BatteryTimer.cpp#L38-L49

Looking at android examples, they always initialize these values to -1, so technically an invalid value which we should check for and not react to. 
https://developer.android.com/training/monitoring-device-state/battery-monitoring#java

lordfolken, 2025-02-08:
Out of curiosity, can you configure the infobox battery percentage, and start in fly mode?

lordfolken, 2025-02-08:
@Blaubart can you try the attached pull request please?

Blaubart, 2025-02-08:
> Out of curiosity, can you configure the infobox battery percentage, and start in fly mode?

How can I try configure battery percentage?? Clicking to the box dosen't work

Blaubart, 2025-02-08:
> [@Blaubart](https://github.com/Blaubart) can you try the attached pull request please?

I can not compile at the moments I can't include this changes. If you could send me an Android testing version I can test.

lordfolken, 2025-02-08:
You can download the apk from the github build action page: https://github.com/XCSoar/XCSoar/actions/runs/13216917352?pr=1668

Blaubart, 2025-02-08:
Yes, Sim mode works

lordfolken, 2025-02-08:
What does the battery percentage box say? is it realitstic?

Blaubart, 2025-02-08:
0%, AC off

lordfolken, 2025-02-08:
Your device has a battery and is a regular phone?

Blaubart, 2025-02-08:
No, it is a no name Android board from China and it has no battery. It is connected with a power adapter.

lordfolken, 2025-02-08:
Ok i added a check that sets plugged to true if no battery is present. 

Can you please install: https://github.com/XCSoar/XCSoar/actions/runs/13218004070

The battery percentage box should show the following:
0% and AC ON

Blaubart, 2025-02-08:
Yes, it works

closed 2025-02-09 by lordfolken