XCSoar/XCSoar #1669: New InfoBox title size do no remember user setting and defaults to 100% after XCSoar restart

Sundown3867, 2025-02-09:
XCSoar versions having the problem
-------------------------------------------------

Latest build of 7.44 

XCSoar versions not having the problem
-------------------------------------------------

- new feature

System information
------------------

Android/ WIN64 (have not tested on other platforms)

Steps to reproduce the behavior
-------------------------------

Set the InfoBox title size to size other than 100% and restart the XCSoar. Size will return to 100%

Expected behavior
-----------------

XCSoar will remember user set value for InfoBox title size

Actual behavior
---------------

Do you have any idea what may have caused this?
-----------------------------------------------

No

Do you have an idea how to solve the issue?
-------------------------------------------

Review the new code

lordfolken, 2025-02-09:
I'm having trouble reproducing this on WIN64 or Linux. 

In your default.prf there should be a line with:
InfoBoxTitleScale="85"

Can you attach your prf file, before and after?

lordfolken, 2025-02-09:
So looking at the code:
  There is a check that invalidates incorrect values here, this one allows a range of 80 to 150. if outside that range, it sets it back to 100. 
https://github.com/XCSoar/XCSoar/blob/68855aa9a9dfa9abb8f81b104378a6936e393fce/src/Profile/InfoBoxConfig.cpp#L113-L115

The Settings Dialog however allows to set the font from 50% to 150%. 
https://github.com/XCSoar/XCSoar/blob/68855aa9a9dfa9abb8f81b104378a6936e393fce/src/Dialogs/Settings/Panels/LayoutConfigPanel.cpp#L187-L189

So setting a small value of below 80% will fail that check when loading from profile. However  will allow it when setting it at runtime. 

@Sundown3867 is it correct that this fails when you set numbers below 80%, but not if you use higher values?

@bomilkar What do you think? This check is only used at startup when reading from profile, but not at runtime. Should we remove it alltogether, or set the range so it matches the dialog's range?

bomilkar, 2025-02-10:
I'd rather change `XCSoar/src/Profile/InfoBoxConfig.cpp line 114` to match the range of the dialog (50% .. 150%).
I can submit a PR if you want. Just let me know, @lordfolken .

Sorry for the mess.

bomilkar, 2025-02-10:
Done: https://github.com/XCSoar/XCSoar/pull/1671

commit a4b93455 (2025-02-10): Profile/InfoBoxConfig: Consitent check for dialog and startup

closes #1669
closed 2025-02-10 by lordfolken