XCSoar/XCSoar #1690: Stratux Support

fraktal-sb, 2025-02-22:
XCSoar version
--------------

7.44

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

FLARM and ADS-B are common traffic awareness protocols in general aviation.
Stratux is an open source project to deliver those data to various EFB's (Electronic Flight Bag's)
Is is described at http://stratux.me/ and https://github.com/stratux/stratux

Because of the NMEA format XCSoar works "nearly" out of the box with but is not usable. The radar page of XCSoar is overloaded with ADS-B targets. see the following screenshots

[image]

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

 A configurable filter is missing (compared to the "safesky app" (see https://www.safesky.app)

Then the traffic radar could be seen as follows

[image]

This could be done by defining a new device "Stratux" in the device list and developing a special driver and configuration user interface.

The Stratux device could be made available at the device list page

[image]

A typical connection could be made by WiFi as follows.

[image]

Then the special setup of the horizontal and vertical filter range could be made as follows by using "Manage"
button.

[image]

[image]

The implementation will be soon available by pull request

fraktal-sb, 2025-03-07:
actually the PR build checks fail with:

/usr/bin/i686-w64-mingw32-ld: /__w/XCSoar/XCSoar/src/Device/Driver/Stratux/Driver.cpp:20: undefined reference to `Profile::Get(std::basic_string_view<char, std::char_traits<char> >, char const*)'
collect2: error: ld returned 1 exit status

and so on...

It seems to be a overload problem or a problem of the build check procedure. Linking and the generated execute file works perfectly on my Debian Bookworm system.

actually no idea how to proceed.

lordfolken, 2025-03-07:
Char isnt going to work on windows. Use the TCHAR macro.

fraktal-sb, 2025-03-09:
Tried different approaches with TCHAR and int. Cant get rid of the failing build checks.

yorickreum, 2025-03-12:
> actually the PR build checks fail with:
> 
> /usr/bin/i686-w64-mingw32-ld: /__w/XCSoar/XCSoar/src/Device/Driver/Stratux/Driver.cpp:20: undefined reference to `Profile::Get(std::basic_string_view<char, std::char_traits<char> >, char const*)'
> collect2: error: ld returned 1 exit status

I'm a bit puzzled why it works for you tbh. Imo the build order/dependencies in the Makefiles are just not right for your case, and it makes sense that the linker can't find the references. See my comment here https://github.com/XCSoar/XCSoar/pull/1710#issuecomment-2715950035.
Is mingw somehow resolving the headers to figure out which files to link?

fraktal-sb, 2025-06-03:
Development is continued in PR https://github.com/XCSoar/XCSoar/pull/1721; code review is actualy pending

closed 2025-07-14 by lordfolken