XCSoar/XCSoar #1693: iOS: Power consumption is intense

yorickreum, 2025-02-22:
XCSoar versions having the problem
-------------------------------------------------

current from master https://github.com/XCSoar/XCSoar/commit/c293d2cdcab6f6e7ce0f72cd343038ea9d9bb34d

XCSoar versions not having the problem
-------------------------------------------------

unknown to me

System information
------------------

iOS 18.3.1 (and others), real device

Steps to reproduce the behavior
-------------------------------

Start XCSoar on an iOS device, feel the device getting warm and watch the battery getting drained.

Expected behavior
-----------------

Consume a reasonable amount of power.

Actual behavior
---------------

Too much CPU time wasted.

Do you have any idea what may have caused this?
-----------------------------------------------

Not yet.

Do you have an idea how to solve the issue?
-------------------------------------------

I'll try to attach the Xcode Time Profiler and try to figure out what is draining so much computational power here.

yorickreum, 2025-03-03:
Xcode Time Profiler of iOS simulator says most of the CPU time goes into ```SDL_WaitEventTimeout``` called in ```UI::EventQueue::Wait(UI::Event&)```, about 30 % a few seconds after startup. Might be normal, not sure? Will investigate again on a real device.

yorickreum, 2025-05-07:
The main issue here might ultimately be OpenGLES-related. Even on the XCSoar start screen, the CPU load caused by the app is about 120 %. Going through the SDL iOS Demo projects, they all show a default CPU load of about 20-30%—with the exception of the Fireworks project, which also uses OpenGLES; there, the initial load is also > 120 %.

TheMiNuS, 2025-06-29:
On the power consumption one of the main problem is that the app do not run in background when the phone is locked. When you unlock the phone XCSoar restart completely and ask again fly/sim.

This improvement could be great to have the app running in background.

yorickreum, 2025-07-12:
> This improvement could be great to have the app running in background and save energy in addition to not restart all the nav.

This would indeed be great; the app restarting after the iPhone went to sleep also bothered me quite a bit when flying with it.

> Am I correct in assuming that something like SDL_WaitEventTimeout(&event.event, 100); won’t help either, because SDL_WaitEvent is a synchronous function that blocks until the next event arrives? 

This is my understanding of things as well. `SDL_WaitEventTimeout()` just makes things worse, as it sets a timeout at which it stops waiting, leading to even more loop iterations.

One option might be to actively throttle the fps via `SDL_Delay()`. Doing so in the SDL fireworks example brings down the idle CPU load there significantly into a somewhat reasonable regime; see https://github.com/libsdl-org/SDL/pull/13353.

> Probably a silly idea that would take too much effort, but could the xcsoar binary be used as a backend, in combination with a SwiftUI frontend that uses something like MapLibre Native (OpenGL or Metal)?

In general, the XCSoar rendering is pretty flexible and already supports different rendering frameworks (OpenGL, GTX, software rendering, ... see e.g., subdirs in `src/ui/canvas`). Of course, one could also think about understanding XCSoar as a library to be used with a SwiftUI frontend, but in this case, it would require a lot of maintenance effort to keep the extra frontend up-to-date with every XCSoar change. So I think if we want to go for Metal support, we should just port it in as we do with the other rendering frameworks...

yorickreum, 2025-07-13:
> This improvement could be great to have the app running in background and save energy in addition to not restart all the nav.

_OpenGL ES May Not Be Used in Background Apps
Apps that are running in the background may not call OpenGL ES functions. If your app accesses the graphics processor while it is in the background, it is automatically terminated by iOS. To avoid this, your app should flush any pending commands previously submitted to OpenGL ES prior to being moved into the background and avoid calling OpenGL ES until it is moved back to the foreground._
from [https://developer.apple.com/library/archive/documentation/3DDrawing/Conceptual/OpenGLES_ProgrammingGuide/Introduction/Introduction.html#//apple_ref/doc/uid/TP40008793](https://developer.apple.com/library/archive/documentation/3DDrawing/Conceptual/OpenGLES_ProgrammingGuide/Introduction/Introduction.html#//apple_ref/doc/uid/TP40008793)

WolfgangDrescher, 2025-07-14:
> One option might be to actively throttle the fps via SDL_Delay(). Doing so in the SDL fireworks example brings down the idle CPU load there significantly into a somewhat reasonable regime; see https://github.com/libsdl-org/SDL/pull/13353.

Your change in `libsdl` looks promising. Does something speak against the integration in XCSoar? It looks like all platforms using SDL would benefit from this change. Or should we wrap it in `#ifdef __APPLE__` to be sure that nothing else breaks?

yorickreum, 2025-07-20:
> > One option might be to actively throttle the fps via SDL_Delay(). Doing so in the SDL fireworks example brings down the idle CPU load there significantly into a somewhat reasonable regime; see [libsdl-org/SDL#13353](https://github.com/libsdl-org/SDL/pull/13353).
> 
> Your change in `libsdl` looks promising. Does something speak against the integration in XCSoar? It looks like all platforms using SDL would benefit from this change. Or should we wrap it in `#ifdef __APPLE__` to be sure that nothing else breaks?

I just tried to implement the `SDL_PollEvent` + `SDL_Delay` change in the XCSoar codebase; however, it sadly does not reduce the CPU load noticeably. I assume that `SDL_WaitEvent` blocks as efficiently as `SDL_Delay`...

So ultimately, I think we can't optimize the event loop any further. Assuming the performance issue is due to inefficient rendering, maybe we should indeed look into implementing a Metal backend. We can maybe do this step by step by just adding a Metal canvas and then drawing the existing OpenGL objects into it…

WolfgangDrescher, 2025-07-20:
Thanks for trying it out @yorickreum and too bad it doesn't make the desired performance difference. Is there still a way to debug if too many events are triggered? In which file in the code could I try to debug further?

I'm happy to help migrate to Metal, but I have no experience with it and would need guidance at the beginning.
How complex is a migration to Metal? A proof of concept would be useful to see if this would solve the CPU problems. It probably still makes more sense to use a Metal backend than to write a new frontend with Swift, right?

Let me know if I can help in any way.

yorickreum, 2025-07-29:
Just discovered something: Using `usleep(16000);` to slow down works and brings down effective CPU load < 5 % in idle, while `SDL_Delay(16);` does not seem to have any effect... Apparently in our setup something is off with `SDL_Delay()`.