XCSoar/XCSoar #1695: iOS: Waypoint icons look weird

yorickreum, 2025-02-22:
XCSoar versions having the problem
-------------------------------------------------

current from master https://github.com/XCSoar/XCSoar/commit/c293d2cdcab6f6e7ce0f72cd343038ea9d9bb34d

XCSoar versions not having the problem
-------------------------------------------------

unknown to me

System information
------------------

iOS 18.3.1 (and others), real device and simulator

Steps to reproduce the behavior
-------------------------------

Launch XCSoar an iOS device, start it, go into sim or fly mode.

Expected behavior
-----------------

Show the icons nicely.

Actual behavior
---------------

Icons look weird, see screenshot.

[image]

Do you have any idea what may have caused this?
-----------------------------------------------

–

Do you have an idea how to solve the issue?
-------------------------------------------

Will look into it; any help is appreciated.

LeNinjaHD, 2025-06-17:
After a restart of the app the icons bug came back and now changing settings doesnt help. Also the icons around the scale are still broken.

lordfolken, 2025-06-17:
The smiley is an icon for teammate IDB_TEAMMATE_POS, and is supposed to signify traffic that is marked as team mate.

yorickreum, 2025-06-17:
> The smiley is an icon for teammate IDB_TEAMMATE_POS, and is supposed to signify traffic that is marked as team mate.

I see. A sky full of team mates sounds a bit too good to be true, so I will look deeper into why this is happening.
Will probably take me a couple weeks to really get to it (next weekend I'll be on a flying camp and then we are busy with [Würzburg Fliegt](https://www.fscw.de/index.php?id=37))... So if anybody wants to debug this in the meantime, any help is more than welcome!

WolfgangDrescher, 2025-06-28:
I’m not entirely sure yet, but it might be related to the alpha channel of colors.

https://github.com/XCSoar/XCSoar/blob/dd1d496283183cd521ae537f52bc6b2dfb482359/src/ui/canvas/Color.hpp#L86-L93

https://github.com/XCSoar/XCSoar/blob/dd1d496283183cd521ae537f52bc6b2dfb482359/src/Look/MapLook.cpp#L35-L40

Using `Color()` instead of `ColorWithAlpha()` or using other colors will always change the appearance of the icons on the map. The transparent backgrounds of the icons are always blurred in some way. Sometimes, there are randomly drawn lines above the icons, sometimes there is a different background color behind the icons, and sometimes a blurred and cloudy looking dotted background appears behind the icons.

---

https://github.com/XCSoar/XCSoar/blob/dd1d496283183cd521ae537f52bc6b2dfb482359/src/Look/WaypointLook.cpp#L12

This line seems to be loading IDB_TEAMMATE_POS in when zoomed out. Is there a resource mapping somewhere else that could mix the icons up? Or a caching layer? But this seems to be another problem than the blurry alpha channel of icons. But I could be wrong.

yorickreum, 2025-07-12:
> I have been digging a bit deeper and it looks like [ui/canvas/Icon.cpp](https://github.com/XCSoar/XCSoar/blob/master/src/ui/canvas/Icon.cpp) or [ui/canvas/opengl/Shaders.cpp](https://github.com/XCSoar/XCSoar/blob/master/src/ui/canvas/opengl/Shaders.cpp) could be related to this problem. So far I've never used OpenGL, so I do not really understand whats going on there. I would appreciate it if someone with more insight into the codebase could help me determine if I am looking in the right place and how to debug the shaders.

Will have a look there as soon as I get to it.

I also realized that the issue doesn't seem to appear on my older iPhone X. On my iPhone 13 Pro and the iPhone 16 Pro simulator it is there.

WolfgangDrescher, 2025-07-12:
I'm not sure how newer iPhones running iOS handle OpenGL ES 2.0, but it looks like `GL_ALPHA` and `GL_LUMINANCE` have been deprecated. So `format` and `internal_format` in `UncompressedImage.cpp` might be relevant as well. Could it be an option to upgrade to a newer version of OpenGL?

yorickreum, 2025-07-13:
iPhone X (physical device, iOS 16.7.10, rendering icons correctly):
```
[2025-07-12T14:03:54Z] Starting XCSoar v7.44-iOS~git#0e003def8c
[2025-07-12T14:03:54Z] SDL_GL config: RGB=5/6/5 alpha=0 depth=24 stencil=8
[2025-07-12T14:03:54Z] GL vendor: Apple Inc.
[2025-07-12T14:03:54Z] GL version: OpenGL ES 2.0 Metal - 99
[2025-07-12T14:03:54Z] GL renderer: Apple A11 GPU
[2025-07-12T14:03:54Z] GL extensions: GL_OES_depth_texture GL_OES_depth_texture_cube_map GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_packed_depth_stencil GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_vertex_array_object GL_EXT_blend_minmax GL_EXT_color_buffer_half_float GL_EXT_debug_label GL_EXT_debug_marker GL_EXT_discard_framebuffer GL_EXT_draw_instanced GL_EXT_instanced_arrays GL_EXT_map_buffer_range GL_EXT_occlusion_query_boolean GL_EXT_pvrtc_sRGB GL_EXT_read_format_bgra GL_EXT_separate_shader_objects GL_EXT_shader_framebuffer_fetch GL_EXT_shader_texture_lod GL_EXT_shadow_samplers GL_EXT_sRGB GL_EXT_texture_filter_anisotropic GL_EXT_texture_rg GL_EXT_texture_storage GL_APPLE_clip_distance GL_APPLE_color_buffer_packed_float GL_APPLE_copy_texture_levels GL_APPLE_framebuffer_multisample GL_APPLE_rgb_422 GL_APPLE_sync GL_APPLE_texture_format_BGRA8888 GL_APPLE_texture_max_level GL_APPLE_texture_packed_float GL_IMG_read_format GL_IMG_texture_compression_pvrtc 
[2025-07-12T14:03:54Z] Display dpi=480,480
[2025-07-12T14:03:54Z] OpenGL: npot=false stencil=0x88f0
```

iPhone 13 Pro (physical device, iOS, not rendering icons correctly):

```
[2025-07-10T22:14:02Z] Starting XCSoar v7.44-iOS~git#0e003def8c
[2025-07-10T22:14:02Z] SDL_GL config: RGB=5/6/5 alpha=0 depth=24 stencil=8
[2025-07-10T22:14:02Z] GL vendor: Apple Inc.
[2025-07-10T22:14:02Z] GL version: OpenGL ES 2.0 Metal - 101
[2025-07-10T22:14:02Z] GL renderer: Apple A15 GPU
[2025-07-10T22:14:02Z] GL extensions: GL_OES_depth_texture GL_OES_depth_texture_cube_map GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_packed_depth_stencil GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_vertex_array_object GL_EXT_blend_minmax GL_EXT_color_buffer_half_float GL_EXT_debug_label GL_EXT_debug_marker GL_EXT_discard_framebuffer GL_EXT_draw_instanced GL_EXT_instanced_arrays GL_EXT_map_buffer_range GL_EXT_occlusion_query_boolean GL_EXT_pvrtc_sRGB GL_EXT_read_format_bgra GL_EXT_separate_shader_objects GL_EXT_shader_framebuffer_fetch GL_EXT_shader_texture_lod GL_EXT_shadow_samplers GL_EXT_sRGB GL_EXT_texture_filter_anisotropic GL_EXT_texture_rg GL_EXT_texture_storage GL_APPLE_clip_distance GL_APPLE_color_buffer_packed_float GL_APPLE_copy_texture_levels GL_APPLE_framebuffer_multisample GL_APPLE_rgb_422 GL_APPLE_sync GL_APPLE_texture_format_BGRA8888 GL_APPLE_texture_max_level GL_APPLE_texture_packed_float GL_IMG_read_format GL_IMG_texture_compression_pvrtc 
[2025-07-10T22:14:02Z] Display dpi=480,480
[2025-07-10T22:14:02Z] OpenGL: npot=false stencil=0x88f0
```

iPhone 16 (simulator, iOS 18.2, not rendering icons correctly):
```
SDL_GL config: RGB=8/8/8 alpha=0 depth=24 stencil=8
GL vendor: Apple Inc.
GL version: OpenGL ES 2.0 APPLE-21.0.27
GL renderer: Apple Software Renderer
GL extensions: GL_OES_depth_texture GL_OES_depth_texture_cube_map GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_packed_depth_stencil GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_vertex_array_object GL_EXT_blend_minmax GL_EXT_color_buffer_half_float GL_EXT_debug_label GL_EXT_debug_marker GL_EXT_discard_framebuffer GL_EXT_draw_instanced GL_EXT_instanced_arrays GL_EXT_map_buffer_range GL_EXT_occlusion_query_boolean GL_EXT_pvrtc_sRGB GL_EXT_read_format_bgra GL_EXT_separate_shader_objects GL_EXT_shader_framebuffer_fetch GL_EXT_shader_texture_lod GL_EXT_shadow_samplers GL_EXT_sRGB GL_EXT_texture_filter_anisotropic GL_EXT_texture_rg GL_EXT_texture_storage GL_APPLE_clip_distance GL_APPLE_color_buffer_packed_float GL_APPLE_copy_texture_levels GL_APPLE_framebuffer_multisample GL_APPLE_rgb_422 GL_APPLE_sync GL_APPLE_texture_format_BGRA8888 GL_APPLE_texture_max_level GL_APPLE_texture_packed_float GL_IMG_read_format GL_IMG_texture_compression_pvrtc 
Display dpi=480,480
OpenGL: npot=false stencil=0x88f0
```

yorickreum, 2025-07-13:
> Apple deprecated opengl es since ios 12. They want you to use metal.

I am aware of that. However, they still ship OpenGLES 2 together with iOS up to the most recent version. And there are some ports of newer versions available. So, in my understanding, with the deprecation they are mainly trying to get people to move to Metal, but it's not strictly necessary yet.

yorickreum, 2025-07-13:
> I'm not sure how newer iPhones running iOS handle OpenGL ES 2.0, but it looks like `GL_ALPHA` and `GL_LUMINANCE` have been deprecated. So `format` and `internal_format` in `UncompressedImage.cpp` might be relevant as well. Could it be an option to upgrade to a newer version of OpenGL?

Upgrading will be a larger task as we are currently just using the version readily provided by Apple. It might just be some sloppy thing that newer GPUs might more strictly enforce (texture alignment, format, or data integrity, e.g., RGBA vs. RGB, alignment of pixel rows, shader code precision e.g. `mediump`, `highp` limits, ... I think we should look deeper into OpenGL errors).

WolfgangDrescher, 2025-07-18:
I logged some OpenGL errors and found multiple occurrences of this errors:

```
[GL ERROR] src/ui/canvas/opengl/UncompressedImage.cpp:48: glPixelStorei - 0x0500 (GL_INVALID_ENUM)
[GL ERROR] src/ui/canvas/opengl/Texture.cpp:64: glTexImage2D 2 - 0x0500 (GL_INVALID_ENUM)
[GL ERROR] src/ui/canvas/Icon.cpp:107: Bind texture - 0x0500 (GL_INVALID_ENUM)
```

The line numbers will be slightly off because I added some debug instructions. I don't really understand OpenGL, so I can't say exactly what's going wrong. I experimented a bit with `internal_format` and `format` in `glTexImage2D()`, and while it does change things, the results are worse (e.g. black boxes instead of the icons).

yorickreum, 2025-07-18:
Hey @coderabbitai, what do you think might be the issue here?

Thanks for investigating, @WolfgangDrescher! I will try to have a deeper look into this within the next couple days.

WolfgangDrescher, 2025-07-19:
Interesting, if coderabbitaI is correct, I was on the right track. However, this seems to require a more extensive migration to OpenGL 3 or even Metal. Trying the `GL_EXT_texture_rg` extension might be another viable option that we should try before.
Without deeper knowledge of OpenGL, I can't be of much help with this problem. At first, it looked like a "good first issue"…

WolfgangDrescher, 2025-07-19:
I already tried that, but I got the following error:

```
error: use of undeclared identifier 'GL_RED'
```

It seems that `GL_RED` is only available in OpenGL ES 3.0 and above.

WolfgangDrescher, 2025-07-20:
Unfortunately, the code from coderabbitai does not work as suggested, and I'm not able to debug further without a better understanding of OpenGL. Is it correct that all changes seem to target `image.GetFormat() == UncompressedImage::Format::GRAY`?

WolfgangDrescher, 2025-08-02:
Since I've been debugging this for days, I'm now taking a more radical approach. I've logged OpenGL error messages after every `gl*` call. Here are my findings:

https://github.com/XCSoar/XCSoar/blob/491fa007d68837331f0e24e9103a021b902d74ed/src/ui/canvas/opengl/Program.hpp#L107
https://github.com/XCSoar/XCSoar/blob/491fa007d68837331f0e24e9103a021b902d74ed/src/ui/canvas/opengl/Texture.hpp#L54
https://github.com/XCSoar/XCSoar/blob/491fa007d68837331f0e24e9103a021b902d74ed/src/ui/canvas/opengl/FBO.hpp#L48
https://github.com/XCSoar/XCSoar/blob/491fa007d68837331f0e24e9103a021b902d74ed/src/ui/canvas/opengl/FBO.hpp#L67

---

| code | fn | count |
|-|-|--|
| 0x500 | `glUseProgram` | 93 |
| 0x500 | `glDeleteTextures` | 3 |
| 0x500 | `glDeleteRenderbuffers` | 2 |
| 0x500 | `glBindFramebuffer` | 32|

WolfgangDrescher, 2025-08-02:
Oh well… if I check for an OpenGL error before those function calls, the same error gets caught. So it seems they're not the problem themselves. I must have missed something. But I'm pretty confident I've logged most `gl*` function calls. Any idea what I am missing?

WolfgangDrescher, 2025-08-02:
https://github.com/XCSoar/XCSoar/blob/491fa007d68837331f0e24e9103a021b902d74ed/src/ui/window/sdl/TopWindow.cpp#L213

WolfgangDrescher, 2025-08-02:
> [XCSoar/src/ui/window/sdl/TopWindow.cpp](https://github.com/XCSoar/XCSoar/blob/491fa007d68837331f0e24e9103a021b902d74ed/src/ui/window/sdl/TopWindow.cpp#L213)
> 
> Line 213 in [491fa00](/XCSoar/XCSoar/commit/491fa007d68837331f0e24e9103a021b902d74ed)
> 
>  SDL_GL_GetDrawableSize(event_window, &real_w, &real_h);

This tracks down to:

https://github.com/XCSoar/XCSoar/blob/491fa007d68837331f0e24e9103a021b902d74ed/src/ui/window/custom/TopWindow.cpp#L161-L162

WolfgangDrescher, 2025-08-02:
It looks like

https://github.com/XCSoar/XCSoar/blob/491fa007d68837331f0e24e9103a021b902d74ed/src/ui/window/custom/TopWindow.cpp#L155-L159

needs to be

```cpp
static constexpr GLenum attachments[3] = {
  GL_COLOR_ATTACHMENT0,
  GL_DEPTH_ATTACHMENT,
  GL_STENCIL_ATTACHMENT
};
```

on Apple OpenEL ES.

With this change all OpenGL error messages are gone. But sadly the icons are still rendered wrong and with artifacts.

WolfgangDrescher, 2025-08-04:
I did not notice before but it actually *is* the same problem:

[image]

[image]

Do you see the purple clown nose in the smiley? It is `map_small` with the `map_teammate` in the background. So maybe there is a `glClear` or `glClearColor` missing somewhere? I don't understand the concept of textures, so I'm not sure where I can debug this.

yorickreum, 2025-08-06:
Okay. I will build a minimal example and see where we can get at the next bad weather day.

yorickreum, 2025-08-20:
Figured it out. See https://github.com/XCSoar/XCSoar/pull/1846. We were barking up the wrong tree. It was not an OpenGL issue in the end, but a `malloc` vs. `calloc` issue (that somehow only showed up in newer iOS versions).

WolfgangDrescher, 2025-08-21:
Good catch, nice job! At least I looked for it in the right file. This is one of those bugs where you spend days and weeks searching, and then two characters make all the difference, and you're not sure whether you're satisfied that you've solved the problem or frustrated because the change is so small :D

With this fix and the solved problems with the battery drain, the iOS version is now a significant step further, thanks for your work @yorickreum. What is your release schedule? Are iOS releases in sync with  XCSoar releases, or are there off-cycle releases based on master? I'm in TestFlight, so I'm not in a hurry, but I know that many club members here would be happy to see these changes deployed.

closed 2025-08-22 by lordfolken