XCSoar/XCSoar #1698: Raspberry pi 5 won't start XCSoar

ludesert, 2025-02-24:
XCSoar versions having the problem
-------------------------------------------------

XCSoar versions not having the problem
-------------------------------------------------

System information
------------------

Steps to reproduce the behavior
-------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2025-02-27:
Plesse build current head. The issue you mention was addressed in https://github.com/XCSoar/XCSoar/commit/b4d66cbbb17670780b2cd3cf81bf48ed38b8d3c7

lordfolken, 2025-02-27:
How die you build xcsoar with which options?
Are u running cli (KMS) Variant?
What os are you using?
Is a GUI running anywhere?

ludesert, 2025-02-27:
> Have you tried to build xcsoar on the target (PI4 or PI5)? Please try this first.
> 
> I've seen this on a plain Debian PC when installing from deb package: `/usr/include/c++/12/string_view:276: constexpr void std::basic_string_view<_CharT, _Traits>::remove_prefix(size_type) [with _CharT = char; _Traits = std::char_traits<char>; size_type = long unsigned int]: Assertion 'this->_M_len >= __n' failed. Abandon` We're still debugging this. I suspect it's missing (or outdated) lib.

I built it on the pi5 directly.
I tried building it on an Ubuntu machine but there are no targets other than PI and PI2 in targets.mk.

> How die you build xcsoar with which options? Are u running cli (KMS) Variant? What os are you using? Is a GUI running anywhere?

Simply using make within the latest Raspberry OS 64 bits version (the full version, not the Lite one). Other GUI apps works fine with an HDMI cable.

> Plesse build current head. The issue you mention was addressed in [b4d66cb](https://github.com/XCSoar/XCSoar/commit/b4d66cbbb17670780b2cd3cf81bf48ed38b8d3c7)

Will try this as soon as possible. Currently I only have access to my Ubuntu but can't built for the PI5 as mentioned above.

bomilkar, 2025-02-28:
Before make:
`sudo ./ide/provisioning/install-debian-packages.sh`

lordfolken, 2025-02-28:
> > How die you build xcsoar with which options? Are u running cli (KMS) Variant? What os are you using? Is a GUI running anywhere?
> 
> Simply using make within the latest Raspberry OS 64 bits version (the full version, not the Lite one). Other GUI apps works fine with an HDMI cable.

XCSoar uses KMS on the raspberry pi. (i.e. it runs on the kernel console without any X11 or Wayland) if you have a ui running, it will not be able to take over the console. (Which is why i recommend the lite variant of rasbpberry os) 

go to the shell: 
```
sudo systemctl stop lightdm
sudo systemctl disable lightdm
./output/UNIX/bin/xcsoar
``` 
Profit.

ludesert, 2025-03-02:
> XCSoar uses KMS on the raspberry pi. (i.e. it runs on the kernel console without any X11 or Wayland) if you have a ui running, it will not be able to take over the console. (Which is why i recommend [...]

I tried without stopping lightdm and here is the result before xcsoar's crash

`Starting XCSoar v7.44-Linux~git#9a036c3725
EGL vendor: Mesa Project
EGL version: 1.4
EGL extensions: EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_context_flush_control EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display 
EGL config: RGB=8/8/8 alpha=0 depth=24 stencil=8
GL vendor: Broadcom
GL version: OpenGL ES 3.1 Mesa 23.2.1-1~bpo12+rpt3
GL renderer: V3D 7.1
GL extensions: GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_filter_anisotropic GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_rgtc GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_draw_instanced GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_AMD_performance_monitor GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_NV_conditional_render GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_NV_pack_subimage GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_pack_reverse_row_order GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_instanced_arrays GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_generate_mipmap_sRGB GL_NV_pixel_buffer_object GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_debug_label GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_copy_image GL_EXT_draw_buffers_indexed GL_EXT_draw_elements_base_vertex GL_EXT_polygon_offset_clamp GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_cube_map_array GL_EXT_texture_view GL_KHR_context_flush_control GL_NV_image_formats GL_NV_shader_noperspective_interpolation GL_OES_copy_image GL_OES_draw_buffers_indexed GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_cube_map_array GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_OES_texture_view GL_EXT_buffer_storage GL_EXT_float_blend GL_EXT_geometry_point_size GL_EXT_geometry_shader GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_geometry_point_size GL_OES_geometry_shader GL_OES_shader_image_atomic GL_EXT_clear_texture GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_EXT_texture_mirror_clamp_to_edge GL_KHR_parallel_shader_compile GL_EXT_EGL_image_storage GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod GL_MESA_bgra 
Display dpi=92,91
OpenGL: mda=true npot=true stencil=0x8d48
drmModePageFlip() failed: -13
xcsoar: ./src/util/Manual.hxx :36 : Manual<T>::~Manual() [with T = SignalMonitor]:  l'assertion « !initialized » a échoué.
Abandon`

I tried stopping lightdm, but I can't type anything after that.

I will try later with a raspberry os lite install to see wether it crashes or not.

But my goal would be to run xcsoar and also use other apps (like Firefox for weather forecast and flight planning) with xcsoar running. As I understand, it's not possible if it is using KMS? Could it be possible to run it using x11 on raspbe|y pi?

ludesert, 2025-03-13:
Before installing a new image with Raspberry OS Lite, I used my non lite installation to build with TARGET=WAYLAND.

The compilation failed:

` make TARGET=WAYLAND
  CXX     output/WAYLAND/dbg/src/Asset.o
src/Asset.cpp: In function ‘bool HasTouchScreen()’:
src/Asset.cpp:26:10: error: ‘UI’ has not been declared
   26 |   return UI::event_queue->HasTouchScreen();
      |          ^~
src/Asset.cpp: In function ‘bool HasKeyboard()’:
src/Asset.cpp:32:10: error: ‘UI’ has not been declared
   32 |   return UI::event_queue->HasKeyboard();
      |          ^~
make: *** [build/compile.mk:104: output/WAYLAND/dbg/src/Asset.o] Error 1`

I will try to execute the UNIX output without Wayland on a fresh Raspberry OS Lite installation, but I would like to solve the Wayland issue because I would prefer it rather than the KMS version.

ludesert, 2025-03-14:
The UINIX executable worked on a headless PI OS Lite install.
I still can't make the waykand variant work but will post another issue for this.

Thanks for your help !

closed 2025-03-14 by ludesert