XCSoar/XCSoar #1700: Safe relative height difference to Flarm targets by colour coding in map and radar page

fraktal-sb, 2025-02-25:
XCSoar versions having the problem
-------------------------------------------------

since version 7.32 Flarm targets are colour coded within the map display. Blue means above, green below, pink means the target is on the same height. With version 7.44 the radar targets will be colour coded the same way.

XCSoar versions not having the problem
-------------------------------------------------

below 7.32 this functionality did not exist

System information
------------------

Unix and Android

Steps to reproduce the behavior
-------------------------------

use Test data https://github.com/user-attachments/files/18930179/StratuxTest.txt and FeedNMEA program

Expected behavior
-----------------

Because the colour information is an essential information about the relative height difference this information should be within a safe range. Then the pink colour can be seen as a warning. 100m above and below is considered as a safe minimum height difference.

Actual behavior
---------------

The software codes the targets as expected but the range for targets on the same height is to narrow. 

[image]

Do you have any idea what may have caused this?
-----------------------------------------------

The range is hardcoded with 50 meters above and below within src/Gauge/FlarmTrafficWindow.cpp and src/MapWindow/MapWindowTraffic.cpp

Do you have an idea how to solve the issue?
-------------------------------------------

Change the range from 50m to 100m within the code

The software change will be soon ready for implementation with pull request.

bomilkar, 2025-02-26:
What is the reason for extending the range to +/-100 m? That will lead to lots of false alarms.
Either keep +/-50 or make it a parameter in the setup. Maybe in Glide_Computer -> Safety_Factors.

fraktal-sb, 2025-04-09:
no further need of this requirement (see https://github.com/XCSoar/XCSoar/pull/1702)

closed 2025-04-09 by fraktal-sb