XCSoar/XCSoar #1711: Down-Left gesture, when there is no reachable site

hjr, 2025-03-10:
XCSoar version
--------------
head

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
The emergency gesture Down-Left results in just no reaction at all, when there is no landable site in reach. That is a bit awkward. I would suggest a response of XCS somehow. E.g. "Sorry, no registered field in the reach", or what ever the reasoning for no result is.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
Retrieve the outland hints, as implemented.

FritsvB, 2025-03-10:
Notams would be nice as an extra airspace file

lordfolken, 2025-03-10:
I agree that silently failing is not a good standard.

lordfolken, 2025-03-10:
You could use AddMessage in the check that prevents the Dialog from opening, and popup a message along the mines of no landables in resch.