XCSoar/XCSoar #1717: Compilation for target WAYLAND fails

ludesert, 2025-03-15:
XCSoar versions having the problem
-------------------------------------------------

latest head [761fac6](https://github.com/XCSoar/XCSoar/commit/761fac6e54c265d9011eaf7abfcac4952261d510)

System information
------------------

Raspberry OS (not lite)

Steps to reproduce the behavior
-------------------------------

```
sudo ./ide/provisioning/install-debian-packages.sh
make TARGET=WAYLAND -j4
```

Expected behavior
-----------------

Compilation for wayland

Actual behavior
---------------

The compilation fails:

```
make TARGET=WAYLAND
  CXX     output/WAYLAND/dbg/src/Asset.o
src/Asset.cpp: In function ‘bool HasTouchScreen()’:
src/Asset.cpp:26:10: error: ‘UI’ has not been declared
   26 |   return UI::event_queue->HasTouchScreen();
      |          ^~
src/Asset.cpp: In function ‘bool HasKeyboard()’:
src/Asset.cpp:32:10: error: ‘UI’ has not been declared
   32 |   return UI::event_queue->HasKeyboard();
      |          ^~
make: *** [build/compile.mk:104: output/WAYLAND/dbg/src/Asset.o] Error 1
```

Do you have any idea what may have caused this?
-----------------------------------------------

No.

Do you have an idea how to solve the issue?
-------------------------------------------

No.

ludesert, 2025-03-27:
Any idea ?

lordfolken, 2025-03-27:
Did you investigate?

One intersting thing is that it only fails on the pi. TARGET=WAYLAND builds fine on x86.

ludesert, 2025-03-27:
I will try to built it on x86.

Looking at the error, I thought it could be a configuration error for the Wayland variant. Unfortunately, I have not enough knowledge to read and try to correct the source code ...

ludesert, 2025-03-30:
I tried to build it on debian x84 (virtual machine)? The process was successful but it just creates a black window.

lordfolken, 2025-03-31:
I started sway and started it in a terminal there. 

Anyway the orginal issue must be misplaced #ifdef or missing include.

lordfolken, 2025-12-12:
This is fixed, but i fail to find the commit.

closed 2025-12-12 by lordfolken