XCSoar/XCSoar #1718: XCSoar Shuts Down a Few Minutes After Start of Flight on Android 14 Tablet

kterentjev, 2025-03-16:
A few minutes after starting a flight, XCSoar shuts down and the tablet displays regular home screen. If the glider just sits on the ground prior to flight, XCSoar does not shut down. Once the flight starts, XCSoar Shuts down within 5 to 10 minutes. Tried XCSoar on two different Android 14 tablets with the same result. One of the tablets is made by BNCF having an 8.4 inch screen size. The other tablet is made by fezawio model F10 and 10.1 inch screen size. Both Android 14 tablets were purchased at Amazon.com.
Also tested the tablets by connecting them to a Condor2 glider simulation running on a laptop. Tablets were connected to Condor2 one at a time through Bluetooth. Same result as described above. If the glider is stationary on the ground, XCSoar does not shut down. Once the glider starts flying, XCSoar shuts down within 5 minutes.
XCSoar version tested was 7.44.

XCSoar version 7.43 and 7.42 do not run on Android 14 tablets. XCSoar fails to start. Some sort of display issue.

XCSoar versions having the problem
-------------------------------------------------

 7.44

XCSoar versions not having the problem
-------------------------------------------------
7.42 running on Galaxy Tab A (2016) Android version 5.1.1
Android security patch level 2017-12-01
7.42 running on Android 5.1.1

System information
------------------

BNCF BPAd Mini tablet Android 14 operating system. 
Fezawio model F10 10.1 inch tablet with Android 14 operating system.

Steps to reproduce the behavior
-------------------------------

Start XCSoar on an Android 14 tablet. Make sure XCSOar is configured properly for your area. Use Fly mode for XCSoar. Get in a car with the tablet running XCSoar. Drive around with the car. Within about 5  to 10 minutes XCSoar shuts down.
Same thing happens if the XCSoar Android 14 tablet is connected through Bluetooth to a laptop computer running a Condor2 flight simulation. Once the glider takes off, XCSoar shuts down within about 5 to 10 minutes.

Expected behavior
-----------------

 Expect XCSoar to operate continuously on the Android 14 tablet without unexpectedly shutting down. Should be able to track progress of my flight with XCSoar.

Actual behavior
---------------

 Within about five to ten  minutes after takeoff XCSoar shuts down and Android 14 tablet displays Application screen or Home screen.

Do you have any idea what may have caused this?
-----------------------------------------------

 Something in Android 14 operating system is causing this issue. Could be file security issue.

Do you have an idea how to solve the issue?
-------------------------------------------

 Look in the XCSoar log file after you reproduce this issue.

[xcsoar.log](https://github.com/user-attachments/files/19267651/xcsoar.log)

[xcsoar-old.log](https://github.com/user-attachments/files/19267664/xcsoar-old.log)

lordfolken, 2025-03-17:
Thank you for the detailed bug report. 😃

From your xcsoar log:

src/Engine/Contest/Solvers/AbstractContest.hpp:116: double AbstractContest::ApplyHandicap(double) const: assertion "handicap != 0" failed

This is an assertion failure. These trigger an exit when one is encountered, but this is only triggered in debug builds.

 Please don't fly with debug i.e. Red builds. 

I may have to rerthink how we do the testing releases.

groundhogxc, 2025-04-07:
> Looking forward to using XCSoar on my Android 14 tables when it is fixed.

Just in case you are not aware of this:
You can already do that now, just use the "regular" releases (the ones with a blue icon, and a blue logo when you start it).

If it shows a red icon or red logo when you start it, then it is a testing release of a new version which is not yet considered a stable "regular" version.

kterentjev, 2025-04-08:
I
The latest released version of XCSoar does not start up properly in Android
14 tablets. Whenever XCSoar is started the initial screen appears and after
selecting Fly or Sim  XCSoar shuts itself down. I tried XCSoar on two
different Android 14 tablets made by different manufacturers with the same
results. XCSoar does not run on Android 14 tablets devices. I wish you
could get this fixed.

Konstantin Terentjev

groundhogxc, 2025-04-08:
I see. Could you upload the error message log that happens in this case? It seems to be slightly different, so this might help finding the problem(s).

kterentjev, 2025-04-09:
Yes, I will do that. I will also enter a budget report.

kterentjev, 2025-04-09:
I am getting the following error message when installing XCSoar version 4.3:

INSTALL_FAILED_UPDATE_INCOMPATIBLE: Existing package org.xcsoar signatures
do not match newer version; ignoring!

How can this error message be fixed?

Thanks

Konstantin Terentjev

groundhogxc, 2025-04-09:
Hi Konstantin -

if you are installing from within the phone using an app store or by directly opening an .APK file, then the only way I know is that you need to uninstall the existing version first in order to replace it with the new version when you get this message.

Beware that you should copy/backup/otherwise secure your log files and other data files before you uninstall, as these seem to get deleted when you uninstall (even when klicking "do not delete app data"), at least on my phone.

If you install it from a PC with ADB I think you can force overwriting the installed version with any other one, without first uninstalling from the phone.

kterentjev, 2025-04-10:
I had the test version installed version 7.44 then I uninstalled it but
still get the error message. There could be a remnant of XCSoar that the
uninstall procedure misses. I think the uninstall procedure in the test
version of XCSoar may not be working properly.

groundhogxc, 2025-04-10:
Hi Konstantin -

That is unfortunate. The uninstall routine is actually not in XCSoar, that is an Android operating system routine. So that can not be changed by XCSoar programmers.

I had this problem also once. In that case, when the uninstall routine asked if I want to "keep data", I clicked "yes, keep data".

I could resolve the problem by re-installing the previous version once more from the old APK, and then de-installing it again, On this new de-install I klicked "do not keep data".

After that, I was able to install the new version that I wanted.

Of course, this requires that you still have the old APK file. With a bit of luck, another APK file from the exact same source can also work, if it has the same signature.

lordfolken, 2025-04-10:
adb uninstall org.xcsoar 

Will always purge whatever is left.

If you select keep data, then it will keep the data. The data is however guarded by the app signature. Since this signature has changed. you can only reinstall an XCSoar version with that signature.