XCSoar/XCSoar #1719: Enhanced Traffic Radar Page

fraktal-sb, 2025-03-21:
XCSoar version
--------------

7.44

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

Currently the radar page has 2 views. They are accessible by gesture (see documentation) and show targets with either climb/sink- rate or the height difference. 

These are shown in the following screenshots:

[image]

View "A" shows the targets with climb/sink. View "B" shows the targets with their heigth difference in height unit (meter or feet)

Because a lot more of information came out of current traffic awareness systems (Stratux, Flarm...) an enhenced version of the radar page **(only "B"**) is proposed. View "A" should still contain the reduced information.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

Actualy a test version (with incomming data from the Stratux receiver) is implemented showing the possibilitys

[image]

Here an explanation of the displayed data:

1) callsign from Mode-S Transponder
2) instead of displaying in full accuracy the heigth difference is only shown in "hundreds" of the height unit (as in the small radar
3) Mode-S targets with no position info are displayed in flight direction with their distance and hight difference. To distinguish these targets from the ones with relative east/north they are displayed as a circle with exclamation mark. It means that the targets can be within every directionn but with the received height and distance. the existing colour coding (green,pink,blue) is not affected.
4) the callsign instead of the ICAO-Hexcode is displayed
5) the north direction is no more displayed as a circle (see 4). It is shown only as a "N" and outside the last distance ring

If discussed/accepted these information can be implemented by single feature request for each of the points above or as a whole.

If possible we should coordinate this with https://github.com/XCSoar/XCSoar/issues/1700 because the colour change in units of 50m is not coincident with the displayed hight difference in hundreds of the height difference. 

The small traffic radar on the map page should also contain "no position traffic" as a circle element

fraktal-sb, 2025-04-14:
see XCSoar Manual page 112: gesture right left: The button toggles in-between average vario and altitude displayed next to the target. For me a relative altitude with a precision in meters (as ist is now) makes no sense. Who want's to know that?

add 1) full call sign was my idea in the first version. In the last version (see last screenshot above) I only display the last 2 characters of the callsign which is common in air traffic communication ( after first communication with the full call sign). Pilots recognize others by the last 2 characters

add 2) within the small radar we have the triangle to indicate "above/below". The number beside is the relative height divided by 100. For my taste the small triangle is hard to read but for a first impression good enough. I think most users will switch to the big radar if the small radar pops up. So the information is not so much different within the small and the big radar.

add3) Mode-S Transponders are standard now but unfortunately not all support ADS-B out. To my opinion it is better to have this information on "no position targets" then nothing. In airfield approach it makes sense to have the radar screen open to get an traffic overview. The minimum information from Mode-S targets I get is the call sign, the distance and the relative height. What is more secure? To have those 3 informations or nothing?

add 5) If (!?)  we display "no position targets" how should we do that? This is my proposal. But yes, it collides with the current implementation of the north indicator

fraktal-sb, 2025-04-15:
I know the method to display no position targets by a circle ring according to their distance. I do not prefer this. It is like an additional distance ring and implies a more visual importance then it has. Because others do it it must not be a good standard. The circle and the exclamation mark within is what I propose. Because this element allways is in flight direction this is the most dangerous situation and means "keep your eyes open". Everybody who encounters this element will read in the manual what this means (or ask a friend)

My proposal is to display the short call sign and the heigth difference only in the second mode (toggle gesture right left). Those who want a decluttered competition display with only the climb/sink rates will use the first mode. In the end we will have two major groups of users: those using the radar in competition with reduced/decluttered information (only climb/sink) and the others who wants to see traffic around or on airfield approach (with height difference and short call sign). The 3th group interesting in both will be small they switch between the views.

To implement this we have plenty enough time because the Flarm driver first must provide the basic information about the no position targets. That is something I surely will not do. I have enough to do with the stratux support where I have all functionality above in my private version.

Do we have the graphic method to display a dotted circle?

lordfolken, 2025-04-15:
There is DrawCircle() and i'm sure that it could be extended to create DrawDottedCircle();

fraktal-sb, 2025-04-17:
I experimented a bit with the "No-Position-Target" circle. Yes, it indicates this object can be anywhere but it overloads the radar (my opinion) no matter what type of circle pen is used (in this case solid-mid-grey-colour).

Actualy I have no idea how to program a dotted circle. I can imagine this will be not trivial. For this graphical element we should rise a separate issue if we need this. If somebody will pick up this challenge it is OK for me.

Anyway we should give the user the chance to suppress the extra ring by having a parameter. I don't like this. But as you can see in the example below with the zoom and menu buttons which can be turned off.

[image]

fraktal-sb, 2025-04-17:
Within the code you cannot find any call of DrawArc. Defining a "Pen" with style "dashed" does not draw a dashed arc.
This is the code I used:

    Pen distance_ring_pen;
    distance_ring_pen.Create(Pen::DASH3, Layout::ScalePenWidth(3), Color(0xD0, 0xD0, 0xD0));
    canvas.Select(distance_ring_pen);
    canvas.DrawArc(radar_mid,radius,Angle::Degrees(0),Angle::Degrees(360));

It seams the dashed styles only work in combination with lines where we have enough examples.

So the only graphical tool to reduce the visual importance of the no-position-target-distance-ring is the light grey colour as in the following screenshot

[image]

fraktal-sb, 2025-04-19:
Does somebody have some test data for me?

I need a Powerflarm NMEA stream with FLARM, ADS-B and Mode-S no-position-targets.

fraktal-sb, 2025-04-20:
Can we fix the future layout as follows?

[image]

1) Within the Radar page there are 2 views on Flarm targets selectable by gestures (see page 112 of the XCSoar Manual); the reduced information (only climb/sink- rate) is untouched. As already programmed this rate is only displayed if significant.

2) No-position-targets (from Mode-S transponder without ADS-B out) will be displayed as circle element. The Information of these elements usually came in with distance, height-difference and callsign. These targets will be displayed all ways in flight direction. To show that these targets can be anywhere a dashed distance ring in displayed in Addition (see 3).

3) No-position-target-distance-ring.

4) The zoom in and zoom out buttons are moved vertically from the top to the right side (more space to display the callsign information)

5) More space to display the selected target callsign (see 4).

6) The north indicator is no more displayed as circle with "N" located on the last distance ring. It is only displayed as "N" as in other systems (see above). Within the distance rings only radar targets are displayed. The old display method as circle element with "N" inside would collide with the no-position-target display (see 2)

7) The second Flarm target view (toggle selection by gesture) contains no more the height difference in feet/meter precision. (Who wants to know information this precise?). Instead the height difference is consistent with the small radar displayed in hundreds of the height difference (feed/meter). In addition the last 2 characters of the callsign are displayed above.

Further things to clear:

- How do we distinct "no-position-targets"? (i.e. Stratux sends the distance information in relative_north and relative_east is zero).
- Getting the callsign information has to be modified. (i.e. Stratux has stores this information in addition to the HexId within the PFLAA field "id").
- How much of the callsign do we display in the map. Actually the full callsign is stored in traffic.name and would be displayed in total.
- How to deal with the current ugly zoom and menu Buttons? They overwrite displayed data.
- We need NMEA test data with Flarm-, ADS-B- and no-position-targets.