XCSoar/XCSoar #1725: FLARM messaging interface PFLAM

lordfolken, 2025-03-27:
### Discussed in https://github.com/XCSoar/XCSoar/discussions/1724

<div type='discussions-op-text'>

<sup>Originally posted by **Scumi** March 26, 2025</sup>
Flarm has implemented a messaging interface, which can provide aircraft information like type, registration, pilot name and others without the need of a dedicated database like FlarmNet or OGN.

I have started to extend the FLARM driver to read-out those PFLAM messages. 
The question is, which data should XCSoar prefer: FlarmNet records or PFLAM data? I would tend to the latter because I expect FlarmNet records to be not as up-to-date as Flarm internal data. Furthermore, FlarmNet records might not be maintained anymore by everyone now that Flarm internal data is available to be read-out. Disadvantage of using PFLAM data is that in high traffic situations, the bandwidth for PFLAM messages might be throttled in favor of actual collision data. I personally do not see this as an issue though as in those situations you have other things to do than looking up a pilot name or whatever.
I thought about merging the data, but with a FlarmNet record potentially not maintained, the result can be very wrong and confusing.

What do you think about this?

Documentation:
https://www.flarm.com/wp-content/uploads/2024/12/FTD-109-FLARM-Messaging-Interface-Control-Document-v1.00.pdf</div>

lyusupov, 2025-03-28:
> The question is, which data should XCSoar prefer: FlarmNet records or PFLAM data? I would tend to the latter because ...

Aircraft data that comes from MESSAGING INTERFACE is periodic. It is supplied at low priority rate 1-2 times per minute or less.

[image]

Data from OGN or FlarmNet file can be accessed momentarily.

My opinion that XCSoar traffic display could resolve aircraft ID from a fast data resource first _(such as a database)_ .
When MESSAGING INTERFACE will provide it's own data _(a dozen of seconds or minutes later)_  - XCSoar could update the traffic information based on this data.

> I thought about merging the data, but with a FlarmNet record potentially not maintained

Even though FlarmNet may not be maintained very well -  but the OGN database is updated quite fast, they grow at rate of approx. 1000-1500 new records every year and they have 25000+ records accumulated at this time.

Scumi, 2025-03-31:
> Aircraft data that comes from MESSAGING INTERFACE is periodic. It is supplied at low priority rate 1-2 times per minute or less.

Yes. I hoped that I get all IDs within that minute (i.e. Flarm sends them more or less together), but the dataset I gathered during the weekend shows between 2.25 to 3 minutes for 3 aircraft around. This is in my view too slow to use the data "live". Maybe we can extend the existing "manual input" file or create a new one for all aircraft discovered. This would after a while solve the not-momentarily issue.

> XCSoar could update the traffic information based on the FLARM message.

I don't know, this can be very confusing I think. Imagine you see information belonging to a certain aircraft. Then you look at the screen a few minutes later and this aircraft has disappeared and a new one replaced it. Not an issue with one aircraft next to you, but in high traffic areas not an option I think.

> Even though FlarmNet may not be maintained very well - but the OGN database is updated quite fast, they grow at rate of approx. 1000-1500 new records every year and they have 25000+ records accumulated at this time.

OGN might have a big database, but I question its accuracy to a certain extend. I know 3 aircraft from my club alone that cannot update their OGN credentials because either the previous owner has forgotten his password, does not reply, etc... This is partially solved in https://github.com/glidernet/ogn-ddb/pull/57, but the question remains how many people will update this in the future, now that another messaging interface is available. Remains to be seen.

I think giving the user the choice would be the best option. For now I look into storing the data locally rather than using it live. 

@lordfolken: how does xcsoar-data-content merge the data into the UNIFIED file (which is a FlarmNET+OGN combination I think)?

lordfolken, 2025-03-31:
The united flarmnet file is generated by this project: https://turbo87.github.io/united-flarmnet a github action that runs multiple times daily. 

we only link to the file: 

```
{
  "description": "FlarmNet, OGN and WeGlide live tracking data",
  "uri": "https://turbo87.github.io/united-flarmnet/united.fln"
}
```

I guess the magic is here:
https://github.com/Turbo87/united-flarmnet/blob/main/src/main.rs

lordfolken, 2025-03-31:
As to the PFLAM, we should gather a few nmea logs that have these messages in them. Then one can engineer and test to that.

Since this data arrives slowly, I would collect it in a file and persist it in the xcsoar data folder. That way, once an aircraft is known, lookup is instant.

* a user added record in the profile, always trumps everything else. 
* if a dataset arrives its recorded to the file
* if an id isn't yet known, its looked up in the file
* if its not found, fallback to flarmnet db. (whichever variant)

* There should be an indication in the FlarmTraffic Details dialog, what the source of the traffic is, and what the source for the information is. 

I'm less worried about updating the records life. Potentially you see these PFLAM sentences long before the traffic becomes a concern.

lordfolken, 2025-03-31:
There is a lot of issues with the Flarm Traffic Details dialog #1730

Scumi, 2025-04-02:
I've attached my nmea log from the weekend, just a few aircraft on the ground, but it gives an impression how the PFLAM messages come in.
[2025-03-29_14-10.nmea.log](https://github.com/user-attachments/files/19562368/2025-03-29_14-10.nmea.log)

> There is a lot of issues with the Flarm Traffic Details dialog [#1730](https://github.com/XCSoar/XCSoar/issues/1730)

I suggest we fix some of them before we move working on the messaging interface. I have done some groundwork already, but now its a matter of where to put the read-out data and in what condition. If we treat it like a database, I would not even translate the hex strings before storing it in the file, just during read-out and display.

lordfolken, 2025-04-02:
Im.a fan of separtion between data and display, but
Currently user editable CSV is still a thing in xcsoar. So at the moment i would prefeer the "DB" to be in that Format. 

But internally use the hex strings. Much safer to parse and pass, then arbituary strings.

groundhogxc, 2025-04-07:
> Since this data arrives slowly, I would collect it in a file and persist it in the xcsoar data folder. That way, once an aircraft is known, lookup is instant.
>
>     * a user added record in the [...]

Yes, that also to me seems the way to go.
Just two remarks:
It is clearly the way to go long as one assumes "fresh" PFLAM data, i.e. from the same day for example. In other cases, it somehow feels to me that if the database is newer than the PFLAM data, it could be the other way round. Or generally for old PFLAM data.

It is probably a very rare problem though.

The other point: In the database data there can be additional data which is not present in PFLAM, such as contact frequency. That should probably not get lost from being displayed  as a result of a PFLAM packet? But the one would have to decide how to merge...

Not many people put a frequency, but it feels like these might be the same people that take the time to configure the PFLAM config :)

Scumi, 2025-04-07:
> It is clearly the way to go long as one assumes "fresh" PFLAM data, i.e. from the same day for example. In other cases, it somehow feels to me that if the database is newer than the PFLAM data, it could be the other way round. Or generally for old PFLAM data.

Yes, I thought about that as well. But old data is old data, you can't do much if users don't update any of the databases in use. 
The question I currently think about is: One day you get an aircraft with a pilotname provided by PFLAM. The next day, you see the same aircraft again without the pilotname provided by PFLAM. Ideally the database removes the name then because it was probably a deliberate choice. The thing is, if a name is not set, you just dont get PNAME supplied by PFLAM but the way it is designed, you could wait minutes for the full information to be distributed and therefor recognize that a sentence (PNAME in this case) is missing. 

> The other point: In the database data there can be additional data which is not present in PFLAM, such as contact frequency. That should probably not get lost from being displayed as a result of a PFLAM packet? But then one would have to decide how to merge...

I think that's data I would treat as "live" data because it can and will be set live during flight i.e. while changing frequencies. I would not store that information in the database (nor METAR, AIRPT, BCST, UCST if ever implemented).

groundhogxc, 2025-04-08:
> Yes, I thought about that as well. But old data is old data, you can't do much if users don't update any of the databases in use. The 
>  [...]
>could wait minutes for the full information to be distributed and therefor recognize that a sentence (PNAME in this case) is missing.

Yeah. Not sure what the best way is.
A typical problem will be with club planes vs. private single owner planes.

> I think that's data I would treat as "live" data because it can and will be set live during flight i.e. while changing frequencies. I would not store that information in the database (nor METAR, [...]

No, what I meant was the frequency value stored in Flarmnet/OGN, which is (like home airfield) supposed to be the frequency at which the aircraft can "typically" be reached. This is not connected to the currently set frequency, and it is very rarely changed in the flarmnet database. I think this is not provided by PFLAM, so the question would be whether to look up this info from the database files separately and merge the PFLAM and Flarmnet info before diplaying.

lordfolken, 2025-04-08:
It can send the VHF frequency that is currently set on the radio. So that field would be actually useful in contrast to the db. Alas everything beyond type, registration, pilot name, comp id is locked out behind an unspecified flarm license.

----

I think we are overthinking this:
 * we simply allow you to choose either a premade db or PFLAM. 
 * the pflam is persisted in a file and is available in subsequent flights
 * if new data is available its overwritten

**I think the most important information is the registration / compid associated with the flarmid. That information gets hardly changed in any scenario.** The pilot name is only visible in the traffic details dialog.

Personally i think that the DDBs will become obsolete very fast, as this protocol is already rolled out in any flying flarm equipped aircraft.

groundhogxc, 2025-04-09:
> The VHF field from a DDB has IMO limited utility. If its the home airbase, you shouldn't be using this frequency more than 15nm away from the field (according to regulations).

I think the idea is to put the "normal" air-to-air frequency there that you are on. But very few people do it.

With the PFLAM command I just realised that this info needs to be triggered, so that needs some kind of button to broadcast. And maybe you have to pay a special license, I guess...

> I think we are overthinking this:
Yes, I think it is fine to not do too many things. Like you say, it is possible the databases will disappear except for current or future ones with special functions like weglide. For example I doubt many people will put pilot name into their flarms to be transmitted out into the landscape. But with the additional step via weglide connect it is more acceptable I guess.