XCSoar/XCSoar #1728: event to open Alternates dialog

bomilkar, 2025-03-31:
XCSoar version
--------------

7.43

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

I'm looking for an event which opens the Alternates dialog. 
There are lots of events in src/Input/InputEvents.hpp but nothing sounds as if it would open the Alternates.
The intention is to define a single key stroke in the xci file to open the Alternates. Here is the snip-it.
mode=default
type=key
data=RETURN
event=StatusMessage List Alternates
event=Alternates

The "event=Alternates" is what I'm looking for.
Is there anything like that already? (I can't believe something basic like this wouldn't already exist. Maybe I'm just too blind to find it??)

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

A friend of mine wants to open the Alternates dialog through a single key stroke.
I thought this should be easy: just add a few lines to the xci file mapping the key stroke to the event opening the Alternates.
But I failed to find such an event.

Unless it's already there, I think it should be just a few lines in
src/Input/InputEventsActions.cpp
src/Input/InputEvents.hpp
to add the feature.
I'm happy to add a PR if needed. But I want to check first.

DanD222, 2025-03-31:
Hi Ronald, I believe Alternates is mapped to the F6 key, that's how I have it mapped on my remote stick. Please check though.

bomilkar, 2025-03-31:
Thanks DanD222 for looking into this.
Can you please look into your xci file to see which event is mapped to F6 ? This would be what I'm looking for.

It won't find any Alternates if it doesn't know where it's at. That makes sense. Hence the Alternates list is empty in such a case.

Scumi, 2025-03-31:
The event is `Setup Alternates`. You can look in the default.xci - line 197 (`Data/Input/default.xci`).

closed 2025-03-31 by bomilkar