XCSoar/XCSoar #1729: Zoom and Menu Button Position and Functionality

fraktal-sb, 2025-03-31:
XCSoar versions having the problem
-------------------------------------------------

7.44

XCSoar versions not having the problem
-------------------------------------------------

7.43

System information
------------------

Unix

Steps to reproduce the behavior
-------------------------------

Change configuration to show the Zoom and Menu Buttons

Expected behavior
-----------------

Zoom and Menu Button should not hide other information

Actual behavior
---------------

The Buttons hide other elements in the map. In addition the zoom Button does not make sense in the radar page and if you press it, it switches to the map display (!?) The buttons in the full map display are in the middle of the map which is inconsistent with the other map pages.

Do you have any idea what may have caused this?
-----------------------------------------------

Bugs

Do you have an idea how to solve the issue?
-------------------------------------------

It should be checked wether the buttons hide oder control elements i.e. the climb/sink rate widget in the radar page or the north indicator in the map

My opinion: The menu button belongs to the upper right corner. You also should think about a more simple button then this big "M" i.e. a Hamburger sign which is more common and a less complicated graphic. Buttons in the lower right corner of browsers are unusual.

see the following screenshots to see the problems

[image]
[image]
[image]

fraktal-sb, 2025-06-25:
We have the option to switch on and off the zoom an menu buttons which is good to have. But if we have the buttons visible it should not destroy other information. Under certain circumstances the buttons are completly hidden. As it is for now it cannot remain.

lordfolken, 2026-10-09:
Overlay placement / radar zoom / hamburger landed earlier (#2546 and follow-ups). The remaining north-arrow clash is addressed by `af4f648c` (`MapHudLayout`: compass draws in `top_right`, inset by the overlay-button column). Closing. Please reopen if a track-up map with the menu overlay still covers the compass. Ozi-style transparent buttons stay a separate wishlist.

closed 2026-10-09 by lordfolken