XCSoar/XCSoar #1731: Google Play Store Relase

lordfolken, 2025-04-01:


andrestotorica, 2025-06-19:
Happy to help on this front on whatever's needed. I was talking to several pilots last couple of weeks and many are just switching to the forks because of this. Also, any new features or bug fixes will go invisible for most.

On a slightly different note, I think the store name change will probably hugely affect XCSoar adoption by pilots by creating a lot of confusion. I don't know much of the history of the project, but does anyone think there's anything the community can do to convince @MaxKellermann to transfer the ownership to a community managed organization account? Feel free to reach out by email if you prefer.

lordfolken, 2025-11-17:
i added you guys @andrestotorica @kobedegeest.
@dssherrill i have no visible email address from you.

Joerg-CH, 2025-11-20:
Another observation with the test version v7.44-Android~git#3819573eea:

The german translation is not up to date. 

The file XCSoar/po/de.po seems to be from one year ago, and contains still errors which are updated since long in weblate (but weblate is currently locked)
Example in de.po:
> line 4406 msgid "Top left (avoid InfoBoxes)"
> line 4407 msgstr "Unten Links (vermeide InfoBoxen)"

lordfolken, 2025-12-11:
Yes, i would love to get there. internal test is required for a new app. But the the internal testing requires 12 testers for 2 weeks. Only 9 people signed up so far.

lordfolken, 2025-12-30:
@zinuzoid The closed testing phase was now successful with 12 testers for 14 days. Can you login an press "apply for production" in the play console? Only the account owner can do that apparently. 

https://play.google.com/console/u/0/developers/8853108088361740008/app/4972532698259227287/app-dashboard

zinuzoid, 2025-12-30:
> [@zinuzoid](https://github.com/zinuzoid) The closed testing phase was now successful with 12 testers for 14 days. Can you login an press "apply for production" in the play console? Only the account [...]

@lordfolken I shared the credentials of the owner account with you via email back in March 2024. You should have full access and control using that account.
It would be great if you could try accessing it directly, but if you prefer, I’m happy to do it, just let me know.

closed 2026-01-01 by lordfolken

lordfolken, 2026-01-01:
176 should roll out soon to closed testing. This should fix 
* full screen support  
* notch 
* (edge-to-edge) display (landscape) 
* downloads.  
* Also the xcsoardatapath is now tapable to reveal the full path. 

Please report any problems.

lordfolken, 2026-01-18:
Currently still applying for open testing. They rejected the update i made on the 11th. Again due to the consent dialogs. They provided a screenshot evidence and no explenation. I appealed because it looked like the previous version.

lordfolken, 2026-02-04:
So i released 182, to internal testers. based on current master.

lordfolken, 2026-02-04:
I would appreciate if you can leave some reviews on the google play store page. That way it looks more alive.

groundhogxc, 2026-02-05:
> Swiping from edge still causes unwanted clicks or gestures (not sure if that issue was already merged tho just know someone was working on it)

Did you test build 176 or 182? In 182 I think the fix shod be included. Not 100% sure through.

kobedegeest, 2026-02-06:
I build with github workflow currently ANDROID i thought both ANDROID and ANDROIDFAT made all targets, used to do ANDROIDARCH64 but that workflow broke at some point so took the new native.ylm where i just keep the android target. And i build the apk not bundel (never tried to install a bundel).

Nothing in the logs. My guess is that is just an issue on the phone side not xcsoar. I can open new issue for it but kinda hard to debug if no one can reproduce.

groundhogxc, 2026-02-06:
But it is true - it does not work anymore. I quickly downloaded the current CI build artifact to my phone for testing (Android 12, but with only top and bottom swipe).
I can confirm the unwanted clicks are there in full screen mode.

And it somehow looks like those commits got lost from PR 2067 at some point. They were in there initially, I remember I tested it, but in the final version of PR2067 they are gone.
Likely it happened when you (@lordfolken ) made the changes that also led to the PR no longer building due to the IsSimulator() issue?

We should probably check if something else also got lost in the process? But how?

I had deleted the branch in my repository after the merge (without checking - my bad). I will recreate it, looks like I still have something in my local git repo.

Also, the commits are still in the main repo as well, as a side track to pr 2067, just no longer in any current branch/PR. So nothing is lost, thankfully.

zinuzoid, 2026-03-17:
> Our records indicate that an email request for additional information with the subject line "Information needed regarding your app submission" was sent on [Jan 26 2026 & Feb 27 2026] to the following email address(es): [[xcsoarorg@gmail.com](mailto:xcsoarorg@gmail.com) & [zinuzoid@gmail.com](mailto:zinuzoid@gmail.com)].
> 
> [@zinuzoid](https://github.com/zinuzoid) could you check?

@lordfolken I've forward you an email.

_On the side topic, let's solve the main account access issue so you're not blocked by me, will follow up in email._

lordfolken, 2026-03-18:
thats not really possible to change, as @zinuzoid is sponsoring the account.