XCSoar/XCSoar #1733: InfoBox content is magenta

bomilkar, 2025-04-04:
XCSoar versions having the problem
-------------------------------------------------

current head

XCSoar versions not having the problem
-------------------------------------------------

7.43

System information
------------------

Steps to reproduce the behavior
-------------------------------

Expected behavior
-----------------

Font color may change to indicate danger, for instance. But if there is no reason than please keep the font color to the default (which is black or white depending on the color scheme.)

Actual behavior
---------------

Numbers of several InfoBoxes including "Next altitude", "Next distance", ..  is always magenta.
What's the point? Why not keep it black? Why reduce contrast for no reason?

Do you have any idea what may have caused this?
-----------------------------------------------

It might have been introduced by Commit 07957700916598c5680067111ff0c06b8dd94616

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2025-04-04:
To make it clear what the magenta line is pointing at. Like in every other EFIS on the market.

bomilkar, 2025-04-06:
I've been looking around: I have not found a single gliding navigation system using magenta **anywhere**.
The only instrument I found is the Garmin 1000 series. (And I've heard it's an EASA requirement to show GPS-locked state and information, not VOR, or other.)

But Garmin addresses an entirely different use case.
Combining the two use cases would be like designing one pair of shoes for a whole family.

BTW: the magenta course line on the map is less than optimal for certain map color schemes.

I strongly suggest to revert this change.

Why not discuss this change in the forum so a wider audience can chime in?

Scumi, 2025-04-06:
I also see this skeptical. This is very user specific and to my knowledge, other glide computers don't do this. If at all, please make this a configurable option. I would for sure disable it because I find it distracting.

> Why not discuss this change in the forum so a wider audience can chime in?

I agree on that one.

lordfolken, 2025-04-06:
So.. lets see. 

Foreflight:

[image]

Dyon:

[image]

LXNavigation LX10:
https://lxavionics.co.uk/image/cache/catalog/LX%20Navigation/lx-10k-tsk19-500x500.png

LXNAV:
 
[image]

LK8000:

[image]

Skydemon:

[image]

Air Navigation Pro:

[image]

EnRoute: 

[image]

Scumi, 2025-04-07:
Thanks for taking the time to make this showcase. I think we talk two things here:

- magenta line and arrow
This is in my view actually fine and I agree that this is pretty much standard on many systems as shown above. Some of them (LXNAV for example) give you the option to configure the color, others do not.
- magenta info boxes
Rarely seen in the examples above and also what I remember using a few of them in the past. This is the change I personally disagree with. I like that XCSoar makes intelligent use of InfoBox colors to highlight certain states, but I don't think magenta for navigation should be part of it.

bomilkar, 2025-04-07:
Only the first 2 tools (Foreflight, Dynon) show values in magenta. All other tools, which are **designed for glider pilots** show values in high contrast font, usually black on white or white on black.

The only exception would be XCSoar:

[image]

If this should really be the new style, then we should make it configurable, like @Scumi suggested.

Scumi, 2025-04-07:
> as far as the infoboxes go it is configurable just turn of colored infoboxes and done.

As I said: "XCSoar makes intelligent use of InfoBox colors to highlight certain states". Prior to this change, there were 22 settings of SetValueColour (other than black which was used to set a default). Except for radio frequency, all of them are based on a certain state to highlight the infobox content. This change just flat changes the 14 task infoboxes to magenta by default. Yes, disabling all colors helps, but you also lose the advantages of the infoboxe state highlighting. It is not really the same thing.

> And why complain after it is implemented, the feature requised and draft pr had been open quite long plenty of time to complain then.

Yes, it is unfortunate that this slipped through. It probably did for a lot of people. It would be nice if you don't see this as a complaint, but a discussion that should have happened before the PR was merged.

bomilkar, 2025-04-07:
I fully agree with @Scumi .

Adding:
> And why complain after it is implemented, the feature requised and draft pr had been open quite long plenty of time to complain then.

It is not uncommon that an Issues is raises vs. something which had previously been implemented and a debate is started on how to improve. A given implementation is **not** the last and final word!

bomilkar, 2025-04-08:
> Then please give a suggestion as to how to make it configurable.

If you tell us what you were trying to achieve then we could name it. Lets assume "Magenta" would that name, then the Help text might be:
"If Magenta, certain InfoBoxes will have colored text.  InfoBoxes related to the next Waypoint will be Magenta. Other InfoBoxes change color to indicate above or below certain threshold."