XCSoar/XCSoar #1734: Update OpenAir Format Parser to new Spec - Opening Hours &Transponder codes in Airspaces

lordfolken, 2025-04-04:
If this gets merged there are two new improvements that we should support.
https://github.com/naviter/seeyou_file_formats/pull/9

The format has a (proposed) change for opening hours of airfields:

```
AA 2023-12-16T12:00Z/2023-12-16T13:00Z 
AA 2024-12-17T00:00Z/2024-12-17T24:00Z
AA 2024-12-17T00:00Z/NONE
AA NONE/NONE
```

#### AX: Transponder Code

*Optional* Transponder code to be used with the airspace. Must be a 4 digit code.

```
AX 7000
```

reskume, 2025-05-05:
Slightly off-topic, but it fits with the general topic. We recently released our updates parser with support for the Naviter v2 format. I'm currently updating openAIP to use the new parser. We will also switch our exports to strictly align with the Naviter v1 and v2 format. In addition that, we are discussing also providing a use-case specific "data exchange" format for use with OpenAIP airspace editor capabilities. This means that we will also change the file export naming conventions to be more clear about the formats in use. Just wanted to inform you about this to avoid possible hic-ups if you are using the OpenAIP export files. Should I open a new ticket to track this if this is relevant for you or must I not worry about XCSoar data ingestion getting interrupted due to the switch?

lordfolken, 2025-05-05:
@reskume As this concerns more the parsing of the files and less the overall file naming, I created an issue here: https://github.com/XCSoar/xcsoar-data-content/issues/442

closed 2025-06-17 by lordfolken