XCSoar/XCSoar #1736: Task declaration to IGC flarm fails via S80

PNich, 2025-04-07:
XCSoar versions having the problem
-------------------------------------------------

7.43

XCSoar versions not having the problem
-------------------------------------------------

Not checked

System information
------------------

XCSoar 7.43 Android running on SteNav platform. Connected to LxNav S80 running v 9.19 and then to Red Box LGC flarm running v 7.4. All serial speeds set to 19,200. XCSoar driver set to Lxnav with pass through device and second driver set to Flarm.

Steps to reproduce the behavior
-------------------------------

XCsoar task created then go through "declare task" process. First attempt after a system reboot or "device reconnect" is always successful and flarm reboots as expected. Any subsequent attempt to declare a task always fails.

Expected behavior
-----------------

Task should be declared correctly every time. If the S80 is removed from the system and XCSoar is configured to interface directly with flarm, then every declaration is successful.

Actual behavior
---------------

As above - 2nd and subsequent declarations fail. XCSoar need either a reboot or a "device reconnect" instruction to declare correctly, but will only do it once.

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

All serial communication between XCSoar / S80 / Flarm captured using DSO with protocol decode to analyze timing issues. Results as follows:

Sequence during 1st (successful) declaration:

1) XCSoar sends "connection direct" command immediately followed by $PFLAE command
2) About 6ms later S80 configures hardware to forward serial data to flarm - exact bit for bit buffer and not message by message
3) Because of the S80 delay in decoding "connection direct" command and reconfiguring hardware, only half of the $PFLAE command gets forwarded to the flarm
4) Flarm ignores this corrupted command
5) XCsoar sends $PFLAV command followed by the declaration sequence
6) Everything is forwarded correctly to flarm which echos back as expected and task is declared successfully 

Sequence during subsequent (unsuccessful) declarations:

1) XCSoar sends "connection direct" command immediately followed by $PFLAC,s, Pilot Name........... ie straight into declaration sequence.
2) The first part the forwarded $PFLAC command is corrupted and flarm ignores all the subsequent declaration sequence and nothing is echoed back
3)  XCSoar gives up after about 2.25 seconds and sends a command to removes the direct connection. 
4) XCSoar reports that the declaration has failed

Summary:

The declaration process does not seem to allow for the delay in the S80 reconfiguring to forward the serial stream and a corrupted message is sent to the flarm.  In the first declaration the corrupted message does not matter, but in subsequent declarations the message sequence sent by XCSoar is different and a critical message is corrupted and the declaration fails.

groundhogxc, 2025-04-07:
For me, with the LXNAV connected to a classic FLARM with passthrough, it is even worse: The FLARM declaration *always* fails, even on the first try after power on / reconnect.

Not sure about the baud rates in my case, out of the top of my head.

What is DSO? Like a Tektronix Digital Signal Oscilloscope? Not sure I can bring that to the fight in my case :)
Any other way I can obtain useful debug info for this?

PNich, 2025-04-07:
Oh dear, that doesn’t sound good!

 

Yes, DSO is Digital Storage Oscilloscope. I found you need to record about 3 seconds of data transfer (at 19,200) to capture the whole sequence, but then you need to examine it at the bit by bit level to find out what was going on.  

 

I was initially just debugging with multiple instances of a terminal emulator (Tera Term) to sniff the 4 serial lines and couldn’t understand the various corrupted messages which initially I thought was something transmitting data at the wrong baud rate. It was only when I started looking at in the time domain did I find that it was the time delay in the S80 decoding the message and reconfiguring itself that caused messages to get corrupted / missed. I suspect that the impact will be different at different baud rates which might explain your situation.

 

As a starting point it might be worth giving the S80 10ms to reconfigure itself after a “connection direct” command before sending any more data. Maybe this would then show the protocol exchange without corrupted messages?

 

If you’re looking in this area, then I also noticed that after XCSoar disables the direct connection, it still tries to send sends some commands aimed at configuring the flarm’s serial ports. Seems a bit odd when you’ve just turned off the passthrough?

groundhogxc, 2025-04-08:
> Oh dear, that doesn’t sound good!
>
>
>
> Yes, DSO is Digital Storage Oscilloscope. I found you need to record about 3 seconds of data transfer (at 19,200) to capture the whole sequence, but then you [...]

Yes, that seems very likely! It is great to finally have some initial specific lead on where the problem might be.

> As a starting point it might be worth giving the S80 10ms to reconfigure itself after a “connection direct” command before sending any more data. Maybe this would then show the protocol exchange without corrupted messages?

It seems like in the case of the S100 there is only a flag change. But I do not really understand the code in the device drivers here. I even failed to find the logic that declares to the S100 AND then to the FLARM behind it. In the code it looks like in the case of pass-through it would ONLY declare to the FLARM. But this is not the case, it does declare to the S100 too, in reality.

> If you’re looking in this area, then I also noticed that after XCSoar disables the direct connection, it still tries to send sends some commands aimed at configuring the flarm’s serial ports. Seems a bit odd when you’ve just turned off the passthrough?

Oh! That is certainly not good.

This could have to do with the fact that the passthrough mode is handled by a flag in S100, and then maybe that flag is ignored for some of the FLARM declaration procedures in the end? Or the flag is reset too early? (I am speculating here, I do not really comprehend the code right now that is doing this)

When you say Teraterm, does it mean you can monitor the communication between XCS and S100 on the bluetooth channel? Or are you purely looking between S100 and Flarm?

PNich, 2025-04-08:
Comments in blue…..

 

Yes, from what I’ve seen on the serial messages, XCS only tries to declare to the flarm via the passthrough as all the declaration sequence is after the “connection direct” command and I assume that at that point XCS thinks it’s only talking to the flarm.  It sounds like your S100 is still listening to the communication and understands the declaration. 

On my S80 there is an option “send decl to GPS port” and I leave this unchecked when using passthrough driver in XCS. If I enable this “send decl to GPS port” S80 option and still use the XCS passthrough driver it doesn’t appear to declare to the S80.  If I enable the “send decl to GPS” option and keep the XCS driver as LXNAV but with no passthrough then the declaration gets sent to the S80 OK and the S80 reports that it has successfully declared the task to Flarm. Unfortunately in this configuration XCS reports an error and that the task wasn’t declared.  Not very satisfactory.

My setup is RS232 serial link between XCS and the S80 another serial link between S80 and Flarm.  As a result I’ve been looking at 4 serial lines in total using TeraTerm.  You need some very specialised tools to eavesdrop on Bluetooth so I can’t help with that.

groundhogxc, 2025-04-08:
> Comments in blue…..

FYI: The blue from the e-mail does not show up in GitHub, so it becomes a bit of search work to find the comments :)

> It seems like in the case of the S100 there is only a flag change. But I do not really understand the code in the device drivers here. I even failed to find the logic that declares to the S100 AND [...]
 
> Yes, from what I’ve seen on the serial messages, XCS only tries to declare to the flarm via the passthrough as all the declaration sequence is after the “connection direct” command and I assume that [...]

Oh, I see. That might make more sense then, very different from what I thought.
Then this could actually also be the cause for #1737, if the initial lost commands contain the turnpoint types. Maybe.

> On my S80 there is an option “send decl to GPS port” and I leave this unchecked when using passthrough driver in XCS. If I enable this “send decl to GPS port” S80 option and still use the XCS [...]

I see. I do not remember how our "send decl to GPS port" is set. However I do observe the second behavior that you describe (successful declaration to LXNav, but with "declaration failed" message from XCSoar in the end), with passthrough enabled in XCS. I guess I will have to check now to make sure, but that is how I set it up at some point. So that would be somewhat opposite to what you describe with respect to the passthrough setting.

In my case it is bluetooth from XCS to LXNav, then serial to classic FLARM.

PNich, 2025-04-09:
That’s all understood. Yes, I noticed the B&W GitHub problem so I think I’ll stop inline comments.

 

Right now I have the setup on the bench beside me as I’m mid panel upgrade. I can investigate fairly easily for the next week or so before the panel goes back in the glider.

 

I think the root cause of my problem is the lost commands and the fact that for some reason XCS send a different sequence on 1st and subsequent declaration attempts. On the first attempt the lost command doesn’t matter.

 

I’m not sure the lost command relates to #1737.  When I get a chance I’ll search through a whole successful sequence to see if XCS does try to send turnpoint types.  Which bit of documentation defines the protocol for sending declarations and turnpoint types?

groundhogxc, 2025-04-09:
The specifications for LXNAV communications are linked in #1248

Looking at that, you are probably right. The two problems seem likely to be unrelated, as the turnpoint type and position are transmitted in the same line, with one line per TP. So this is a separate problem.

But your setup would still be useful to see what is actually sent exactly to the LXNAV in terms of the declaration file content, and to the FLARM behind it, for the two declaration modes.

groundhogxc, 2025-04-09:
By the way, can you download igc files from your vario? And does it work for large IGC files (long flights, 5 hours or more)?
It does not work for me, only for short flights.
It might be a bluetooth-specific issue like #1322, but the symptoms are different from that issue, so would be a nice way to find out.

PNich, 2025-04-10:
I’ll have a look at the LXNAV spec when I get the chance and see what XCS sends when declaring to S80 and what gets passed onto flarm.

 

Is there a similar spec for declaring to flarm?

 

My S80 has never been in a glider so can’t check out IGC file download

groundhogxc, 2025-04-10:
You can find the most recent version of the data port specification at
[https://www.flarm.com/en/support/downloads/](https://www.flarm.com/en/support/downloads/)

It is in the bottom half, "DATA PORT INTERFACE CONTROL DOCUMENT (ICD)"
Currently it is April 2024 version.

PNich, 2025-04-12:
Putting a 10ms sleep into that function could be a usable solution and could be used to test the theory, but the actual required period would depend on the serial transmission speed.

 

After requesting a direct connection, could XCS send a number of simple non-critical commands until it gets a valid response and knows that it’s talking to the flarm correctly and can continue with the flarm protocol?

 

Is this what XCS does on it’s initial (successful) attempt to declare to flarm as the sequence includes extra commands just after the connection direct?

groundhogxc, 2025-04-14:
As a side remark - it seems now that what I obesrve, that the Task is (almost) correctly declared to the LXNav even though there is a FLARM behind it, is actually a bug.
We would expect ONLY the FLARM to be programmed in this scenario, it is just that something goes wrong with the pass through in my case and thus the commands are received by LXNav, not the FLARM.

If this is by design - should we not change that to make XCSoar attempt to declare to *all* "declarable" devices when declaring the task? Or was the current logic (declare to only one of the two devices) important for some reason? I could not really find anything historic about this.

One big problem I think I see would be if there is a non-igc FLARM behind an LXNav - as far as I understand the current logic only checks for pass through, not for "IGC-ness" of the FLARM. As a result, it will ignore the IGC LXNav, try a (doomed to fail) declaration to the non-igc FLARM, so in the end nothing is declared.

Of course, with a non-igc FLARM you can just disable pass-through to avoid this. But then you lose the ability to communicate with the flarm for all other purposes, including the new configuration functions etc  that came in recently.

groundhogxc, 2025-04-14:
Actually, with a quick look into the FLARM protocol file:

It seems that FLARM does not use the same commands for task upload - instead of uploading a config file, the tasks are programmed point by point with NEWTASK and ADDWP commands.

For *reading* the task from the FLARM, a file of this type is returned, however (The file format is described in the IGC GNSS dataspec definition of the IGC).

So unless there is an undocumented "write declaration file" serial command in the FLARM compatible with the LXNav command, or unless I am overlooking something, the LXNav declaration must come from LXNav code in my case, not from misdirected FLARM commands ending up int he LXNav.

This continues to be mysterious...

PNich, 2025-04-14:
In my case, serial between XCS and LXNav (S80) then serial between LXnav and classic flarm, with passthrough enabled the only communication instigated by XCS after the user tries to declare a task uses the Flarm protocol sequence. No idea how LXNav gets hold of it.

 

On the missing turnpoint issue (#1737), in my setup XCS only send the locations and not types.

Scumi, 2025-04-14:
> One big problem I think I see would be if there is a non-igc FLARM behind an LXNav - as far as I understand the current logic only checks for pass through, not for "IGC-ness" of the FLARM. As a [...]

We could check for IGC capability by extending the FLARM/Hardware.hpp class we introduced lately. That would not be a big issue.

PNich, 2025-04-15:
This issue now seems to involve two bugs:

1) Second and subsequent declarations to flarm via S80 fail due to lost/corrupted messages when setting up passthrough in S80
2) When just using LXNAV driver without passthrough, all declarations to LXNav (S80) reported as failed by XCS, but are reported as declared OK by S80. Reason unknown as traffic captured (see #1737) looks OK 

Issue #1737 looks like an enhancement to me as XCS makes no attempt to transfer turnpoint types.

groundhogxc, 2025-04-15:
That seems the right way of describing it. Personally though I have never succeeded in doing *any* declaration through the LXNav to (classic) FLARM. But I will now have to try it again doing it specifically after a fresh start of both LXNav and XCS to see if it works.

As for No. 1, Adding some 10ms or 50ms pause seems like a good thing to try - maybe I can do it one of my next visits to the airfield (however I can't check the results with hardware logging on the serial line, only see whether it works or not),

As for No. 2, looking at the source code in LX/Declare.cpp, one of two things must happen
`https://github.com/XCSoar/XCSoar/blob/c916f292cd5c628fd308e534562b6182e72eb892/src/Device/Driver/LX/Declare.cpp#L139`

Either it fails on the "LX::ExpectACK(port, env);" after sending the declaration, or it fails after the last "LX::ExpectACK(port, env);", after sending the "Contest Type" string - which seems to never happen, or, even worse, it is the weird "s1?" line in your serial log in #1737.
At any rate, both of these would result in a "failed declaration" return value, but after having sent the entire declaration to the LXNav.

I suspect it is the second, as ther is nothing coming back from LXNav in the log - unless the "LX::ExpectACK(port, env);" function has a bug already.

Whether bug or improvement on #1737, question of taste, I guess. For someone who sees TP type as "part of task", it is a bug, for someone who sees this as extra info, it is an enhancement... Does not really matter.

groundhogxc, 2025-04-24:
I was just able to try it out on our device (LXNav with pass-through to classic FLARM, both on most recent firmware).
With pass-through enabled, and no additional delays added I get the following:

- Declaration to FLARM always ends in failure message, and I never hear a FLARM beep (which should confirm successful declaration as far as I understand)
- On the first time after restarting XCSoar, I get (maybe not everytime, not sure), I briefly get one or two red LEDs on the FLARM display.
- In subsequent attempts the LEDs doe not blink red, the rest is the same

If I add 10ms as discussed I get the same behavior, but:
- in addition the LXNav shows a "FLARM Error" in the end. 
- Also, the progress bar in XCSoar seems to go up to around 20% - not sure if this is always the case though, I tried many combinations.

For longer delays, and with adding more delay commands also before and after the mode change, I get the same result.

So overall, the additional delay does seem to make it better, but it does not really solve the issue.

I did set the device to "debug mode" to log the communication, but unfortunately even the debug mode seems to not log the declaration attempt, only the regular NMEA data. So unfortunately I have no idea of what was going on on the ports. I need to build something to be able to log this.

Without pass-through, there seems to be no attempt to declare to the FLARM, but rather it declares to the LXNav, and this ends successfully - except that there is an error message in XCSOar at the end which falsely claims the declaration was not successful.

groundhogxc, 2026-02-15:
> Tested with an LXNAV S series vario + external PowerFLARM: both LXNAV and FLARM declarations now succeed in a single operation.

That sounds glorious!! 🚀🚀

PNich, 2026-02-23:
> I see, thanks. It should not be necessary to reboot in fact, but sometimes it is. It is more of a "try everything" approach. Another is to try end clear the cache (somwhere in the Android memory settings). Should not be necessary, but...
>
> The downloaded version probably has a different app signing key now, or it is a debug version, not sure. Either way it would cause this effect if there is anything left from the prior install.
>
> The real problem being of course that there should be nothing left of the previous install, if you uninstalled with data removal.
>
> Was 7.43 the first version of XCSoar you had on the device?

Yep, a reboot fixed it and I’ve been playing with it for a while. Thanks for the suggestion.

I’ve been using XCS since 5.xx on kobos, but this is my first Android tablet and it came with 7.43 installed.

The good news is that my task declaration problem has been fixed. It seems completely reliable.

I’ve noticed a couple of differences in the look using the same .prf file….

1)      In 7.43 “dark mode” only impacts the info boxes on the main screens. In 7.44 “dark mode” impacts the info boxes and the menu pages. I’d prefer them to stay separate.

2)      In 7.43 the text at the top of the turn points page wraps to fit in the tab boxes (very nice). In 7.44 the same size text overlaps the boxes (not so nice)