XCSoar/XCSoar #1737: LXNav S8/80/10/100 varios: Task declaration start finish/point types incorrect

groundhogxc, 2025-04-07:
XCSoar versions having the problem
-------------------------------------------------

System information
------------------

Seen with Android version

Steps to reproduce the behavior
-------------------------------

* create task in XCSoar or e.g.  download from weglide 
* use default turnpoint types: A cylinder start point and finish point, and symmetric sectors for the turnpoints
* declare task to LXNav device

Expected behavior
-----------------

Turnpoint types should be the same in LXNav as in XCSoar

Actual behavior
---------------

The task shows up  in the LXNav device with symmetric sector type turnpoints for ALL task points, including start and finish

Do you have any idea what may have caused this?
-----------------------------------------------

It looks like somewhere in the declaration process the turn point type info is lost and something (driver or device) then reverts to sector types as a default, maybe?

PNich, 2025-04-13:
See attached - Output from XCS RS232 traffic captured and response from LXNAV.

XCS sends turnpoint positions OK but does not send turnpoint types.

LXNAV reports task declared, XCS reports declaration failed.

[XCS decl.txt](https://github.com/user-attachments/files/19725043/XCS.decl.txt)

[LXNAV resp.txt](https://github.com/user-attachments/files/19725044/LXNAV.resp.txt)

groundhogxc, 2025-04-14:
Looking at the dataport specification file V 1.04 (Linked in issue #1248), it seems to be supported by the protocol at least- their example declaration file is:

`HFPLTPILOT:Uros Krasovic
HFCM2CREW2:
HFGTYGLIDERTYPE:
HFGIDGLIDERID:
HFCIDCOMPETITIONID:
HFCCLCOMPETITIONCLASS:
C050221133458050221020502
C0000000N00000000E
C4846667N01015883EAALEN HEIDENHEI::585.00000
C4711500N00942100ED21CH NENZING::572.00000
C4748667N01346550EEBENSEE EBXTRAUN::420.00000
C4744110N01133501EWACKERSBERG::831.00000
C0000000N00000000E
LLXVOZ=-1,Style=2,R1=3000m,A1=45.0,R2=0m,A2=0.0,A12=193.6,Maxa=0m,Line=1,Autonext=1,Lat=4846.667N,Lon=01015.883E,Near=0
LLXVOZ=0,Style=1,R1=5100m,A1=180.0,R2=0m,A2=0.0,A12=44.7,Maxa=0m,Autonext=1,AAT=1,Lat=4711.500N,Lon=00942.100E,Near=0
LLXVOZ=1,Style=1,R1=10000m,A1=89.5,R2=0m,A2=0.0,A12=261.9,Maxa=0m,Line=1,Autonext=1,Lat=4748.667N,Lon=01346.550E,Near=0
LLXVOZ=2,Style=3,R1=500m,A1=180.0,R2=0m,A2=0.0,A12=87.9,Maxa=0m,Autonext=1,Lat=4744.100N,Lon=01133.100E,Near=1
LLXVTSK,StartOnEntry=false,Short=false,Near=true`

Of course, it could still be that the S80 / S100 then ignores the additional information in the LLXVOZ lines.

Maybe someone can check what happens when declaring from an  LX9000 to an igc LX100 (If someone here is willing to admit to running an LX9000 ;-) )

lordfolken, 2025-04-15:
[LXNAV_DataPort_Specification1.04.pdf](https://github.com/user-attachments/files/19754231/LXNAV_DataPort_Specification1.04.pdf)