XCSoar/XCSoar #1740: Turn Back Marker - new map element

DavisChappins, 2025-04-12:
A new feature called the "Turn Back Marker" a green triangle on the track line that indicates where along the track your arrival altitude will equal 0 if you maintained that track (and had still air). You can use this information to indicate where you will need to "turn back" to your goto waypoint.

The assumptions that go into arrival altitude go into the TBM calculation (MC setting, etc).

In use - 

Pic below shows the TBM as a green triangle. Leg A can be flown until arrival alt (2243) is equal to 0 - at that point you'll have just enough altitude to fly leg B (0 arrival altitude still).

[image]

Pic below shows another example of overflying your goto waypoint maybe to get a few more OLC points. TBM is again circled and indicates where on your current track (leg A) your arrival alt (2543) would equal 0 and you would have to turn back and fly leg B to arrive at Valence.

[image]

TBM is only shown when arrival alt is > 0.
The idea is to visually indicate on the map how far you'll be able to deviate or overshoot and still make it back to your goto waypoint.
When i get a configuration on/off page working i'll post some code.

dssherrill, 2025-04-12:
Very nice!  Could you make the target arrival altitude be the configured
safety height instead of zero?

   …david

groundhogxc, 2025-04-24:
I was thinking a bit about these calculations, so I am putting this here so it is not lost, in particular with respect to the wind calculation for the turnback marker.

One general point: If you consider all possible turn back points in all directions which maximize the distance flown from the current point to the destination point in two legs, then these lie on an ellipse. This can be used to make the calculation more general.

(In fact, I think it would be quite useful to also show a section of that ellipse around the green triangle, for example +/- 10 degrees around the triangle. This allows to plan better for optimizing where to fly, in particular in combination with the FAI triangle regions that XCSoar already can show - so one can fly the path to optimize triangle easier.)

Without wind:
- The ellipse is centered on the point in the middle between the current position and the destination point.
- These two points also lie on the major axis of the ellipse
- The major and minor axis radii are such that the current position and the destination point are the two focal points of the ellipse
- The current turnback point marker position  (TBM) is the intersection point of the current course line in forward direction with the ellipse

With wind:
- The set of all possible turnback points is still an ellipse, and the center point is still the center between the current position and destination (this can be shown quickly with a relatively simple mathematical proof). 
- The major and minor axes however get changed due to the presence of wind. This is the harder part. The easiest way to figure out this ellipse could be to calculate four points of it, as four points determine all parameters of an ellipse
- Two points of the ellipse along the connection line between current position and destination are fairly easy by just calculating the glide slopes along that line in both direction.

- A completely different way to do the calculation in the presence of wind is to do it by displacing the destination point by the wind drift, and then doing the calculation as one would do it without wind. 
- This is easy because the total time of flight to destination is fixed by the known sink rate for the current MC value, and the known wind vector.
- The current course line needs to be modified too, by correcting the track direction (vector) with the wind vector
- Using this modified course, the MC STF TAS (instead of ground speed) and the no-wind algorithm, the turning point is calculated as well as the time needed to reach that point
- This turning point is then moved by the inverse wind drift (wind vector times time to that point) to get the position in real space, which should then coincide with the real course line

None of these methods, by the way,  fulfill my earlier claim that ideally the same code be used that is already used for arrival height and reach calculations... But the current pull request also does not do that so at least it is not worse :-)
Using the existing code would be easily possible by an iterative approach to solve for the point, but that would require a bit more CPU power, as one would need to run that code ten or more times to determine the TBM with reasonable precision.

DavisChappins, 2025-04-26:
yea so im going to employ the FSIS principle (First Make It Simple) I don't think ive seen that before so im claiming it as mine. If you want to get super complicated with this go for it later but im going to keep it straightforward.

groundhogxc, 2025-05-02:
> yea so im going to employ the FSIS principle (First Make It Simple) I don't think ive seen that before so im claiming it as mine. If you want to get super complicated with this go for it later but im going to keep it straightforward.

I totally agree with the principle as such. Also, I only wrote the "complicated stuff" to have it available here to be found whenever someone wants to extend something such as add a glide ellips segment or sth. like that.

But I do see a big problem with the current approach in that it displays a calculated reach value which is not consistent with the other reach calculations. So, if I were the user, I would assume that it is consistent with the way that direct flight is calculated (meaning with MC taken into account but also wind and safety altitude for that waypoint).

If the calculation is completely wind-free, it would be very problematic - in that case, if you have a headwind component after the turn at the turn back point, you would fly to the TBM and then find that the altitude is not enough to reach the destination point even according to XCSoar's own reach/arrival height calculator. In my personal opinion such information should not be displayed.

I am not sure if that would happen with the current implementation - I think there might be an influence of the wind value due to the way the glide rate is calculated. It looks like this is done by obtaining the glide rate for the direct path, and I think this takes the wind into account for that virtual flight path. 

This would mean that as you proceed, the point would move, and as you pass the destination point, your TBM point would move significantly. If you fly directly overhead the destination point, it would even jump to a new value in that moment.