XCSoar/XCSoar #1746: Airspace entry warnings on invisible airspaces

fraktal-sb, 2025-04-27:
XCSoar versions having the problem
-------------------------------------------------

7.44

XCSoar versions not having the problem
-------------------------------------------------

all other versions before the AY (airspace type) was introduced

System information
------------------

UNIX

Steps to reproduce the behavior
-------------------------------

use airspace filter (uncomment warning and display to airspace E)

use France airspace and the following test data withe the replay function

[Test24.txt](https://github.com/user-attachments/files/19930395/Test24.txt)

the plane enters airspace class E/TMA Sarre

[image]

Expected behavior
-----------------

the uncommented airspace (class) "E" should not issue a warning

Actual behavior
---------------

visible and audio warning is issued. The reason is that the AY record where "display and warning" is active

Do you have any idea what may have caused this?
-----------------------------------------------

Bug

Do you have an idea how to solve the issue?
-------------------------------------------

Disable TMA warnings is not the solution; they exist also in combination with airspace class D

fraktal-sb, 2025-04-28:
The above solution solves my local problem, but after further analysis of the French airspace data, I found combinations of airspace classes E and G together with the TMZ airspace type. Normally, I don't want to see E or G, but rather TMZs. So, the only solution would be to configure combinations of airspace classes and types. However, I think that would be difficult to implement.

bomilkar, 2025-05-05:
> airspace classes E and G together with the TMZ airspace type

I came across this issue, too.
This airspace doesn't display or warn if Class E is switched off:
```
AC E
AY TMZ
AN TMZ-EDDN XPDR A4404
```
What XCSoar should do is to warn and display if TMZ is switched on in the airspace filter, regardless of the airspace class.

I'm experimenting with it and found this to work:
```
AirspaceClass GetClassOrType() const noexcept {
    if (GetType() == AirspaceClass::TMZ) return (AirspaceClass::TMZ);
    return GetClass() == AirspaceClass::UNCLASSIFIED ? GetType() : GetClass();
  }
```
This is in src/Engine/Airspace/AbstractAirspace.hpp
But I'm not sure about the side-effects of the change. And what about RMZ or other Types? Unfortunately every country seems to have it's own flavor of characterizing airspaces. However, we should follow OpenAIP's notation!

PR #1751 (2025-05-19): Airspace/AirspaceVisibilty.cpp: Also check type
With the extended openair format airspace have at class and type We need to check both of these to know weather or not to draw the airspace

Renamed the boolean to reflect that it now checks both.

closes #1746
commit 8ad0596f (2025-05-19): Airspace/AirspaceVisibilty.cpp: Also check type

With the extended openair format airspace have at class and type
We need to check both of these to know weather or not to draw the airspace

Renamed the boolean to reflect that it now checks both.

closes #1746
closed 2025-05-19 by lordfolken