XCSoar/XCSoar #1752: Changing symbol from STF to Vario

Blaubart, 2025-05-05:
If I correctly find out these two commands are changing the mode in XCSoar:

This command "info.switch_state.flight_mode = SwitchState::FlightMode::CRUISE;" should put XCSoar into STF mode and display the airplane symbol in the bottom right.

This command "info.switch_state.flight_mode = SwitchState::FlightMode::CIRCLING;" should put XCSoar into Vario mode and the airplane symbol in the bottom right should be replaced by a circle symbol.

The FreeVario driver calls these functions but the symbol does not change on the workbench. Now the question is whether I did something wrong or whether XCSoar actually has to fly to perform the switching.

lordfolken, 2025-05-06:
These are the correct functions and are in use by a variety of drivers. Personally i'm not sure if at the moment they work as intended. 

I would suggest to write a Test for your Driver.  An example is the v7Test. That way you can check its behavior even without hardware attached. 

https://github.com/XCSoar/XCSoar/blob/1834e5968b69bcba064b37ac2b01e9a765eb6233/test/src/TestDriver.cpp#L1188-L1219

lordfolken, 2025-05-06:
So reading the source code, and using an LX Vario i could reproduce what you want, but you need to set "Flap forces cruise" in the GlideComputer Settings.

XCNav, 2025-05-07:
Reading the driver for XCVARIO,  the flightmode should also be changed when switching XCVARIO from Thermaling to S2F. This is also not working here.

iltis42, 2025-05-08:
Confirm, neither with Borget B50, nor with XCVario driver this works ATM e.g. in recent 7.43 XCSoar

lordfolken, 2025-05-08:
@XCNav @iltis42 
So i noticed that we have no tests for the xcvario driver:
https://github.com/XCSoar/XCSoar/commit/84fb5b4855fc49bde1c876943e5af7abcc70b8f6

Could one of you finish this and submit it?

I believe the Switch state check in the XCVario.cpp is broken (the tests also indicate this.):
https://github.com/XCSoar/XCSoar/commit/efc4f38f007854663559af405b962784a7e24fc0
Could you guys confirm this fix to work?

dspreitz, 2026-10-07:
Summary of what I found, with a PR to follow:

- The switch is read the right way round. The XCVario firmware sends
  `!cruise` (0 in cruise, 1 in climb) since iltis42/XCVario@7a2d934 for
  #812, where Mike Borgelt confirmed that the Borgelt documentation has
  the bit inverted. XCSoar's B50 and XCVario drivers read it that way.
  @iltis42 the protocol comments in `main/Protocols.cpp` (both `$PXCV` and
  `$PBB50`) still say "G = 0 in climb, 1 in cruise"; it might be worth
  correcting them to match what the code sends.
- The setting only takes effect while XCSoar detects flight
  (`CirclingComputer::Turning()` returns on the ground), so on the
  workbench the mode never changes, with or without "Flap forces cruise".
  That explains the test in the comment of 2025-06-04.
- The setting's name and help only talk about the Vega flap, although it
  applies to every vario that reports the switch, and it is off by
  default.

#3293 renames it to "Vario switch sets flight mode", rewrites the help
and the manual entry (mentioning that it only works in flight), corrects
the inverted comments in our drivers, and adds a test for the B50 cruise
value. Whether the switch should also work on the ground for the
pre-flight check is a separate question: #3292.

closed 2026-10-14 by lordfolken