XCSoar/XCSoar #1753: TMZ generates Warning but not Display although both are on

bomilkar, 2025-05-05:
XCSoar versions having the problem
-------------------------------------------------
Current Head (7.44 git:1834e5968b)
in conjunction with OpenAIR Extended airspace file

XCSoar versions not having the problem
-------------------------------------------------

System information
------------------

Steps to reproduce the behavior
-------------------------------

Use the attached airspace file (from OpenAIP in OpenAir-Extended format)
Set Config->System->Map_Display->Airspace->Filter
   Class E off
   TMZ Warn Display
Set QNH to 1000 hPa
Replay the attached Testcase nmea File.

About 2 Minutes into the replay you see "Near Airspace TMZ-EDDN" warnings.
But you don't see TMZ-EDDN on the map. 

Expected behavior
-----------------

If TMZ Warn Display are both on, it should warn AND display.

Actual behavior
---------------

Do you have any idea what may have caused this?
-----------------------------------------------
It is probably related to OpenAir-Extended, which introduced Airspace Type in addition to Class.

[de_asp_extended.txt](https://github.com/user-attachments/files/20039613/de_asp_extended.txt)

[Testcase_TMZ_2025-04-30.nmea.txt](https://github.com/user-attachments/files/20039654/Testcase_TMZ_2025-04-30.nmea.txt)

Do you have an idea how to solve the issue?
-------------------------------------------

bomilkar, 2025-05-05:
This is related to PR #1751 as requested

Joerg-CH, 2025-05-05:
I could not repeat your experience. (v7.44 from april,17)
With your airspace file and your .nmea file, I see the TMZ, as long as the warning is displayed. 
To have the warning longer displayed, I increased the warning time from 45" to 2 min. When I then stop the replay, I can also pan to the then visible airspace and check its details in 'Map elements at this  location' (just before the warning, with the TMZ hidden, there is nothing in 'Map elements...').

[image]

Perhaps you didn't see the TMZ, because it overlaps the CTR Ansbach.

bomilkar, 2025-05-05:
I think you do see the issue: when the warning is NOT active the TMZ is **invisible** on the map.
Despite the airspace filter is set to "Warn **Display**".

bomilkar, 2025-05-05:
> warnings work as intended
> 
> displaying is skipped if class is set to not show for code analisis see [#1751](https://github.com/XCSoar/XCSoar/pull/1751) (this is the part that is not work as intended)
> 
> Please stop making this about warnings all your explaining confirms warnings are working as expected.

This is about the display! Warning seem to work.
You suggest to turn on Display for Class E in order to make TMZ visible? That is not the expected behavior. I would expect the TMZ to be visible, based on the display attribute for TMZ (not for Class E).
OpenAIP confirms in OpenAir-Extended format the Class attributes are limited to the ICAO classes A through G and SUA. "Class TMZ" does not exist in the Extended format. See https://docs.openaip.net/#/Airspaces/get_airspaces

closed 2025-05-19 by lordfolken