XCSoar/XCSoar #1756: XCVario: S2F is not passed to XCSoar

lordfolken, 2025-05-08:
> @XCNav @iltis42 
> So i noticed that we have no tests for the xcvario driver:
> https://github.com/XCSoar/XCSoar/commit/84fb5b4855fc49bde1c876943e5af7abcc70b8f6
> 
> Could one of you finish this and submit it?
> 
> I believe the Switch state check in the XCVario.cpp is broken (the tests also indicate this.):
> https://github.com/XCSoar/XCSoar/commit/efc4f38f007854663559af405b962784a7e24fc0
> Could you guys confirm this fix to work?
> 
> 
> 
>  

 _Originally posted by @lordfolken in [#1752](https://github.com/XCSoar/XCSoar/issues/1752#issuecomment-2864374885)_

lordfolken, 2025-05-08:
Better fix fa3bf570bd35852750f9ec64b40e06027bbb454e

iltis42, 2025-05-09:
The line 65, taken 1:1 from Borgelt documentation is wrong, Mike Borgelt confirmed that some time ago. Its exactly the same in the Borgelt Driver. In the file history you should be able to See the discussion. We can make a final Note in the code and close this.

iltis42, 2025-05-26:
Latest 7.44 testing on my mobile, XCSoar with XCVario driver connected, added some pressure to fake a takeoff that took place. 'Flight Time' Info-Box counts, and i see the cruise bit toggling in the XCVario protocol NMEA stream $PXCV, the first 4 sentences its zero, then one:
[image]
but the mode doesn't change between cruise and vario in here. What are we missing?

lordfolken, 2025-05-27:
You also need a valid gps signal that supplies the time. (there is a reason)

So this is another feature that has mutated over time. 

The initial idea was to force xcsoar into cruise as soon as you set the flaps to negative. This was a feature of the Vega Vario/Altair computer.

This feature also  makes sense if you want to interlock a vario that has S2F in auto mode. I.e. for example locked to speed or G-force. So when the vario detects circling, xcsoar will start its calculation for circling and thermalling. 

However if you wire a manual s2f switch to the vario, this is where things fall apart. 

When coming up to a (suspected) thermal, i switch the vario and xcsoar to vario mode, but stay at cruise speed until i actually feel the rising air and start to circle. 

If the integrator and thermal average calculation already starts at that point in time upsetting the average thermal strength, thermal time and other calculations. 

There should be another option to lock this to eventUserDisplayModeForce defined in src/Input/InputEventActions.cpp which sets DisplayMode::Circling

Sundown3867, 2025-06-27:
Iltis42 - Just out of curiosity what values are set for Config-Glide Computer -Glide Computer Cruise/Circling and Circling/Cruise when you encounter delays in mode change?

lordfolken, 2026-10-05:
The protocol comment was corrected in 32b2c26 (7.44). Switching works in flight once XCSoar thinks you are flying (needs GPS time). Confirmed with XCVario and FreeVario. Closing.

closed 2026-10-05 by lordfolken