XCSoar/XCSoar #1758: Switching Transponder off will crash XCSoar

Joerg-CH, 2025-05-12:
**XCSoar versions having the problem**
v7.44

**XCSoar versions not having the problem**
v7.43 - but I am not absolutely sure

**System information**
XCSoar v7.44 (build made 17.04.2025, with PRs 1703 and 1708 included) running on Android-11
Transponder VT-01
Air Control Display (firmware ACD57_VER_0_45.air) as frontend for transponder (gets power from transponder, and has data connection via RS232 to device with XCSoar)
Device configured with driver Air Control Display / 4800 baud
Sync to/from device enabled

**Steps to reproduce the behavior**
1. Transponder via Air Control Display (ACD) connected to XCSoar
2. Switch transponder on,  SBY
3. Switch transponder off
==> XCSoar crashes with xcsoar.log ending in last line without a time stamp:
_src/ActionInterface.cpp:366: void ActionInterface::SetTransponderCode(TransponderCode, TransponderMode, bool): assertion "mode.IsDefined()" failed_
4. Switch transponder on
5. Restart XCSoar
==> After 2-3 seconds, XCSoar crashes again

**Expected behavior**
Switching the transponder on and off should have no impact on XCSoar running

**Actual behavior**
XCSoar crashes when transponder in use / switched off.

**Do you have any idea what may have caused this?**
Probably originated by PRs #1703 and #1708. But only the (new) assertion "mode.IsDefined()" fails, not the (old) "code.IsDefined()"

**Do you have an idea how to solve the issue?**
No, but I found a workaround:
Disable "Sync from device" in the driver, but keep "Sync to device" enabled to send GPS-data from FLARM to the transponder. The data stream is checked on the ACD to be present.

lordfolken, 2025-05-13:
Please dont fly with debug builds. Asserts are debug statements and will trigger an exit.

lordfolken, 2025-05-13:
However it is a bug and shouldn't happen.

Scumi, 2025-05-13:
I'll look into it.

Joerg-CH, 2025-08-13:
Problem solved! 
To test the function to set the XPDR code directly from the (OpenAIP 2.0) airspace map (commit 9988c2c) , I compiled a build with DEBUG: n, and tested it in flight. 
The function works well, and there was no more crash of XCSoar.

Apparently, the crash was triggered by DEBUG: y.

Scumi, 2025-08-13:
Yes, this is what @lordfolken said. It will only cause issues in debug builds, but those shouldn't be used for flying anyway. However, there is still an underlying bug that needs fixing because it might cause issues later. I will probably take care of that in winter, when my instruments are on my desk.

lordfolken, 2025-08-13:
Yes, i'd like to reopen the bug. Because the assert is a safety fuse. There might be erronous data passed now in the function, that might lead to unpredictable results.

Joerg-CH, 2025-09-21:
I made a test in my glider with a debug build based on the solution coderabitai suggested. Switching the XPDR on and off did not result in an error anymore.