XCSoar/XCSoar #1759: Auto-MacCready increases to 6.7 m/s while average climb rate is lower

StdLibelle, 2025-05-13:
XCSoar versions having the problem
-------------------------------------------------

 V7.43 (on Android)

XCSoar versions not having the problem   
-------------------------------------------------

 Unknown, the configuration was different when I used previous versions

System information
------------------

 SteFly NAV (Android), vario information comes from XCVario (version 24.0930) (PowerFLARM information is routed by XCVario to XCSoar)

Steps to reproduce the behavior
-------------------------------

 Several times reduced MC value manualy, whem thermaling Auto MC value increased even when I didn't expect it (average climbrate lower than value), up to 6.7 m/s (with MC speed = max cruise speed in profile).
Please note that May 6, 2024 someone else raised a quite similar issue #1421, but unlike #1421, in my case the valute didn't jump from 2.3. I haven't noticed the jump, but I have had an MC value of 2.5 m/s on the display.

Expected behavior
-----------------

 Realistic Auto MC value, reflecting the average climb rate.

Actual behavior
---------------

 Value increased when the average climb rate was lower than the Auto MC value, and it ended at 6.7 m/s which was never actually achieved.

Do you have any idea what may have caused this?
-----------------------------------------------

 It didn't happen last year, with pre 7.40 versions. However, last year the XCVario version was 24.0410 and data from the PowerFLARM went directly to the SteFly NAV.

Do you have an idea how to solve the issue?
-------------------------------------------

 I;m sorry, but no.

lordfolken, 2025-05-14:
This sounds like final glide mode. If you are on a task and achieve final glide the vario will be configured to an MC value that in S2F mode that will get you to your final waypoint at maximum velocity (limited by max cruise speed) down to the minium saftey margin that you have set. 

To activate:
In glide-computer -> Glide Computer -> Auto MC mode: Final Glide or Both
In glide-computer -> saftey factors: Set arrival height
In Look -> InfoBox Sets: "Use final glide mode"  on. 
In config -> plane -> Max Cruise Speed

StdLibelle, 2025-05-14:
OK, that looks very plausible.Thinking back about the situation of May 11, I had finished the task but stayed aloft, quite high for the distance to the finishing point.I will check next time, but I think you are right.Thank you,Regards,JanVan: Philipp Wollschlegel ***@***.***>Verstuurd: woensdag, mei 14, 2025 9:56:08 p.m.Aan: XCSoar/XCSoar ***@***.***>CC: StdLibelle ***@***.***>; Author ***@***.***>Onderwerp: Re: [XCSoar/XCSoar] Auto-MacCready increases to 6.7 m/s while average climb rate is lower (Issue #1759)
lordfolken left a comment (XCSoar/XCSoar#1759)This sounds like final glide mode. If you are on a task and achieve final glide the vario will be configured to an MC value that in S2F mode that will get you to your final waypoint at maximum velocity (limited by max cruise speed) down to the minium saftey margin that you have set.To activate:
In glide-computer -> Glide Computer -> Auto MC mode: Final Glide or Both
In glide-computer -> saftey factors: Set arrival height
In Look -> InfoBox Sets: "Use final glide mode"  on.
In config -> plane -> Max Cruise Speed

lordfolken, 2025-05-15:
This needs to be improved in the manual. 
* There should be a popup message that says that now final glide is activated. 
* Its also unclear that all the four settings need to be set for it to function as intended

Sundown3867, 2025-05-16:
'Maybe we could add a third icon in the bottom right now we have circling or cruise could add final glide symbol as well.' 

It is already implemented and working fine

StdLibelle, 2025-06-02:
I could reproduce this and it happens indeed when in final glide mode at (far) too high altitude.

lordfolken, 2025-12-28:
there is now a popup for final glide

closed 2025-12-28 by lordfolken