XCSoar/XCSoar #1764: thermal assistant is always visible

Blaubart, 2025-05-18:
self compiled XCSoar 7.43:

The thermal assistant remains in the foreground throughout the flight and obscures important parts of the map. It should only be visible when actually circling.

lordfolken, 2025-05-18:
Thats generally not the case. if XCSoar isnt detecting circling, the Thermal assistant gauge is hidden.

https://github.com/XCSoar/XCSoar/blob/b4205f57bb8e870240e42345d953f3e861755c5e/src/MainWindow.cpp#L809-L811

can zou describe zour setup, provide a nmea log? potentiallz zour profile?

Blaubart, 2025-05-18:
Thanks für your fast answer! I will check again and send you more information.

Blaubart, 2025-05-21:
I think I've figured out where the problem is. My OV with Cubieboard is generally not performing well at the moment. That's probably why the screen wasn't refreshing correctly, and the thermal assistant remained. The reason for the poor performance is the map that I regenerated with the map generator at XCSoar.org. The map is 590MB in size!! The same section has always been about 100MB in recent years. After I loaded last year's map onto the OV, the device is running normally again.

lordfolken, 2025-05-22:
550 is huge.. I would sugguest to use the lowest detail selection on the mapgen for ov with banana pi CPU board.

Blaubart, 2025-05-22:
You mean „High resolution terrain“? I didn‘t activate. I generate the map always the same way. I have no idea why the map is so huge.

lordfolken, 2025-05-22:
> You mean „High resolution terrain“? I didn‘t activate. I generate the map always the same way. I have no idea why the map is so huge.

I meant level of detail. The topography not the terrain is causing the size increase.

Blaubart, 2025-05-24:
I always used the same settings. Is there a reason why the map is so huge now?

lordfolken, 2025-05-27:
Yes. because the OSM dataset is a lot more detailed and hence  larger, than the 25 years out of date vmap0 data. 

Alps_HighRes.xcm from the filemanager is 71mb http://download.xcsoar.org/source/map/region/

The same coordinates in the mapgen with LevelofDetail1 (which is comparable to what vmap0 had) is 127mb. 

But judging by the download sizes of other navigational maps, i would say that XCSoar is very moderate for the same area.

freevariode, 2025-05-28:
Thanks for explaining!!

closed 2025-06-24 by Blaubart