XCSoar/XCSoar #1769: XCSoar with a FT4232HL

Blaubart, 2025-05-19:
I use an Android device with XCSoar Testing (05.03.2025). I connected a FT4232 to use 4 serial ports for other devices. I programmed the FT4232 with FTProg so that manufacturer is FreeVario and device description ist Serial Port. If I go to NMEA Devices at XCSoar I see the 4 ports as "Serial Port (FreeVario)#1", "Serial Port (FreeVario)#2", "Serial Port (FreeVario)#3" and "Serial Port (FreeVario)#4". But if I try to use them, device is not connected. I also have "ttyUSB0", "ttyUSB1", "ttyUSB2" and "ttyUSB3". I have to use this ports to connect the device.

Here is a video: https://www.dropbox.com/scl/fi/093b8m8ln8uh3piahbb3w/IMG_4802.MOV?rlkey=di14v8izys4k4eizwhxb3z54i&dl=0

lordfolken, 2025-05-20:
Please attach an xcsoar.org log of when you try to attach these devices. (Please always include logs)

Also a listing of /dev/ files would be of interest.

https://github.com/XCSoar/XCSoar/blob/master/src/Device/Port/TTYEnumerator.cpp

Blaubart, 2025-05-24:
Here is the log file:

> [2025-05-24T14:04:44Z] Starting OpenSoar v7.43-3.24-Android git: 518ac9026f
> [2025-05-24T14:04:45Z] EGL vendor: Android
> [2025-05-24T14:04:45Z] EGL version: 1.4 Android META-EGL
> [2025-05-24T14:04:45Z] EGL extensions: EGL_ANDROID_front_buffer_auto_refresh EGL_ANDROID_get_frame_timestamps EGL_ANDROID_get_native_client_buffer EGL_ANDROID_presentation_time EGL_EXT_surface_CTA861_3_metadata EGL_EXT_surface_SMPTE2086_metadata EGL_KHR_get_all_proc_addresses EGL_KHR_swap_buffers_with_damage EGL_ANDROID_image_native_buffer EGL_ANDROID_native_fence_sync EGL_ANDROID_recordable EGL_EXT_create_context_robustness EGL_EXT_image_gl_colorspace EGL_EXT_pixel_format_float EGL_EXT_yuv_surface EGL_IMG_context_priority EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_fence_sync EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_mutable_render_buffer EGL_KHR_no_config_context EGL_KHR_partial_update EGL_KHR_surfaceless_context EGL_KHR_wait_sync EGL_NV_context_priority_realtime 
> [2025-05-24T14:04:45Z] EGL config: RGB=8/8/8 alpha=8 depth=24 stencil=8
> [2025-05-24T14:04:45Z] GL vendor: ARM
> [2025-05-24T14:04:45Z] GL version: OpenGL ES 3.2 v1.g7p1-01bet0.961949323ff055ba323dec41a0013d32
> [2025-05-24T14:04:45Z] GL renderer: Mali-G52
> [2025-05-24T14:04:45Z] GL extensions: GL_EXT_debug_marker GL_ARM_rgba8 GL_ARM_mali_shader_binary GL_OES_depth24 GL_OES_depth_texture GL_OES_depth_texture_cube_map GL_OES_packed_depth_stencil GL_OES_rgb8_rgba8 GL_EXT_read_format_bgra GL_OES_compressed_paletted_texture GL_OES_compressed_ETC1_RGB8_texture GL_OES_standard_derivatives GL_OES_EGL_image GL_OES_EGL_image_external GL_OES_EGL_image_external_essl3 GL_OES_EGL_sync GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_required_internalformat GL_OES_vertex_array_object GL_OES_mapbuffer GL_EXT_texture_format_BGRA8888 GL_EXT_texture_rg GL_EXT_texture_type_2_10_10_10_REV GL_OES_fbo_render_mipmap GL_OES_element_index_uint GL_EXT_shadow_samplers GL_OES_texture_compression_astc GL_KHR_texture_compression_astc_ldr GL_KHR_texture_compression_astc_hdr GL_KHR_texture_compression_astc_sliced_3d GL_EXT_texture_compression_astc_decode_mode GL_EXT_texture_compression_astc_decode_mode_rgb9e5 GL_KHR_debug GL_EXT_occlusion_query_boolean GL_EXT_disjoint_timer_query GL_EXT_blend_minmax GL_EXT_discard_framebuffer GL_OES_get_program_binary GL_OES_texture_3D GL_EXT_texture_storage GL_EXT_multisampled_render_to_texture GL_EXT_multisampled_render_to_texture2 GL_OES_surfaceless_context GL_OES_texture_stencil8 GL_EXT_shader_pixel_local_storage GL_ARM_shader_framebuffer_fetch GL_ARM_shader_framebuffer_fetch_depth_stencil GL_ARM_mali_program_binary GL_EXT_sRGB GL_EXT_sRGB_write_control GL_EXT_texture_sRGB_decode GL_EXT_texture_sRGB_R8 GL_EXT_texture_sRGB_RG8 GL_KHR_blend_equation_advanced GL_KHR_blend_equation_advanced_coherent GL_OES_texture_storage_multisample_2d_array GL_OES_shader_image_atomic GL_EXT_robustness GL_EXT_draw_buffers_indexed GL_OES_draw_buffers_indexed GL_EXT_texture_border_clamp GL_OES_texture_border_clamp GL_EXT_texture_cube_map_array GL_OES_texture_cube_map_array GL_OES_sample_variables GL_OES_sample_shading GL_OES_shader_multisample_interpolation GL_EXT_shader_io_blocks GL_OES_shader_io_blocks GL_EXT_tessellation_shader GL_OES_tessellation_shader GL_EXT_primitive_bounding_box GL_OES_primitive_bounding_box GL_EXT_geometry_shader GL_OES_geometry_shader GL_ANDROID_extension_pack_es31a GL_EXT_gpu_shader5 GL_OES_gpu_shader5 GL_EXT_texture_buffer GL_OES_texture_buffer GL_EXT_copy_image GL_OES_copy_image GL_EXT_shader_non_constant_global_initializers GL_EXT_color_buffer_half_float GL_EXT_unpack_subimage GL_EXT_color_buffer_float GL_EXT_YUV_target GL_OVR_multiview GL_OVR_multiview2 GL_OVR_multiview_multisampled_render_to_texture GL_KHR_robustness GL_KHR_robust_buffer_access_behavior GL_EXT_draw_elements_base_vertex GL_OES_draw_elements_base_vertex GL_EXT_protected_textures GL_EXT_buffer_storage GL_EXT_external_buffer GL_EXT_EGL_image_array GL_EXT_texture_filter_anisotropic GL_ARM_texture_unnormalized_coordinates GL_EXT_shader_framebuffer_fetch 
> [2025-05-24T14:04:45Z] Display dpi=60,59
> [2025-05-24T14:04:45Z] OpenGL: mda=false npot=true stencil=0x88f0
> [2025-05-24T14:04:45Z] Startup dialog
> [2025-05-24T14:04:45Z] Loading device port information
> [2025-05-24T14:04:45Z] Loaded profile from /storage/emulated/0/Android/media/de.opensoar/device_map.map
> [2025-05-24T14:04:45Z] Loading profiles
> [2025-05-24T14:04:45Z] Loaded profile from /storage/emulated/0/Android/media/de.opensoar/openvario.prf
> [2025-05-24T14:04:45Z] Loading language file
> [2025-05-24T14:04:45Z] Language: loading file '/storage/emulated/0/Android/media/de.opensoar/de.mo'
> [2025-05-24T14:04:45Z] Language: could not load file: Failed to open '/storage/emulated/0/Android/media/de.opensoar/de.mo': No such file or directory
> [2025-05-24T14:04:45Z] Language: loading resource 'de.mo'
> [2025-05-24T14:04:45Z] Loaded translations from resource 'de.mo'
> [2025-05-24T14:04:45Z] Loading input events file
> [2025-05-24T14:04:45Z] Invalid key data:  APP1 at 934
> [2025-05-24T14:04:45Z] Invalid key data:  APP2 at 941
> [2025-05-24T14:04:45Z] Invalid key data:  APP3 at 948
> [2025-05-24T14:04:45Z] Invalid key data:  APP4 at 955
> [2025-05-24T14:04:45Z] LoadTerrain
> [2025-05-24T14:04:45Z] Loading Topography File...
> [2025-05-24T14:04:45Z] ReadWaypoints
[... cut ...]
> [2025-05-24T14:05:34Z] Opening device USB Seriell: Serial Port (FreeVario)#4
> [2025-05-24T14:05:40Z] Opening device /dev/ttyUSB1
> [2025-05-24T14:05:46Z] Entering shutdown...
> [2025-05-24T14:05:46Z] Stop devices
> [2025-05-24T14:05:46Z] Stop threads
> [2025-05-24T14:05:46Z] Waiting for calculation thread
> [2025-05-24T14:05:46Z] delete MapWindow
> [2025-05-24T14:05:46Z] Save default task
> [2025-05-24T14:05:46Z] Close Progress Dialog
> [2025-05-24T14:05:46Z] Close Windows - main
> [2025-05-24T14:05:46Z] Finished shutdown

Blaubart, 2025-06-27:
at file usb_device_filter.xml line 153 is the reason for # at device names.