XCSoar/XCSoar #1784: Font and line thickness too small at high display resolution

freevariode, 2025-05-31:
I'm using a 7" Texim display (CH070OLDLTH-002-CHEF) with an Android board. I forgot to check the resolution in the settings, but I'll provide it later. I set the font size to 200%, but that's still too small. I think between 250 and 300% would be necessary. Also, the line that marks the maximum flight distance is so thin it's almost invisible. I have to turn off the topography to be able to see it. The line thickness should be thicker, or better, adjustable.

freevariode, 2025-06-01:
I checked the display settings, and it says 120 dpi. If I set the resolution in XCSoar to 120 dpi instead of automatic, the font is also large enough. Alternatively, I compiled an APK where you can set the font to 300%. This has roughly the same effect.

200%
[image]
[image]

300%

[image]
[image]

lordfolken, 2025-06-01:
Please use the bug tracker template and provide the basics, such as log files, hardware, xcsoar versions, etc.

freevariode, 2025-06-02:
ok, thanks

lordfolken, 2025-06-02:
I really need a log to know why your quite low resolution (800x480) (around 120dpI?)  device is having such dpi scaling issues. XCSoar scale's mostly well with current android devices that have 480 dpi and more. 

Please test with current master not with 2 year old code. https://download.xcsoar.org/testing/

freevariode, 2025-06-02:
Thanks a lot!! 

I will test changing dpi value, like kobedegeest has suggested, too. Next weekend I will test your version and send you a logfile!!

freevariode, 2025-06-05:
I installed XCSoar, that lordfolken has sent to me. Here are two log files

[xcsoar.log](https://github.com/user-attachments/files/20609950/xcsoar.log)
[xcsoar-old.log](https://github.com/user-attachments/files/20609951/xcsoar-old.log)

I think to adjust resolution to 120DPI is the best solution. Fonts and airfields are big enough. I did fly, so I couldn't test the line that marks the maximum flight distance

Blaubart, 2025-06-05:
I have checked the config file for the display. DPI would be 120:

#lvds,output 单：0 双：1
#DPI参数 支持80 100 120 160 200 240
dpi = 120;

freevariode, 2025-06-07:
DPI Checker says 120dpi. I testet with the smaller, 5.7 inch Texim display.

[image]

lordfolken, 2025-06-07:
You need to compare same with same, not a different display. 
What does xcsoar say in the log when you start it on that?

freevariode, 2025-06-08:
Sorry, You are right! I have this device at may workbench at the moment.
Display dpi=47,60