XCSoar/XCSoar #1790: Cannot decalre PEV start parameters to S100 (not 100% sure if this is a bug or feature request)

niemiha, 2025-06-06:
XCSoar version having the problem
-------------------------------------------------

I am using v 7.43-rc0 .

XCSoar versions not having the problem
-------------------------------------------------

I haven't tested any other versions

System information
------------------

I am running XCSoar on a dedicated Samsung Galaxy A34 5G, Android  version is 14. The S100 is connected via its internal Bluetooth.

Steps to reproduce the behavior
-------------------------------

When I set the task up, I also fill in the PEV start settings and declare the task to S100 via Bluetooth. The declaration goes without hiccups but when I activate the Pilot Event (and it shows on the S100 display) the S100 infobox just says WAIT! but no counter appears. I haven't found even a way to enter the PEV window settings manually for the task (not a XCSoar problem, I know).

And I want to emphasize that we are now talking about declaring the task, not using the XCSoar PEV functionality. I prefer the S100 display on the hectic start gaggle conditions.

Expected behavior
-----------------

If the task would have declared correctly, there should be a counter going down for the waiting time and then again for the opening time left on S100 info box.

Actual behavior
---------------

The S100 infobox shows only wait and the time is --:--.

Do you have any idea what may have caused this?
-----------------------------------------------

Internally not, but my hunch is that the PEV window parameters are not send as a part of the flight declaration.

Do you have an idea how to solve the issue?
-------------------------------------------

The PEV start window settings should be sent as a part of the S100 task declaration.

niemiha, 2025-06-06:
> In the meantime you should be able to manually set the timings in the S100, and XCSoar will trigger the pilot event and countdown (I believe people at my club do this)

After re-re-re-reading the manual I think I now know how to set the paramrters on the S100. Still would prefer the declaration to enter them too. 

h

lordfolken, 2025-06-07:
> This is not currently implemented in XCSoar. The PEV parameters are not sent on task declaration.
> 
> I don't know if it's possible for the S100 to receive these parameters, it would require some investigation, but maybe someone familiar with the LXNav driver would know better.

The way the lxnav driver sends a declaration is by basically uploading a text file. It doesn't care what the individual records are, as long as the s100 can read them. 

3.4 Read/Write Declaration (ASCII, BASE64)
Syntax:
Request1: PLXVC,DECL,R,<Start row>,<End row>
Request2: PLXVC,DECL,W,<Current line>,<Number of lines>,<Line content>