XCSoar/XCSoar #1803: Airspace Class "Unclassified" no more in Airspace Details

fraktal-sb, 2025-06-19:
XCSoar versions having the problem
-------------------------------------------------

7.44 current head

XCSoar versions not having the problem
-------------------------------------------------

7.44 previous head

System information
------------------

Unix, Windows

Steps to reproduce the behavior
-------------------------------

use french airspace and airspace FF-R150D1 and show it in both versions above

Expected behavior
-----------------

Airspace should be displayed in Details the same way

Actual behavior
---------------

see following screenshots. In addition airspace color has changed from red to pink.

[image]
[image]

Do you have any idea what may have caused this?
-----------------------------------------------

no

Do you have an idea how to solve the issue?
-------------------------------------------

change in airspace class handling

fraktal-sb, 2025-06-19:
adding

{ "UNCLASSIFIED", UNCLASSIFIED },

in 

XCSoar/src/Airspace/AirspaceParser.cpp

after

static constexpr AirspaceClassStringCouple airspace_class_strings[] = {

helps

[image]

colour is again red

lordfolken, 2026-04-06:
Current OpenAIP Naviter 2.0 exports use `AC UNC` for unclassified class; XCSoar maps `UNC` to `UNCLASSIFIED` and applies `AY` types as intended. OpenAIR 2.0 support is in 7.44 (see NEWS). The failure mode described here (missing Unclassified / wrong colours) matches older transitional parser/data behaviour when `AC` did not parse as ICAO or unclassified, so `AY` was skipped.

If anyone still sees this with a file that uses the long token `AC UNCLASSIFIED` or another edge case, please reopen and attach the airspace extract.